zephyrproject-rtos/zephyr #52057: tests: kernel: timer: starve: kernel.timer.starve hangs

nordic-piks, 2022-11-08:
**Describe the bug**
The tests/kernel/timer/starve/kernel.timer.starve hangs

Observed for
- nrf5340dk_nrf5340_cpunet
- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpuapp_ns
- nrf9160dk_nrf9160
- nrf9160dk_nrf9160_ns
- nrf52dk_nrf52832
- nrf52840dk_nrf52840
- 

Occurrence: Always

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf5340dk_nrf5340 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/timer/starve -p nrf5340dk_nrf5340--device-testing --device-serial /dev/ttyACM2 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2022-11-08T00:23:59.574Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.2.0-1303-g9cb1ff7fcee2 ***[0m
[2022-11-08T00:23:59.574Z] [0mRunning TESTSUITE starve_fn[0m
[2022-11-08T00:23:59.574Z] [0m===================================================================[0m
[2022-11-08T00:23:59.574Z] [0mSTART - test_starve[0m
[2022-11-08T00:23:59.574Z] [0mCycle clock runs at 32768 Hz[0m
[2022-11-08T00:23:59.574Z] [0mThere are 1 cycles per tick (32768 Hz ticks)[0m
[2022-11-08T00:23:59.574Z] [0m[     0.387] 1: still running, would pass at 3600 s[0m
[2022-11-08T00:23:59.574Z] [0m[    60.091] 594: still running, would pass at 3600 s[0m
[2022-11-08T00:23:59.574Z] [0m[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: v3.2.0-1303-g9cb1ff7fcee2

anangl, 2022-11-10:
The test does not hang. It just needs 60 minutes to complete (as the message says: "still running, would pass at 3600 s"). It is properly marked: https://github.com/zephyrproject-rtos/zephyr/blob/9b7d3657c9f715e2e47466c2a6e009e21688e43e/tests/kernel/timer/starve/testcase.yaml#L3 and when called with the command mentioned above, slightly corrected to:
```
./scripts/twister -T tests/kernel/timer/starve -p nrf5340dk_nrf5340_cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs
```
the test is skipped, so I don't see an issue here that could be corrected in the Zephyr tree.

The problem seems to be a too short timeout (2 minutes?) used when the test is somehow forced to be executed.

anangl, 2022-11-10:
> I don't see an issue here that could be corrected in the Zephyr tree.

It looks like I was a bit too quick with making this statement and closing the issue. `timeout: 3600` added in `testcase.yaml` could probably solve the problem. I'll create a PR with such patch.

PR #52134 (2022-11-11): tests: kernel: timer: starve: Add proper timeout value
The test in its default configuration needs 3600 seconds to complete, so use such timeout value in testcase.yaml so that twister called with --enable-slow option can successfully execute it.

Fixes #52057
closed 2022-11-11 by carlescufi