zephyrproject-rtos/zephyr #52059: Bluetooth: conn:  in multi role configuration incorrect adress is set after advertising resume

ppryga-nordic, 2022-11-08:
**Describe the bug**
Advertising resume in Host is limited operation in comparison to regular advertising enable operation. Limited in a way, there is no device address check/update before the advertising is resumed. When scanner is resumed and identity address is not used, Host uses NRPA address for a device when scanning. This address is different than static random address. When connection in peripheral role is disconnected, Host resumes advertising and continues use of address set by scan resume.
 
Now, assume that an application starts a connection in central role. Both functions `bt_le_create_conn_ext` and `bt_le_create_conn_legacy` call `bt_id_set_create_conn_own_addr`. The last one mentioned function attempts to change an address to static random address: https://github.com/zephyrproject-rtos/zephyr/blob/e104e031b7d3ddacd2b90527fff438fc4056b72a/subsys/bluetooth/host/id.c#L1519-L1528. Unfortunately that will not happen, because there is an advertising running and Controller returns error `BT_HCI_ERR_CMD_DISALLOWED` for Host command: `BT_HCI_OP_LE_SET_RANDOM_ADDRESS`.

It looks to me that, to fix the issue we need to run Bluetooth address setup before advertising is resumed. Unfortunately that would require storage of some information that are provided to Host when advertising is started by `bt_le_adv_start`.

**To Reproduce**
Steps to reproduce the behavior:
The issue may be reproduced according to steps in #50438.

**Expected behavior**
It should be possible to established a connection in central even though there is an advertising pending in the background.

**Impact**
Not sure how sever is the issue and what impact it has on end users.

**Additional context**
The issue was found during investigation of #50438.

jori-nordic, 2022-12-08:
doesn't seem to happen any more. I made a bsim test to reproduce it, but the DUT still uses the identity address (the static random you talk about I think) when re-starting advertising.

ppryga-nordic, 2022-12-09:
I would suggest to see in code to check if there were changes about it. In my humble opinion it would have been fixed if one changed advertising resume implementation, but I'm not following PRs in Host now.

jori-nordic, 2022-12-09:
I just reproduced it again, using the steps in the parent issue. I'll dig a bit more, see if downstream is missing any patches or I've missed something in the bsim test.

PR #52940 (2022-12-19): Bluetooth: host: adv: set the address in bt_le_adv_resume
Bluetooth: host: adv: set the address in bt_le_adv_resume

The address could have changed. The issue linked (and bsim test in previous
commit) shows a scenario where that could happen:

- we start scanning, host will use NRPA
- we start advertising, host will use identity address
- a device connects (as a central)
-> advertising resume fails for some unrelated reason
- another device connects (as a peripheral)
- the central device disconnects, we resume scanning
- the peripheral disconnects, the stack resumes advertising
-> but it mista [... cut]
commit dc4d63e9 (2022-12-19): Bluetooth: host: adv: set the address in bt_le_adv_resume

The address could have changed. The issue linked (and bsim test in previous
commit) shows a scenario where that could happen:

- we start scanning, host will use NRPA
- we start advertising, host will use identity address
- a device connects (as a central)
-> advertising resume fails for some unrelated reason
- another device connects (as a peripheral)
- the central device disconnects, we resume scanning
- the peripheral disconnects, the stack resumes advertising
-> but it mistakenly advertises using the NRPA set by the scanner

Fixes  [... cut]
closed 2022-12-19 by fabiobaltieri