zephyrproject-rtos/zephyr #52079: TLS handshake failure (after client-hello) with big_http_download sample

gokultv-private, 2022-11-08:
**Describe the bug**
A clear and concise description of what the bug is.
TLS handshake failure (after client-hello) with big_http_download sample

Please also mention any information which could help others to understand
the problem you're facing:
 - Target platform: native_posix
 - Captured the wireshark traces (attached here) and identified the 3-way handshake & TLS client-hello

**To Reproduce**
Steps to reproduce the behavior:
1. [terminal 1] Go to Zephyr tools folder & run "sudo ./net-setup.sh
2. [terminal 2] Compile: west build -b native_posix samples/net/sockets/big_http_download -p -- -DCONF_FILE="prj.conf overlay-tls.conf"
3. [terminal 2] run: west build -t run
4. [terminal 2] see the handshake error in the console

**Expected behavior**
TLS handshake should be successful and the TLS data communication should be started.

**Impact**
Major: Without HTTPS/TLS, we cannot proceed with our investigation.

**Logs and console output**
Uploaded wireshark traces -> tls-handshare-error.zip. Within this zip file, attached the following pcap files
 - tls.pcap -> contains full trace
 - tls-stream-3.pcap -> contains network trace relevant for this bug (TCP stream = 3)

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 20.04)
 - Toolchain: Zephyr version: 3.1.99; west version: 0.13.1
 - taken the 'big_http_download' sample

**Additional context**
 - Same behavior is seen, if we try to connect with any other HTTPS server.
 - If we disable TLS, the sample is able to download the file using HTTP.

[tls-handshake-error.zip](https://github.com/zephyrproject-rtos/zephyr/files/9963355/tls-handshake-error.zip)

laurenmurphyx64, 2022-11-08:
Setting as medium for now, assignee / networking folks can downgrade if necessary

rlubos, 2022-11-09:
It seems to be a problem with the web server the sample uses, I've encountered the issue recently. Either the certificate has expired and we need to use a new one in the sample, or we need to find a new server to connect to.

@gokultv-private In a meantime, you could evaluate https://docs.zephyrproject.org/latest/samples/net/sockets/http_get/README.html, I see no issues with TLS communication with  the server used by that sample.

Given that the issue appears to be releated to an individual sample, not the TLS stack per se, I think we could lower the issue priority to low.

gokultv-private, 2022-11-09:
Thank you @rlubos . The http_get sample is working for me. This tells the core TLS library is not having any issue.

Interestingly, if I replace the http_get sample to connect our internal HTTPS server, we are seeing the same issue. I believe it is not to do with the certificate but with the default supported cipher suites from the client-side.

I've attached the following wireshark traces,
1. http_get_unmodified.pcap -> Unmodified http_get sample connecting to https://google.com:443 and it is working as expected.
2. http_get_modified_to_new_server.pcap -> Modified http_get sample to use new HTTPS server URL & cert. Not working.

[http_get traces.zip](https://github.com/zephyrproject-rtos/zephyr/files/9972920/http_get.traces.zip)

I'll debug further & share the updates tomorrow.

rlubos, 2022-11-10:
> I believe it is not to do with the certificate but with the default supported cipher suites from the client-side.

The ciphersuites avaialble depend on the mbed TLS configuration (https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/mbedtls/Kconfig.tls-generic), if you know exactly which ciphersuite is expcted from the client you can enable specific configs, otherwise you could try to enable everything ([here's an example](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/benchmarks/mbedtls/prj.conf#L20)) and then optimize.

If you still see issues with the handshake after enabling the correct ciphersuite in mbed TLS, make sure the mbed TLS heap is large enough (`CONFIG_MBEDTLS_HEAP_SIZE`) the library is known to be a heavy heap user.

gokultv-private, 2022-11-11:
Thank you @rlubos . Once we enable all the cipher suites in mbedtls config, we are not seeing issues even with big_http_download!

But with our private server, when we connect we see the client_hello & server_hello, certificates - but after than the client is rejecting with the error 'bad certificate'. 
Our private server's root certificate is from 'Entrust Root Certification Authority (G2)', serial num: 4a 53 8c 28, expiry: 12/7/2030 and downloaded the certificate from [entrust_download_link](https://www.entrust.com/resources/certificate-solutions/tools/root-certificate-downloads) . The downloaded certificate is converted to der using the command 
`openssl x509 -outform der -in entrust_g2_ca.cer -out entrust_g2_ca.der`

I've attached the wireshark trace. Not sure what we are missing?
[hget_privsrv_entrust_ca.zip](https://github.com/zephyrproject-rtos/zephyr/files/9990913/hget_privsrv_entrust_ca.zip)

rlubos, 2022-11-14:
Did you set the hostname with `TLS_HOSTNAME` socket option accordingly? Note, that by default hostname verification is enabled and is a part of the certificate check (it must much the target certificate CN field). This check can be disabled by providing NULL pointer to the `TLS_HOSTNAME` socket option.

You can also try to enable mbed TLS logs (https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/benchmarks/mbedtls/prj.conf#L10) to see what's going under the hood of mbed TLS (it should give a more specific information about the reason of the failure than the alert message).

gokultv-private, 2022-11-14:
Thank you very much @rlubos . I found the root-cause. The issue is our server's (target) certificate CN field is different from its hostname (found it via wireshark). Surprisingly, the Windows/Google-browser UI is showing the actual hostname (when compared to the wireshark & mbedtls logs).

Until, our IT team fixes this certificate issue, I'm disabling the hostname verification.

PS:- let me know if you want to raise a separate issue to fix the http_big_download sample.

closed 2022-11-14 by gokultv-private