zephyrproject-rtos/zephyr #52103: STM32u5 dual bank flash issue

Vinay96110, 2022-11-09:
HI,
   I am using **stm32u575zi** board, which has **2mb flash size, with dual bank concept**. I am facing issue accessing the flash of 2ndbank i.e. starting from flash address 0x8100000 end of the flash am not able to access not even read also. but same am able to do it for 1st bank (0x8000000 to 0x8100000 ). while debugging I could see that it was failing in flash_page_layout function in file **flash_stm32l5_u5.c**. anything I am missing here please let me know.

kpyush, 2022-11-10:
@Vinay96110 Could you add some more detail and reproduction steps to the description?

FRASTM, 2022-11-10:
could this have a link with https://github.com/zephyrproject-rtos/zephyr/pull/51525 ?

Vinay96110, 2022-11-17:
@FRASTM  , that PR helped to solve my issue. now able to access 2nd bank as well. Thanks

FRASTM, 2022-12-01:
closed as the PR is now merged

closed 2022-12-01 by FRASTM