zephyrproject-rtos/zephyr #52105: Bluetooth: make the stacksizes for the ECDH processing thread and BT workqueue user configurable

vidarbe, 2022-11-09:
**Is your feature request related to a problem? Please describe.**
One of our users experience stack overruns in the ECDH processing thread and in the BT workqueue after they enable debugging options in their build, but they can't increase the stack sizes because the BT_LONG_WQ_STACK_SIZE and the BT_CTLR_ECDH_STACK_SIZE symbols are defined without a prompt.

**Describe the solution you'd like**
I would like the symbols to be made visible to allow users to increase the stack sizes when needed.

carlescufi, 2022-11-10:
`BT_LONG_WQ_STACK_SIZE ` should have a prompt since application code may execute in this context. The alternative is to ask the user to define an `app/Kconfig` and override the stack size there.
`BT_CTLR_ECDH_STACK_SIZE ` is an NCS option and should have a different default when `CONFIG_NO_OPTIMIZATIONS` is enabled.

tejlmand, 2022-11-11:
This is a topic to which there is not a clear answer.

Having sane calculated defaults on cases like this is very important, and something has definitely gone wrong if the default stack sizes does not consider optimization level (debug).

Of course having a prompt is a easy solution and is used generally by stack sizes in the kernel:
https://github.com/zephyrproject-rtos/zephyr/blob/main/kernel/Kconfig

but having a promt leads to other issues.
Must importantly the stuck Kconfig symbol syndrome.

But also the following case.
Example, today the default size of 1300 is found to be adequate when enabling `BT_GATT_CACHING`.
A user discovers that 1400 is needed when debug is enabled and sets this in prj.conf.

Later Zephyr maintainers discovers that 1300 is too small in some cases and increases the value to 1500.
Now, users who has 1400 in prj.conf could get stuck at this setting if they don't notice the change in the default.
Being stuck on a too low value can lead to lot of effort in tracing down the error.

The option of `<setting>_WITH_PROMPT` at first glance seems like a nice approach, but i'm not so fond of it.
Cause we can risk a lot of duplicated settings just to make the prompt available, and users would need to do:
```
# prj.conf
CONFIG_<setting>_WITH_PROMPT=y
CONFIG_<setting>=<my-value>
```

Another approach could be to introduce adjustment, like this:
```
config BT_LONG_WQ_STACK_SIZE_ADJUST
    int "Bluetooth long work queue size adjustment"
    default 0
    help
      This setting can be used for adjustment of the Bluetooth long work queue.
      The  BT_LONG_WQ_STACK_SIZE value will be adjusted with the value defined at this setting.
```

And then in the code here:
https://github.com/zephyrproject-rtos/zephyr/blob/6578b218ca02c22ec7f09b179d25d5f3a4a4cbdf/subsys/bluetooth/host/long_wq.c#L11
could be adjusted to:
```
K_THREAD_STACK_DEFINE(bt_lw_stack_area, CONFIG_BT_LONG_WQ_STACK_SIZE + CONFIG_BT_LONG_WQ_STACK_SIZE_ADJUST);
```

That would leave the user with only a single setting to set, while at the same time benefit from any changes that might happen to default size due to ongoing Zephyr development.
```
# prj.conf
CONFIG_BT_LONG_WQ_STACK_SIZE_ADJUST=+200 # Give me 200 bytes more. (Note, the + sign can be omitted)
```

The `BT_LONG_WQ_STACK_SIZE` setting is kept promptless and thus never impacted by the stuck symbol syndrome.

I believe such an approach would be nice, but I can also accept giving the setting a prompt, as that is what is already the pattern used in Kernel Kconfig.

denisbohm, 2022-11-16:
I brought up this issue with Nordic a couple of times.  Each major release the apps I'm working on for clients start having stack faults when built with CONFIG_NO_OPTIMIZATIONS=y.

From the above discussion it seems like there are a couple of things with the way stack sizes are set:
1) If user code can't run on a thread then Zephyr intentionally tries to make those stack sizes non-settable.  This only works if the Zephyr maintainers consider every possible compiler setting and compiler and architecture that will be used and tests for all of those.  That clearly isn't the case today, so seems like a bad policy.
2) If user code can run on a thread then Zephyr has an additional _WITH_PROMPT symbol to enable users to change the stack size.  This just seems to complicate life.  What is the benefit?

From my user point of view I would like to see:
1) All stack sizes are user configurable without changing Zephyr / NCS sources.  And ideally, without the extra _WITH_PROPMT.
2) There is at least a "release" stack size and a "debug" stack size for every stack.  When any debug compile option is on then the "debug" size is used.  The debug stack size should be conservative so it works under all conditions.

jori-nordic, 2023-01-20:
> The only solution for now seems to be to fork Zephyr / NCS and make changes and then reference that fork from all the projects I'm working on.

@denisbohm @alwa-nordic will look into this soon, but in the meantime, you don't have to fork/patch Zephyr.
You can just create a file named `Kconfig` in your app folder (next to the `prj.conf`) with the options you want to override, e.g. if you want to set 2000 as the size for both, it can look like this:
```
config BT_LONG_WQ_STACK_SIZE
	int "Long workqueue stack size"
	default 2000

config BT_CTLR_ECDH_STACK_SIZE
	int "ECDH HCI thread stack size"
	default 2000

source "Kconfig.zephyr"
```

denisbohm, 2023-02-21:
Another stack that isn't configurable and is overrun with no optimizations:

In zephyr/drivers/bluetooth/hci/spi.c:
static K_KERNEL_STACK_DEFINE(spi_rx_stack, 512);

denisbohm, 2023-02-21:
Not sure why this was changed to enhancement from bug.  Crashing isn't a bug?

jori-nordic, 2023-02-22:
Yes it is, but the user can override the stack sizes in question. See my previous comment.
For the SPI stack, I made this PR https://github.com/zephyrproject-rtos/zephyr/pull/55062.

PR #55062 (2023-02-22): Bluetooth: HCI: Make driver stack sizes configurable
Adding them as hidden kconfig options to avoid the stuck kconfig syndrome. Users are free to redefine them in their app to force a value.

Fixes #52105
closed 2023-02-22 by carlescufi