zephyrproject-rtos/zephyr #52111: Incompatible LTO version of liblt_9518_zephyr.a

aectaan, 2022-11-09:
Target: TLSR9518ADK80D
Zephyr versions:  v3.2.0, main.
Host OS: Ubuntu 22.04
```
lto1: fatal error: bytecode stream in file '/workdir/modules/hal/telink/tlsr9/ble/proj_lib/lib/liblt_9518_zephyr.a' generated with LTO version 9.3 instead of the expected 12.0
compilation terminated.
lto-wrapper: fatal error: /home/egor/.local/zephyr-sdk-0.15.1/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc returned 1 exit status
compilation terminated.
/home/egor/.local/zephyr-sdk-0.15.1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.1.0/../../../../riscv64-zephyr-elf/bin/ld.bfd: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
Reproduce:
add `CONFIG_FPU=y` to zephyr/boards/riscv/tlsr9518adk80d/tlsr9518adk80d_defconfig
``` bash
west build -p -b tlsr9518adk80d ../zephyr/samples/bluetooth/peripheral
```

Expected: successful build

nordicjm, 2022-11-09:
@carlescufi Query on this, the command above seems to imply that a closed source blob is being included in a hal repo - see https://github.com/zephyrproject-rtos/hal_telink/tree/main/tlsr9/ble/proj_lib

I was under the impression that blobs had to be fetched using `west blobs` and could not be included in the default zephyr instance?

aectaan, 2022-11-10:
@andy-liu-telink Could you please fix it? I would like to use main Zephyr repo instead of Telink fork

jhedberg, 2024-05-24:
@AlexKolosovTelink are you be able to help out here? If this is not fixed I think we'll have to remove B91 Bluetooth support from Zephyr.

jhedberg, 2024-05-25:
I've opened a pull request to remove the HCI driver, since it turns out it doesn't build even with the binary blob made manually available: #73289

andy-liu-telink, 2024-05-27:
@jhedberg 
I have notified the proper team members to look into this issue.

@s07641069
@andriy-bilynskyy 
Could you take a look?

andriy-bilynskyy, 2024-05-29:
Since HCI driver is no longer available #73289 the initial issue is no longer valid

jhedberg, 2024-05-29:
> Since HCI driver is no longer available #73289 the initial issue is no longer valid

That PR hasn't been merged yet, but from the responses I'd conclude that there are no objections to proceeding with the removal of the HCI driver.

PR #73289 (2024-05-30): Bluetooth: drivers: Remove unmaintained B91 HCI driver
The driver isn't currently buildable due to "west blobs" support never having been added for hal_telink. Furthermore, even if the blob dependency is manually made available it turns out the code has bitrotten to the point where it doesn't build anymore. This situation has continued for several years without anyone taking action, so I think it's safe to assume this is unmaintained and should be removed.

Fixes #52111