zephyrproject-rtos/zephyr #52125: Timer accuracy issue with STM32U5 board

vibhor-meshram, 2022-11-10:
**Describe the bug**
I am using b_u585i_iot02a board and running samples/drivers/counter/alarm example on it. I modified the code to toggle a GPIO pin in the interrupt handler so that can be observed on DSO. 

**Observation**
If alarm is configured for 1 ms (auto reload) the interrupt would trigger 5 us early. Attached is the DSO screenshot. 

**Expected behavior**
The timer interrupt should trigger every 1 ms .

**Impact**
HIGH

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: gnuarmemb
 - Commit SHA or Version used: 3.1.0
[image]
[main.txt](https://github.com/zephyrproject-rtos/zephyr/files/9977882/main.txt)
[overlay.txt](https://github.com/zephyrproject-rtos/zephyr/files/9977884/overlay.txt)

FRASTM, 2022-11-10:
Do you get a better accuracy with LSE 32768 kHz as RTC clock source by :
CONFIG_COUNTER_RTC_STM32_CLOCK_LSE=y
CONFIG_COUNTER_RTC_STM32_CLOCK_LSI=n

vibhor-meshram, 2022-11-11:
> Do you get a better accuracy with LSE 32768 kHz as RTC clock source by : CONFIG_COUNTER_RTC_STM32_CLOCK_LSE=y CONFIG_COUNTER_RTC_STM32_CLOCK_LSI=n

@FRASTM This configuration works. But I need microseconds accuracy. With RTC 1sec = 1 tick.  So instead of RTC I have configured timer2 as counter. Using timer2 I am getting the results shown above.

vibhor-meshram, 2022-11-11:
> Do you get a better accuracy with LSE 32768 kHz as RTC clock source by : CONFIG_COUNTER_RTC_STM32_CLOCK_LSE=y CONFIG_COUNTER_RTC_STM32_CLOCK_LSI=n

@FRASTM This works. But I need microsecond accuracy. With RTC 1sec = 1 tick. So instead of RTC I configured timer2 in this example. Using timer2 I get the results shown above.

FRASTM, 2022-11-14:
In that case would you be able to compare what is written to the timer2 autoreload register compared to the 1ms value.
Maybe be the conversion factors from ms to ticks and from ticks to timer2 counter units inducing an uncertainty.

vibhor-meshram, 2022-11-15:
> In that case would you be able to compare what is written to the timer2 counter compared to the 1ms value. Maybe be the conversion factors from ms to ticks and from ticks to timer2 counter units induce some uncertainty.

@FRASTM I am using 159 pre-scaler in my .overlay file. As per that 1000 us = 1000 ticks (this I got from counter_us_to_ticks() api ) and value of CCR1 register is 1001.

`*** Booting Zephyr OS build zephyr-v3.1.0-21-gf6d4aba54f53  ***`
`Counter alarm sample`
`Set alarm in 0.000000 sec (1000 ticks)`
`Timer Control CCR1 = 1001`

FRASTM, 2022-11-15:
@vibhor-meshram that could help a lot if you could share your sample based on zephyr-v3.2.0

vibhor-meshram, 2022-11-21:
@FRASTM Any updates/suggestions ?

FRASTM, 2022-11-21:
I have measured 
[image]
 on my nucleo_u575zi_q with LSE as RTC clock source

FRASTM, 2022-11-22:
> @FRASTM Any updates/suggestions ?

in the board overlay, I set :
```
&rtc {
	/delete-property/ clocks;
        status = "okay";
        clocks = <&rcc STM32_CLOCK_BUS_APB3 0x00200000>,
		 <&rcc STM32_SRC_LSE RTC_SEL(1)>;
	label = "rtc";
};
```
to source RTC by the LSE clock.

vibhor-meshram, 2022-11-25:
> I have measured on my nucleo_u575zi_q and b_u585i_iot02a with LSE as RTC clock source. On the PB6 I get 1.0 ms:
> 
> [image]

@FRASTM  I am unable to go less than 1 second using LSE as RTC clock source. Is there anything else that needs to be modified other than .overlay file? How did you configure 1ms alarm?

FRASTM, 2022-11-25:
Nothing special, I took your example (main.c) and your overlay files and put them in a  new samples/boards/stm32/timer_pulse. A counter node was included in the timers2 node of the stm32u5 device Tree.
See my branch [here](https://github.com/FRASTM/zephyr/tree/counter_u5/).

vibhor-meshram, 2022-11-25:
Thanks @FRASTM I am now getting the same results as yours. I am closing this issue now.

closed 2022-11-25 by vibhor-meshram