zephyrproject-rtos/zephyr #52139: ppp modem doesn't send NET_EVENT_L4_CONNECTED event

wojciechslenska, 2022-11-10:
**Describe the bug**
In the current zephyr version, ppp modem doesn't send the event `NET_EVENT_L4_CONNECTED`, and the network is not working. The issue is related probably to the following pull request https://github.com/zephyrproject-rtos/zephyr/pull/51240

**Test setup**
 gsm_modem sample with the following extra configuration
```
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=y
CONFIG_MODEM_GSM_QUECTEL=y
CONFIG_MODEM_GSM_APN="apn"
```

**Proper behavior**
Test on zephyr version cf23b312c350 which doesn't contain https://github.com/zephyrproject-rtos/zephyr/pull/51240

```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-762-gcf23b312c350  ***
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x8013ec8 (gsm-modem)
uart:~$ sample resume
[00:00:16.015,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:16.336,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:16.389,000] <inf> modem_gsm: Model: BG95-M3
[00:00:16.443,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:16.496,000] <inf> modem_gsm: IMEI: 123456
[00:00:16.550,000] <inf> modem_gsm: IMSI: 123456
[00:00:16.604,000] <inf> modem_gsm: ICCID: 123456
[00:00:16.661,000] <inf> modem_gsm: Attached to packet service!
[00:00:36.747,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:36.782,000] <inf> sample_gsm_ppp: Network connected
uart:~$ net ping 8.8.8.8
PING 8.8.8.8
28 bytes from 8.8.8.8 to 10.234.18.44: icmp_seq=0 ttl=58 time=260 ms
28 bytes from 8.8.8.8 to 10.234.18.44: icmp_seq=1 ttl=58 time=249 ms
28 bytes from 8.8.8.8 to 10.234.18.44: icmp_seq=2 ttl=58 time=168 ms
```
The callback is executed and the ping is working.

**Current version**
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-1477-gc9e07fc38515 ***
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x80140ec (gsm-modem)
uart:~$ sample resume
[00:00:22.445,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:22.768,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:22.821,000] <inf> modem_gsm: Model: BG95-M3
[00:00:22.874,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:22.928,000] <inf> modem_gsm: IMEI: 123456
[00:00:22.982,000] <inf> modem_gsm: IMSI: 123456
[00:00:23.036,000] <inf> modem_gsm: ICCID: 123456
[00:00:23.092,000] <inf> modem_gsm: Attached to packet service!
[00:00:43.179,000] <inf> net_ppp: Initializing PPP to use serial@40004400
uart:~$ net ping 8.8.8.8
PING 8.8.8.8
Cannot send IPv4 ping
```
Callback `Network connected` does not exist in the log and the ping is also not working.

**Workaround**
With the following workaround, the current zephyr version is working properly
```
--- a/subsys/net/l2/ppp/ppp_l2.c
+++ b/subsys/net/l2/ppp/ppp_l2.c
@@ -282,6 +282,7 @@ static void carrier_on_off(struct k_work *work)
 
        if (ppp_carrier_up) {
                ppp_mgmt_raise_carrier_on_event(ctx->iface);
+               net_if_up(ctx->iface);
                net_if_carrier_on(ctx->iface);
        } else {
                if (ppp_lcp) {
```
logs
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-1477-gc9e07fc38515 ***
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x80140f4 (gsm-modem)
uart:~$ sample resume
[00:00:17.090,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:17.413,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:17.466,000] <inf> modem_gsm: Model: BG95-M3
[00:00:17.519,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:17.573,000] <inf> modem_gsm: IMEI: 123456
[00:00:17.626,000] <inf> modem_gsm: IMSI: 123456
[00:00:17.680,000] <inf> modem_gsm: ICCID: 123456
[00:00:17.736,000] <inf> modem_gsm: Attached to packet service!
[00:00:37.823,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:37.857,000] <inf> sample_gsm_ppp: Network connected
uart:~$ net ping 8.8.8.8
PING 8.8.8.8
28 bytes from 8.8.8.8 to 10.99.154.233: icmp_seq=0 ttl=58 time=229 ms
28 bytes from 8.8.8.8 to 10.99.154.233: icmp_seq=1 ttl=58 time=230 ms
28 bytes from 8.8.8.8 to 10.99.154.233: icmp_seq=2 ttl=58 time=137 ms
```

**Impact**
PPP modem is not working

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: SDK 0.15.1
 - Zephyr: c9e07fc38515351aebd25d87f3e87f0d3e1dcfc9

rlubos, 2022-11-10:
Given the workaround you've provided and `CONFIG_PPP_NET_IF_NO_AUTO_START=y` in your config file I assume that the PPP interface is not brought up. You need to either call `net_if_up()` manually from your application or disable `CONFIG_PPP_NET_IF_NO_AUTO_START` if you want the interface to start automatically.

wojciechslenska, 2022-11-10:
When I disable this flag `CONFIG_PPP_NET_IF_NO_AUTO_START` situation looks similar, but in that case, my workaround is also not working.
***old zephyr***
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-762-gcf23b312c350  ***
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x8013ed4 (gsm-modem)
uart:~$ sample resume
[00:00:12.466,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:12.789,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:12.842,000] <inf> modem_gsm: Model: BG95-M3
[00:00:12.895,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:13.113,000] <inf> modem_gsm: Attached to packet service!
[00:00:33.200,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:33.234,000] <inf> sample_gsm_ppp: Network connected
uart:~$ net ping 8.8.8.8
PING 8.8.8.8
28 bytes from 8.8.8.8 to 10.108.194.41: icmp_seq=0 ttl=58 time=334 ms
28 bytes from 8.8.8.8 to 10.108.194.41: icmp_seq=1 ttl=58 time=129 ms
28 bytes from 8.8.8.8 to 10.108.194.41: icmp_seq=2 ttl=58 time=121 ms
```
**new zephyr**
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0-1477-gc9e07fc38515 ***
[00:00:00.003,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:00.010,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x80140e8 (gsm-modem)
uart:~$ sample resume
[00:00:22.150,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:22.472,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:22.525,000] <inf> modem_gsm: Model: BG95-M3
[00:00:22.578,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:22.796,000] <inf> modem_gsm: Attached to packet service!
[00:01:40.004,000] <inf> modem_gsm: GSM network interface down
```
Also please noticed that I enable my modem manually using the command `modem resume` which call `gsm_ppp_start` function.

rlubos, 2022-11-10:
@wojciechslenska Can you please enable `CONFIG_NET_IF_LOG_LEVEL_DBG=y` and `CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y` and post the logs? It should shed some light on what's going on under the hood.

rlubos, 2022-11-10:
That's with `CONFIG_PPP_NET_IF_NO_AUTO_START=y`, I'm interested to see what's happening if the option is disabled (interface is brought up on boot). Currently the second log just shows that the interface remains in the admin DOWN state (but the carrier goes ON which is a good sign).

wojciechslenska, 2022-11-10:
```
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=n
CONFIG_MODEM_GSM_QUECTEL=y
CONFIG_NET_IF_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y
```
log
```
[00:00*** Booting Zephyr OS build zephyr-v3.2.0-1477-gc9e07fc38515 ***
:00.000,000] <dbg> net_if: net_if_init: (main): 
[00:00:00.000,000] <dbg> net_if: init_iface: (main): On iface 0x200003c0
[00:00:00.003,000] <dbg> net_if: update_operational_state: (main): iface 0x200003c0, oper state DOWN admin DOWN carrier OFF dormant OFF
[00:00:00.003,000] <dbg> net_if: net_if_post_init: (main): 
[00:00:00.003,000] <dbg> net_if: net_if_up: (main): iface 0x200003c0
[00:00:00.003,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:00.003,000] <dbg> net_if: update_operational_state: (main): iface 0x200003c0, oper state DOWN admin UP carrier OFF dormant OFF
[00:00:00.003,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state UP admin UP carrier ON dormant OFF
[00:00:00.009,000] <dbg> conn_mgr: conn_mgr_initial_state: (conn_mgr): Iface 0x200003c0 UP
[00:00:00.009,000] <dbg> conn_mgr: conn_mgr_handler: (conn_mgr): Connection Manager started
[00:00:00.009,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:00.010,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x80148e0 (gsm-modem)
[00:00:10.003,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:20.003,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
uart:~$ sample resume
[00:00:23.810,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:24.132,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:24.185,000] <inf> modem_gsm: Model: BG95-M3
[00:00:24.238,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:24.455,000] <inf> modem_gsm: Attached to packet service!
[00:00:30.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:40.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:50.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:01:00.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:01:10.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:01:20.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:01:30.004,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:01:40.004,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state DOWN admin UP carrier OFF dormant OFF
[00:01:40.004,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface event 3489726465 received on iface 1 (0x200003c0)
[00:01:40.004,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface index 0
[00:01:40.004,000] <inf> modem_gsm: GSM network interface down

```

wojciechslenska, 2022-11-15:
@rlubos I checked your fix, unfortunately not working
***config***
```
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=y
CONFIG_MODEM_GSM_QUECTEL=y
CONFIG_NET_IF_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y
```
***log***
```
[00:*** Booting Zephyr OS build zephyr-v3.2.0-1501-g2566cb386dc2 ***
00:00.000,000] <dbg> net_if: net_if_init: (main): 
[00:00:00.000,000] <dbg> net_if: init_iface: (main): On iface 0x200003c0
[00:00:00.003,000] <dbg> net_if: update_operational_state: (main): iface 0x200003c0, oper state DOWN admin DOWN carrier OFF dormant OFF
[00:00:00.003,000] <dbg> net_if: net_if_post_init: (main): 
[00:00:00.009,000] <dbg> conn_mgr: conn_mgr_handler: (conn_mgr): Connection Manager started
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x80147f4 (gsm-modem)
uart:~$ sample resume
[00:00:06.803,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:07.126,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:07.179,000] <inf> modem_gsm: Model: BG95-M3
[00:00:07.232,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:07.450,000] <inf> modem_gsm: Attached to packet service!
uart:~$ 
[00:00:27.537,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state DOWN admin DOWN carrier ON dormant OFF
[00:00:27.537,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:00:27.537,000] <err> os: ***** BUS FAULT *****
[00:00:27.537,000] <err> os:   Precise data bus error
[00:00:27.537,000] <err> os:   BFAR Address: 0x8
[00:00:27.537,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x0000007e  r2/a3:  0x00000021
[00:00:27.537,000] <err> os: r3/a4:  0x20002747 r12/ip:  0x00006b91 r14/lr:  0x0801397d
[00:00:27.537,000] <err> os:  xpsr:  0x81000000
[00:00:27.537,000] <err> os: Faulting instruction address (r15/pc): 0x08013950
[00:00:27.537,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:27.537,000] <err> os: Current thread: 0x20000ba8 (ppp_tx)
[00:00:27.654,000] <err> os: Halting system
```
***config***
```
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=n
CONFIG_MODEM_GSM_QUECTEL=y
CONFIG_NET_IF_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y
```
***log***
```
[00:00*** Booting Zephyr OS build zephyr-v3.2.0-1501-g2566cb386dc2 ***
:00.000,000] <dbg> net_if: net_if_init: (main): 
[00:00:00.000,000] <dbg> net_if: init_iface: (main): On iface 0x200003c0
[00:00:00.003,000] <dbg> net_if: update_operational_state: (main): iface 0x200003c0, oper state DOWN admin DOWN carrier OFF dormant OFF
[00:00:00.003,000] <dbg> net_if: net_if_post_init: (main): 
[00:00:00.003,000] <dbg> net_if: net_if_up: (main): iface 0x200003c0
[00:00:00.003,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:00.003,000] <dbg> net_if: update_operational_state: (main): iface 0x200003c0, oper state DOWN admin UP carrier OFF dormant OFF
[00:00:00.009,000] <dbg> conn_mgr: conn_mgr_handler: (conn_mgr): Connection Manager started
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x80147e0 (gsm-modem)
uart:~$ sample resume
[00:00:14.017,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:14.339,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:14.391,000] <inf> modem_gsm: Model: BG95-M3
[00:00:14.445,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:14.662,000] <inf> modem_gsm: Attached to packet service!
[00:00:34.749,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state UP admin UP carrier ON dormant OFF
[00:00:34.749,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface event 3489726466 received on iface 1 (0x200003c0)
[00:00:34.749,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface index 0
[00:00:34.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:00:44.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:00:54.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:01:04.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:01:14.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:01:24.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:01:34.749,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:01:44.750,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:01:54.750,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:02:04.750,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ed4, prio 1) network packet iface 0x200003c0/1
[00:02:14.750,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state DOWN admin UP carrier OFF dormant OFF
[00:02:14.750,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface event 3489726465 received on iface 1 (0x200003c0)
[00:02:14.750,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface index 0
[00:02:14.750,000] <inf> modem_gsm: GSM network interface down
```

rlubos, 2022-11-15:
> I checked your fix, unfortunately not working

That's a pity. Unfortunately, w/o a hardware I don't really see a way to investigate the issue further. Can you please check if reverting https://github.com/zephyrproject-rtos/zephyr/commit/dd535f611d6c81275e64f2fe849a033e7e49f8bb helps? Note, that `net_if_carrier_down()` no longer exists, so it'll have to be replaced with `net_if_down()`.

wojciechslenska, 2022-11-15:
For the following setup everything is working
***setup***
Zephyr commit: dae6d2f84b47cd6c295296dde7dd7d97c40d652d
+ revert
```
    Revert "net: ppp: Align PPP driver/L2 with iface state upgrade"
    
    This reverts commit dd535f611d6c81275e64f2fe849a033e7e49f8bb.
```
+ changes
```
--- a/subsys/net/l2/ppp/lcp.c
+++ b/subsys/net/l2/ppp/lcp.c
@@ -210,7 +210,7 @@ static void lcp_finished(struct ppp_fsm *fsm)
        /* take the remainder down */
        ppp_mgmt_raise_carrier_off_event(ctx->iface);
 
-       net_if_carrier_down(ctx->iface);
+       net_if_down(ctx->iface);
 }
 
 #if defined(CONFIG_NET_L2_PPP_OPTION_MRU)
diff --git a/subsys/net/l2/ppp/ppp_l2.c b/subsys/net/l2/ppp/ppp_l2.c
index 102f4a7463..81a4e9a29c 100644
--- a/subsys/net/l2/ppp/ppp_l2.c
+++ b/subsys/net/l2/ppp/ppp_l2.c
@@ -291,7 +291,7 @@ static void carrier_on_off(struct k_work *work)
                        ppp_change_phase(ctx, PPP_DEAD);
 
                        ppp_mgmt_raise_carrier_off_event(ctx->iface);
-                       net_if_carrier_down(ctx->iface);
+                       net_if_down(ctx->iface);
                }
        }
 }
```

***config***
```
CONFIG_GSM_PPP_AUTOSTART=n
CONFIG_PPP_NET_IF_NO_AUTO_START=n
CONFIG_MODEM_GSM_QUECTEL=y
CONFIG_NET_IF_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y
```
***logs***
```
[00:*** Booting Zephyr OS build zephyr-v3.2.0-1607-g11931e27cc80 ***
00:00.000,000] <dbg> net_if: net_if_init: (main): 
[00:00:00.000,000] <dbg> net_if: init_iface: (main): On iface 0x200003c0
[00:00:00.003,000] <dbg> net_if: net_if_post_init: (main): 
[00:00:00.009,000] <dbg> conn_mgr: conn_mgr_handler: (conn_mgr): Connection Manager started
[00:00:00.009,000] <inf> sample_gsm_ppp: Board 'b_u585i_iot02a' APN 'apn' UART 'serial@40004400' device 0x8014868 (gsm-modem)
uart:~$ sample resume
[00:00:11.346,000] <inf> sample_gsm_ppp: GSM modem on callback fired
[00:00:11.669,000] <inf> modem_gsm: Manufacturer: Quectel
[00:00:11.722,000] <inf> modem_gsm: Model: BG95-M3
[00:00:11.775,000] <inf> modem_gsm: Revision: BG95M3LAR02A03
[00:00:11.993,000] <inf> modem_gsm: Attached to packet service!
[00:00:32.080,000] <inf> net_ppp: Initializing PPP to use serial@40004400
[00:00:32.080,000] <dbg> net_if: net_if_up: (sysworkq): iface 0x200003c0
[00:00:32.080,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state UP admin UP carrier ON dormant OFF
[00:00:32.080,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:32.081,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface event 3489726466 received on iface 1 (0x200003c0)
[00:00:32.081,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface index 0
[00:00:32.086,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:32.093,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:32.094,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:32.103,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:32.106,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[... 39 lines cut ...]
[00:00:44.030,000] <dbg> net_if: net_if_up: (sysworkq): iface 0x200003c0
[00:00:44.031,000] <dbg> net_if: update_operational_state: (sysworkq): iface 0x200003c0, oper state UP admin UP carrier ON dormant OFF
[00:00:44.031,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:44.032,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface event 3489726466 received on iface 1 (0x200003c0)
[00:00:44.032,000] <dbg> conn_mgr: conn_mgr_iface_events_handler: (net_mgmt): Iface index 0
[00:00:44.037,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:44.044,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007edc, prio 1) network packet iface 0x200003c0/1
[00:00:44.045,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:44.054,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:44.057,000] <dbg> net_if: net_if_tx: (ppp_tx): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:44.065,000] <dbg> net_if: net_if_ipv4_addr_add: (rx_q[0]): [0] interface 0x200003c0 address 10.164.244.74 type MANUAL added
[00:00:44.065,000] <dbg> conn_mgr: conn_mgr_ipv4_events_handler: (net_mgmt): IPv4 event 3758358529 received on iface 1 (0x200003c0)
[00:00:44.065,000] <dbg> conn_mgr: conn_mgr_ipv4_events_handler: (net_mgmt): Iface index 0
[00:00:44.065,000] <dbg> conn_mgr: conn_mgr_ipv4_status: (conn_mgr): IPv4 connected on iface index 1
[00:00:44.065,000] <dbg> conn_mgr: conn_mgr_notify_status: (conn_mgr): Iface 1 (0x200003c0) connected
[00:00:44.065,000] <inf> sample_gsm_ppp: Network connected
uart:~$ net ping 8.8.8.8
PING 8.8.8.8
28 bytes from 8.8.8.8 to 10.164.244.74: icmp_seq=0 ttl=58 time=1392 ms
28 bytes from 8.8.8.8 to 10.164.244.74: icmp_seq=1 ttl=58 time=393 ms
28 bytes from 8.8.8.8 to 10.164.244.74: icmp_seq=2 ttl=58 time=142 ms
[00:00:49.368,000] <dbg> net_if: net_if_tx: (shell_uart): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:50.373,000] <dbg> net_if: net_if_tx: (shell_uart): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
[00:00:51.378,000] <dbg> net_if: net_if_tx: (shell_uart): Processing (pkt 0x20007ea0, prio 1) network packet iface 0x200003c0/1
```

PR #52257 (2022-11-16): net: ppp: Revert interface management changes
This reverts changes introduced in commit dd535f611d6c81275e64f2fe849a033e7e49f8bb, as they broke the gsm_ppp driver integration with PPP L2. Apparently, a more thorough refactoring is needed to use the new interface management scheme with PPP.

Fixes #52139
closed 2022-11-16 by carlescufi