zephyrproject-rtos/zephyr #52169: Mesh provisioning static OOB incorrect zero padding

eanderlind, 2022-11-10:
https://github.com/zephyrproject-rtos/zephyr/blob/f98a240537f34da826f4dc6eb95f63b4e8bb845b/subsys/bluetooth/mesh/prov_device.c#L162

`
if (atomic_test_bit(bt_mesh_prov_link.flags, OOB_STATIC_KEY)) {
		memcpy(bt_mesh_prov_link.auth + 16 - bt_mesh_prov->static_val_len,
		       bt_mesh_prov->static_val, bt_mesh_prov->static_val_len);
		(void)memset(bt_mesh_prov_link.auth, 0,
			     sizeof(bt_mesh_prov_link.auth) - bt_mesh_prov->static_val_len);
	}
`

It appears the auth field for staticOOB has incorrect zero-padding. Per MeshPrf sec 5.4.2.4 a binary value 
shall be padded with zeros **after** the (variable length) array of octets, ie format shall be same as for alphanumeric string. 
Code currently follows numeric format where zeros are pre-pended.

alxelax, 2022-11-14:
Hi @eanderlind, I think the current implementation does exactly what the specification claims. Let's take a look at the subclause `5.4.2.4 Authentication`.
Spec doesn't claim Static OOB is Numeric or Alphanumeric value. Static OOB is just some binary value.
Then take a look at samples how to build 128-bit AuthValue from different OOB types.

For Numeric type:
```
For example, if the Authentication with Output OOB method is used with Output OOB Action as Display Number and the
displayed number is 019655, then the AuthValue shall be 0x00000000000000000000000000004CC7. 
The AuthValue is then encoded as defined in Section 5.1 in order to compute confirmation values, resulting in an array 
consisting of [0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0xC7].
```
For Alphanumeric type:
```
For example, if the Authentication with Output OOB method is used with Output OOB Action as Display String and the
displayed string is ”123ABC”, then the AuthValue shall be 0x31323341424300000000000000000000, resulting in an array
consisting of [0x31, 0x32, 0x33, 0x41, 0x42, 0x43, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00].
```
For Binary data (Static OOB case):
```
If the value is shorter than 128 bits, the remaining bits shall be set to 0.
For example, if the value is [0x12, 0x34, 0x56], the AuthValue is an array consisting of 
[0x12, 0x34, 0x56, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00].
```

As you can see AuthValue composing for Binary and Alphanumeric looks the same. The data is stored in the most significant octets of the memory area. The least significant octets ate padded to zero. It is exactly what the source code, that you highlighted, does.

Hopefully, I clarified the situation with padding.

eanderlind, 2022-11-14:
Hi @alxelax,
Agree that alphanumeric and staticOOB shall have same format.

In prov_device.c lines 162+  the buffer bt_mesh_prov_link.auth is set for staticOOB by **prepending zeros**

In prov.c for the case of input_alphanumeric
  1. the function get_auth_string() on lines 162-163 creates the expected result by **appending zeros**
  2. in bt_mesh_input_string() on line 316, copies the string to bt_mesh_prov_link.auth buffer
  3. in bt_mesh_prov_auth() on line 267 the previous function is invoked.

Similarly looking at cases of output auth with alphanumeric string, the .auth buffer is constructed the same way and I couldn't find anywhere that the buffer is subsequently reversed for one auth method, but not another.

It appears the implementation currently populates the buffer differently for alphanumeric and staticOOB.
Looking at how .auth is constructed for numeric, it appears that the case of staticOOB is the faulty one.

alxelax, 2022-11-15:
yeah, seems, @eanderlind, you are right. I didn't consider this subclause.
```
5.1 Endianness
Unless stated otherwise, all multiple-octet numeric values in this layer shall be “big endian”, as described
in Section 3.1.1.1.
```
Static OOB should append zeroes since spec sample is in big endian.  
Thanks for this finding.

closed 2022-11-16 by carlescufi