zephyrproject-rtos/zephyr #52171: Inappropriate l2cap channel state set/get

jerryyang35, 2022-11-11:
**Describe the bug**
`void bt_l2cap_chan_set_state(struct bt_l2cap_chan *chan, bt_l2cap_chan_state_t state)`
`static inline bt_l2cap_chan_state_t bt_l2cap_chan_get_state(struct bt_l2cap_chan *chan)`
`void bt_l2cap_chan_set_state_debug(struct bt_l2cap_chan *chan, bt_l2cap_chan_state_t state, const char *func, int line)`
In above API, the state of l2cap channel is only handle by `BT_L2CAP_LE_CHAN`. But the state had better to be handled according to the l2cap channel type.

**To Reproduce**
Steps to reproduce the behavior:
1. Create br l2cap connection between to boards.
2. Master does not send configure req.
3. Fail to establish l2cap connection.

**Expected behavior**
Master is able to send configure req to slave after received connection rep.

**Impact**
 - When a br l2cap connection is tried to be established, the connection state cannot be set correctly. This will cause master fail to send 'configure request' to slave after received 'connection response', which does not follow l2cap protocol.

**Logs and console output**
[image]

**Refer scheme**
-previous
`		BT_L2CAP_LE_CHAN(chan)->state = state;`
-after
`    struct bt_conn* temp_conn = chan->conn;`
`    if (temp_conn->type == BT_CONN_TYPE_LE) {`
`		BT_L2CAP_LE_CHAN(chan)->state = state;`
`    }`
`	else {`
`		BR_CHAN(chan)->state = state;`
`    }`

jhedberg, 2022-11-14:
BR/EDR is an experimental feature of the Zephyr Bluetooth stack, and not very well supported. You are welcome to propose a fix through a poll request, if you want.

carlescufi, 2022-11-17:
@jerryyang35 as @jhedberg mentioned, BR/EDR is just not maintained anymore, nor tested. If you are interested in reviving this feature please let us know and we can discuss how to make accommodations for this.