zephyrproject-rtos/zephyr #52186: Add options for handling pending bit when enabling/disabling GPIO interrupt

LSChyi, 2022-11-14:
**Is your feature request related to a problem? Please describe.**
Currently the gpio driver only exposes an API `pin_interrupt_configure` for enabling/disabling the interrupt, yet this function did not specify the detailed behavior for whether the pending bit is cleared.
This causes the application behaves differently when using different chips. It would be nice to have the ability to control the behavior of whether to clear the pending bit when enabling/disabling GPIO interrupt.

**Describe the solution you'd like**
Exposes new GPIO functions `enable` and `disable` that merely turn on/off interrupt register bit, but not changing other registers, such as the pending bit register.

**Describe alternatives you've considered**
Extends the existing `pin_interrupt_configure` API to accept more parameters. However, I think this may cause compatibility issue.

LSChyi, 2022-11-16:
Create PR #52273 to better illustrate the idea.

keith-zephyr, 2022-11-30:
Instead of adding two new GPIO driver APIs, can you extend `api->pin_interrupt_configure` to support this mode?

Currently the function `gpio_pin_interrupt_configure()` processes the `gpio_flags_t flags` parameter to set the `enum gpio_int_mode` and `enum gpio_int_trig` parameters passed to the driver.

You could extend `enum gpio_int_mode` as follows:
```
#define GPIO_INT_ENABLE_DISABLE_ONLY (1U << 27)
enum gpio_int_mode {
	GPIO_INT_MODE_DISABLED = GPIO_INT_DISABLE,
	GPIO_INT_MODE_LEVEL = GPIO_INT_ENABLE,
	GPIO_INT_MODE_EDGE = GPIO_INT_ENABLE | GPIO_INT_EDGE,
	GPIO_INT_MODE_DISABLE_ONLY = GPIO_INT_DISABLE | GPIO_INT_ENABLE_DISABLE_ONLY,
	GPIO_INT_MODE_ENABLE_ONLY = GPIO_INT_ENABLE | GPIO_INT_ENABLE_DISABLE_ONLY,
};
```

The new public APIs `gpio_pin_interrupt_enable/disable()` routines would set `enum gpio_int_mode` as needed and call the existing `api->pin_interrupt_configure`.

henrikbrixandersen, 2022-12-01:
Another option could be to specify the detailed behavior for whether the pending bit is cleared when disabling an interrupt via `pin_interrupt_configure`. I would prefer that over adding yet another set of GPIO API calls.

LSChyi, 2022-12-02:
I think specifying the detailed behavior is something we should do in the documentation, and adapting the approach from @keith-zephyr is good. Let me update the PR.

LSChyi, 2023-04-07:
PR merged, thank you all!

closed 2023-04-07 by LSChyi