zephyrproject-rtos/zephyr #52189: Flash file system not working on stm32h735g_disco

maxstreitberger, 2022-11-14:
**Describe the bug**
I'm trying to use the [usb/mass](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/mass) sample to create a file system on the external flash of the `stm32h735g_disco` board. I have followed the [documentation](https://docs.zephyrproject.org/latest/services/storage/disk/access.html) and created a block device for the flash partition. 

```
/ {
	storage_disk {
		compatible = "zephyr,flash-disk";
		partition = <&storage_partition>;
		disk-name = "NAND";
		cache-size = <4096>;
                sector-size = <4096>;
	};
};
```
My whole flash looks like this:
```
&octospi1 {
	mx25lm51245: ospi-nor-flash@0 {
		compatible = "st,stm32-ospi-nor";
		reg = <0>;
		ospi-max-frequency = <DT_FREQ_M(50)>;
		size = <DT_SIZE_M(64)>;
		spi-bus-width = <OSPI_OPI_MODE>;
		data-rate = <OSPI_DTR_TRANSFER>;
		status = "okay";
		sfdp-bfp = [
			53 46 44 50 06 01 02 ff
			00 06 01 10 30 00 00 ff
			C2 00 01 04 10 01 00 ff
			84 00 01 02 C0 00 00 ff
			00 00 00 00
		];

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			storage_partition: partition@0 {
				label = "nor";
				reg = <0x00000000 DT_SIZE_M(4)>;
			};
		};
	};
};
```
I removed the usb code from the sample, because it gave me an error and I actually do not need it. However, the flash code is untouched.
When I'm running this command:
```
west build -d build-flash/ -b stm32h735g_disco -s zephyr/samples/subsys/usb/mass/ -p always -- -DCONFIG_APP_MSC_STORAGE_FLASH_FATFS=y && west flash -d build-flash/
```

I receive this output:
```
Area 0 at 0x0 on ospi-nor-flash@0 for 4194304 bytes
[00:00:00.100,000] <dbg> flash_stm32_ospi: stm32_ospi_config_mem: OSPI configuring OctoSPI mode
[00:00:00.140,000] <inf> flash_stm32_ospi: OSPI flash config is OPI / DTR
[00:00:00.140,000] <dbg> flash_stm32_ospi: flash_stm32_ospi_init: ospi-nor-flash@0: SFDP v 1.6 AP ff with 3 PH
[00:00:00.140,000] <dbg> flash_stm32_ospi: flash_stm32_ospi_init: PH0: ff00 rev 1.6: 16 DW @ 30
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Unexpected flash size: 8
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: ospi-nor-flash@0: 0 MiBy flash
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Erase 0 with 00
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Address width: 4 Bytes
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Page size 1 bytes
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Flash size 8 bytes
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Using read mode: 0, instr: 0x0, dummy cycles: 0
[00:00:00.140,000] <dbg> flash_stm32_ospi: spi_nor_process_bfp: Using write instr: 0x12ED
[00:00:00.140,000] <dbg> flash_stm32_ospi: flash_stm32_ospi_init: PH1: ffc2 rev 1.0: 4 DW @ 110
[00:00:00.140,000] <dbg> flash_stm32_ospi: setup_pages_layout: layout 8388608 x 1 By pages
[00:00:00.140,000] <dbg> flash_stm32h7: stm32h7_flash_init: Flash initialized. BS: 32
[00:00:00.140,000] <dbg> flash_stm32h7: stm32h7_flash_init: Block 0: bs: 131072 count: 8
[00:00:00.140,000] <dbg> flash_stm32h7: flash_stm32h7_write_protection: Disable write protection
[00:00:00.145,000] <dbg> disk: disk_access_register: disk interface(NAND) registered
[00:00:00.150,000] <inf> flashdisk: Initialize device NAND
[00:00:00.150,000] <inf> flashdisk: offset 0, sector size 4096, page size 1, volume size 4194304
[00:00:00.150,000] <dbg> flash_stm32_ospi: flash_stm32_ospi_read: OSPI: read 1 data
[00:00:00.150,000] <dbg> flash_stm32_ospi: HAL_OSPI_RxCpltCallback: Rx Cplt cb
[00:00:00.150,000] <dbg> flash_stm32_ospi: flash_stm32_ospi_read: OSPI: read 1 data
[00:00:05.151,000] <err> flash_stm32_ospi: 1: Failed to send OSPI instruction
[00:00:05.151,000] <err> fs: fs mount error (-5)
[00:00:05.151,000] <err> main: Failed to mount filesystem
```
The expected behavior is:
```
Area 0 at 0x0 on ospi-nor-flash@0 for 4194304 bytes
Mount /NAND:: 0
/NAND:: bsize = 512 ; frsize = 512 ; blocks = 216 ; bfree = 216
/NAND: opendir: 0
End of files
```

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.15.0
 - Latest `main` commit

FRASTM, 2022-11-14:
@maxstreitberger could you please share what you mean by "removed the usb code from the sample"
also could you take https://github.com/zephyrproject-rtos/zephyr/pull/47962 and check

maxstreitberger, 2022-11-14:
@FRASTM here is what I removed:
prj.conf:
```
#USB related configs
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr MSC sample"
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
CONFIG_USB_MASS_STORAGE_LOG_LEVEL_ERR=y
```
Kconfig:
```
config USB_DEVICE_PID
	default USB_PID_MASS_SAMPLE
```
main.c:
```
#include <zephyr/usb/usb_device.h>
setup_disk();
int ret = usb_enable(NULL);
if (ret != 0) {
    LOG_ERR("Failed to enable USB");
    return;
}
```

FRASTM, 2022-11-14:
Then what I get is 
```
*** Booting Zephyr OS build zephyr-v3.2.0-1312-gb052143077a0 ***
Area 0 at 0x0 on ospi-nor-flash@0 for 4194304 bytes
[00:00:00.140,000] <inf> flash_stm32_ospi: OSPI flash config is OPI / DTR
[00:00:00.140,000] <inf> flash_stm32_ospi: Read SFDP from octoFlash
[00:00:00.140,000] <inf> flash_stm32_ospi: Read SFDP from octoFlash
[00:00:00.150,000] <err> fs: fs mount error (-5)
[00:00:00.150,000] <err> main: Failed to mount filesystem
[00:00:00.150,000] <inf> main: The device is put in USB mass storage mode.
```

Laczen, 2022-11-14:
@maxstreitberger if you use a sector-size of 4096 you will use the disk as if it has a sector-size of 4096 byte instead of the fat standard 512byte. I don't think you want this. The cache-size is what you want to be equal to the flash sector-size.

maxstreitberger, 2022-11-15:
@FRASTM I looked into https://github.com/zephyrproject-rtos/zephyr/issues/49406 to see how I can fix the `flash_stm32_ospi: OSPI wr in OPI/STR mode is for 32bit address only` error and I found out that I need to change line `1469` in `zephyr/drivers/flash/flash_stm32_ospi.c` to 
```
if (jesd216_bfp_addrbytes == JESD216_SFDP_BFP_DW1_ADDRBYTES_VAL_3B4B)
```
Which is also done in https://github.com/zephyrproject-rtos/zephyr/pull/47962

maxstreitberger, 2022-11-15:
But I still get multiple flashdisk initializations, which is a bit odd
```
*** Booting Zephyr OS build zephyr-v3.2.0-1561-g67e8d03280ad ***
Area 0 at 0x0 on ospi-nor-flash@0 for 4194304 bytes
Mount /NAND:: 0
/NAND:: bsize = 512 ; frsize = 2048 ; blocks = 2038 ; bfree = 2038
/NAND: opendir: 0
End of files
[00:00:00.140,000] <inf> flash_stm32_ospi: OSPI flash config is OPI / DTR
[00:00:00.150,000] <inf> flashdisk: Initialize device NAND
[00:00:00.150,000] <inf> flashdisk: offset 0, sector size 512, page size 4096, volume size 4194304
[00:00:00.150,000] <inf> flashdisk: Initialize device NAND
[00:00:00.150,000] <inf> flashdisk: offset 0, sector size 512, page size 4096, volume size 4194304
[00:00:00.221,000] <inf> flashdisk: Initialize device NAND
[00:00:00.221,000] <inf> flashdisk: offset 0, sector size 512, page size 4096, volume size 4194304
[00:00:00.282,000] <inf> main: The device is put in USB mass storage mode.
```

maxstreitberger, 2022-11-15:
@Laczen I initially thought that `sector-size` and `page-size` meant the same thing, so that is the reason why I had `sector-size = <4096>`

de-nordic, 2022-11-15:
People get confused by what sector-size and page-size do all the time...

Laczen, 2022-11-15:
> People get confused by what sector-size and page-size do all the time...

Yes, maybe it should be renamed to `disk-sector-size` in the binding yaml.

FRASTM, 2022-11-18:
@Laczen, Do you think of something like : 
```
diff --git a/dts/bindings/misc/zephyr,flash-disk.yaml b/dts/bindings/misc/zephyr,flash-disk.yaml
index 5098858801..ea0d52b97b 100644
--- a/dts/bindings/misc/zephyr,flash-disk.yaml
+++ b/dts/bindings/misc/zephyr,flash-disk.yaml
@@ -20,14 +20,14 @@ properties:
       description: |
         Disk name.
 
-    sector-size:
+    disk-sector-size:
       type: int
       required: false
       default: 512
       description: |
         Emulated block device sector size in bytes.
 
-    cache-size:
+    disk-cache-size:
       type: int
       required: true
       description: |
```

jfischer-no, 2022-11-21:
Inability to read description of the property and resulting confusion is not a bug in Zephyr.

stephanosio, 2022-11-22:
Closing since this is not a bug. Please file a separate enhancement issue if you think this is necessary.

closed 2022-11-22 by stephanosio