zephyrproject-rtos/zephyr #52190: [lpcxpresso55s28] flash range starts from NS address which is not compatible with latest Jlink config

hakehuang, 2022-11-14:
**Describe the bug**
in lpcxpresso55s28 dts config, starts from 0x10000000

build/zephyr/include/generated/autoconf.h:13:#define CONFIG_FLASH_BASE_ADDRESS 0x10000000
```
     iap: flash-controller@34000 {
                                ranges = < 0x0 0x10000000 0x3020000 >;

```
which is not a NS address and Jlink V7.82 does not support flash to this address. see below error

```
J-Link>loadfile "/home/shared/disk/zephyr_project/zephyr_lpcxpresso55s28/zephyr/samples/hello_world/build/zephyr/zephyr.bin" 0x10000000
'loadfile': Performing implicit reset & halt of MCU.
ResetTarget() start
Reset strategy selected in debugger ignored. Performing device specific reset.
Reset handler addr: 0x10001030
ResetTarget() end
AfterResetTarget() start
AfterResetTarget() end
Downloading file [/home/shared/disk/zephyr_project/zephyr_lpcxpresso55s28/zephyr/samples/hello_world/build/zephyr/zephyr.bin]...
Writing target memory failed.
```
and if write to 0x0 works.

Please also mention any information which could help others to understand
the problem you're facing:
- all lpcxpresso55 series has the same problem

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b lpcxpresso55s28
4.  west flash --runner=jlink

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
showstopper

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK,)
 - Commit SHA or Version used: zephyr-v3.2.0-1553-g45e1ff94cd

hakehuang, 2022-11-14:
@dleach02 , @mmahadevan108

dleach02, 2022-11-16:
@hake, this appears to be a problem with all the different versions of jlink I've tried. If you change the device type to LPC5528JBD64 then it works. We haven't seen this before because the board is setup to use pyocd runner by default.

dleach02, 2022-11-17:
Functionality of Segger changed with version v7.54b

dleach02, 2022-11-18:
Segger confirmed the bug and will fix this in next week's release (11/21/2022)

dleach02, 2022-11-23:
Confirmed fixed with v7.82d

closed 2022-11-23 by dleach02