zephyrproject-rtos/zephyr #52218: ADC read locked forever with CONFIG_DEBUG_OPTIMISATION=y on STM32U5

tpennors, 2022-11-14:
**Describe the bug**
ADC read locked forever with `CONFIG_DEBUG_OPTIMISATION=y` on STM32U5
I used gdb to analyse this. I saw that the code gets locked in : 
`drivers/adc/adc_context.h: k_sem_take(&ctx->sync, K_FOREVER);`
I checked and `adc_stm32_isr` is never called.

When using `CONFIG_NO_OPTIMISATION=y`, everything is fine.

**Temporary fix**
I found a temporary fix but I would like to understand if it is a real bug from the driver. My solution is to add a very little waiting period between those two calls: `LL_ADC_EnableIT_EOC(adc);` and `adc_context_start_read(&data->ctx, sequence);`

**Other**
I didn't use overlay as it is described in `sample/drivers/adc/boards/stm32l562e_dk.overlay`, would that be a prolem ?

nordicjm, 2022-11-14:
Probably a stack overflow, increase the stack sizes and try again

tpennors, 2022-11-15:
Hello,

I tried to increase stack size and still doesn't work. Hardware interrupt is never received. I would have think of a timing between writing to registers which is not the same between optimizations. Did anyone see this kind of behaviour ?