zephyrproject-rtos/zephyr #52219: jlink runner doesn't flash bin file if a hex file is present

MeisterBob, 2022-11-14:
**Describe the bug**
If you try to flash a binary using `west flash -r jlink --bin-file path/to/my.bin` but have `CONFIG_BUILD_OUTPUT_HEX` enable the jlink runner will always flash the generated .hex file and ignore the .bin file it got explicitly given in the call.

**To Reproduce**
See description

**Expected behavior**
if `--elf-file`, `--hex-file` or `--bin-file` argument is present west should only pass this file to the runner so the runner knows which file the user wants to flash.

**Impact**
After post-processing the binary you can't flash it (using west).

**Logs and console output**
```bash
➭ west flash -r jlink --bin-file build/zephyr/post_processed.bin --skip-rebuild
-- west flash: using runner jlink
-- runners.jlink: JLink version: 7.70e
loadfile "/workspaces/impinj_ex10_app/build/zephyr/zephyr.signed.hex"
-- runners.jlink: Flashing file: /workspaces/impinj_ex10_app/build/zephyr/zephyr.signed.hex
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA 21141042239980edb03332f938958e446e56b7c2

nordicjm, 2022-11-14:
If you're flashing a bin, why not disable hex output, since you aren't using it?

MeisterBob, 2022-11-14:
That's correct, but why should west flash the hex when the user explicitly asked to have the bin flashed?

mbolivar-nordic, 2022-11-15:
It's bad UX, but the meaning of the --bin-file argument is not "flash this bin file". It's "if you are going to flash a bin file, use this one instead of the default one". That causes confusion as it is here but it's not a bug. Sorry for the confusion and thanks for taking the time to file an issue.

mbolivar-nordic, 2022-11-15:
Additional references: https://github.com/zephyrproject-rtos/zephyr/issues/23945 https://github.com/zephyrproject-rtos/zephyr/pull/18159#issuecomment-535715795

closed 2022-11-15 by mbolivar-nordic

MeisterBob, 2022-11-15:
The way I implemented it in #51365 it respects the `--hex-file`, `--bin-file` and `--elf-file` with the only drawback that it falls back to the .elf file if the given hex/bin file doesn't exist.

mbolivar-nordic, 2022-11-15:
I filed an enhancement which summarizes the proposed solution that has gotten the most acceptance over the years: https://github.com/zephyrproject-rtos/zephyr/issues/52262

I don't have any plans to implement this, but would of course review with an eye towards merging an implementation of htis.