zephyrproject-rtos/zephyr #52242: Fatal exception LoadProhibited from mcuboot when enabling newlib in wifi sample (esp32)

rillbert, 2022-11-15:
Hi,

The current main branch of mcuboot on an esp32 gives me a 'Fatal exception (28): LoadProhibited' when trying to run `samples/net/wifi/` after enabling newlib. (similar to #46093).

My target platform is esp32 in the form of a ESP-DevkitC v4 dev board with the ESP-WROOM-32UE module with 4MB flash.

### This succeeds
- build mcuboot from current main branch (36bac4f6a54a3) for the esp32 platform using the espressif docker image as toolchain.  
`docker run --rm -v $PWD:/project -w /project espressif/idf <my-build-script.sh>`)

- flash mcuboot to target using `esptool.py`

- build the `samples/net/wifi/` project for esp32 within zephyr after setting `CONFIG_BOOTLOADER_MCUBOOT=y` in proj.conf
`west build -p -b esp32 samples/net/wifi/`

- flash the application
`west -v flash`

The application loads and then presents the following on the serial console (mcuboot was compiled with log_level_debug in this case):
```
[esp32] [INF] Disabling RNG early entropy source...
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [DBG] flash_area_id_from_multi_image_slot
[esp32] [DBG] flash_area_open: ID=1
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=1 size=20 src_addr=10020 src_addr_aligned=10000
[esp32] [INF] DRAM segment: start=0xcde0, size=0x1d04, vaddr=0x3ffb0000
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=1 size=1d04 src_addr=1cde0 src_addr_aligned=10000
[esp32] [INF] IRAM segment: start=0xeae4, size=0x14b1c, vaddr=0x40080000
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=3 size=14b1c src_addr=1eae4 src_addr_aligned=10000
[esp32] [INF] start=0x400935d0
�*** Booting Zephyr OS build zephyr-v3.2.0-1221-g95202ae3da4e ***
running network sample

uart:~$ 
```

### This fails
- Add `CONFIG_NEWLIB_LIBC=y` to the prj.config file

- build and flash the application in the same way as above

mcuboot shows the following on the serial console and then the system reboots
```
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=1 size=20 src_addr=10020 src_addr_aligned=10000
[esp32] [INF] DRAM segment: start=0xda7c, size=0x2544, vaddr=0x3ffb0000
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=1 size=2544 src_addr=1da7c src_addr_aligned=10000
[esp32] [INF] IRAM segment: start=0xffc0, size=0x1c4b8, vaddr=0x40080000
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=3 size=1c4b8 src_addr=1ffc0 src_addr_aligned=10000
Fatal exception (28): LoadProhibited
```

**Impact**
Depending on the cause, this is a showstopper for me. My actual goal is to build the `samples/subsys/mgmt/hawkbit` sample but that currently fails with the below IllegalInstruction exception so my strategy was to backtrack to a simpler sample where I could reproduce the error by turning one knob (newlib enablement).

```
[esp32] [INF] Disabling RNG early entropy source...
[esp32] [INF] br_image_off = 0x10000
[esp32] [INF] ih_hdr_size = 0x20
[esp32] [DBG] flash_area_id_from_multi_image_slot
[esp32] [DBG] flash_area_open: ID=1
[esp32] [INF] Loading image 0 - slot 0 from flash, area id: 1
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=1 size=20 src_addr=10020 src_addr_aligned=10000
[esp32] [INF] DRAM segment: start=0xe0f4, size=0x2604, vaddr=0x3ffb0000
[esp32] [DBG] [bootloader_flash] mmu set paddr=00010000 count=2 size=2604 src_addr=1e0f4 src_addr_aligned=10000
[esp32] [INF] IRAM segment: start=0x106f8, size=0x1ba0c, vaddr=0x40080000
[esp32] [DBG] [bootloader_flash] mmu set paddr=00020000 count=2 size=1ba0c src_addr=206f8 src_addr_aligned=20000
Fatal exception (0): IllegalInstruction
epc1=0x210002e0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```

**Environment:**
 - OS: Ubuntu 22.04
 - Toolchainf. Zephyr sdk-0.15.1
 - Commit SHA for zephyrproject: 95202ae3da4e8c41a
 - Commit SHA for mcuboot: 36bac4f6a54a3
 
I am not experienced enough within this platform/ecosystem to know if this is a problem with mcuboot, a space issue (too little embedded RAM??) or something else. When reading the memory layout for the resulting elf files, it looks to me as it should fit in available memory though.

Regards
/Anders

rillbert, 2022-11-16:
After testing different things, I have found something that seems to work...at least so far into my yet limited testing.

If I change the memory layout in `<mcuboot_repo_root>/boot/espressif/port/esp32/ld/bootloader.ld` to mimic the values found in `<mcuboot_repo_root>/boot/espressif/hal/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld` both the wifi sample and the hawkbit sample runs!

The updated memory layout looks as:
```
MEMORY
{
  iram_seg (RWX) :                  org = 0x40080400, len = 0xfc00
  iram_loader_seg (RWX) :           org = 0x40078000, len = 0x8000
  dram_seg (RW) :                   org = 0x3FFF0000, len = 0x10000

  /* 
  the original layout for mcuboot
  iram_seg (RWX) :                  org = 0x40093000, len = 0x8800
  iram_loader_seg (RWX) :           org = 0x4009B800, len = 0x4800
  dram_seg (RW) :                   org = 0x3FFF5000, len = 0x8900 
  */
}
```

With my limited knowledge of this platform, it does make some sense since:
 1.  the comments in the esp-idf bootloader.ld file provides arguments for choosing these exact ranges for the bootloader.
 2. The entry point for the chain-loaded application are within the original iram_seg of mcuboot (eg `0x400935d0`) which feels a bit odd.

I don't have a clue if the changed memory layout has other negative impacts but I will continue my testing with the wifi and hawkbit applications now that they actually runs.

Any comments or suggestions are greatly appreciated.

rillbert, 2022-11-17:
How about this theory:

1. the mcuboot iram_loader_seg is set to start at 0x40098000
2. in my two failed cases above, the size of the IRAM segment is 0x1ba0c and 0x1c4b8 respectively. 
3. The application's IRAM segment starts at 0x40080000.

The above would mean that mcuboot will overwrite the start of its own iram_loader_seg with the chain-loaded application's IRAM segment and bad things will probably happen.

In the case above that succeeds, the size of the application's IRAM segment is 0x14b1c, meaning that it will end at 0x40094b1c and thus the mcuboot iram_loader_seg would remain intact after application loading.

sylvioalves, 2022-11-22:
@rillbert, thanks for pointing this out. I'll start looking into this soon.

rillbert, 2022-11-22:
Thanks @sylvioalves !

I am currently focused on trying to understand/fix this so I will be glad to help in any way I can.

sylvioalves, 2022-11-29:
@rillbert, I've addressed the issue regarding mcuboot/zephyr IRAM overlapping.  
The reason it happened is that ESP32 MCUBoot porting in upstream's got new features related to security. During this process, IRAM area needs to be increase, which make less room for IRAM in Zephyr's linker. And we haven't updated Zephyr's.  
The PR linked to this issue makes room for newlibc + mcuboot + wifi.  
Once we make MCUboot build working within Zephyr's environment, this IRAM region optimization will take place.

PR #52629 (2022-11-30): driver: wifi: esp32: linker changes to free up iram
This PR adds some linker changes to increase IRAM area.
This also fixes IRAM length when MCUBoot is enabled.
Fix build warnings when newlibc is enabled.  

Fixes #52242
closed 2022-11-30 by carlescufi