zephyrproject-rtos/zephyr #52267: http client includes chunking data in response body when making a non-chunked request and server responds with chunked data

mrclark25, 2022-11-15:
**Describe the bug**
When making an HTTP get request, the http server I am communicating with responds with chunked data. The trailing chunking information is included in the http response body returned by zephyr. The server is able to fit the entire response in one "chunk", so chunking isn't really used in this case but the chunking data is still included anyway (as per http spec most likely?).

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? nrf52840
 - What have you tried to diagnose or workaround this issue? My device is receiving JSON, so I am able to accommodate this bug by removing trailing characters from the request body until the closing json bracket is found.

**To Reproduce**
Steps to reproduce the behavior:
1. open a socket like:
\`\`\`
m_socketID = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
int status = connect(m_socketID, &m_resolvedDNSAddress->ai_addr, m_resolvedDNSAddress->ai_addrlen);
\`\`\`
2. make http request like:
\`\`\`
struct http_request req;
memset(&req, 0, sizeof(req));
req.method = HTTP_GET;
req.url = URL;
req.host = HOSTNAME;
req.protocol = "HTTP/1.1";
req.payload = m_payload.c_str();
req.payload_len = m_payload.length();
req.response = nonMemberHandleHTTPResponse;
req.recv_buf = m_response.data();
req.recv_buf_len = m_response.size();
http_client_req(m_socketID, &req, WEB_OPERATION_TIMEOUT, nullptr);
\`\`\`
3. If the server responds with the header "Transfer-Encoding: chunked" and accompanying chunking data at the end of the packet, see an ascii zero and extra whitespace at the end of the body returned by the http client.

This zero and extra whitespace is the chunking data sent by the http server which is not part of the response body.

from: https://en.wikipedia.org/wiki/Chunked_transfer_encoding
HTTP chunking example
\`\`\`
4\r\n        (bytes to send)
Wiki\r\n     (data)
6\r\n        (bytes to send)
pedia \r\n   (data)
E\r\n        (bytes to send)
in \r\n
\r\n
chunks.\r\n  (data)
0\r\n        (final byte - 0)
\r\n         (end message)
\`\`\`

The last two lines are the exact extra bytes I see in the http response body.

**Environment:**
 - OS: Windows 10 compiling for and running on nrf52840
 - Toolchain: Zephyr SDK, c++ enabled
 - Zephyr version 3.2.0
 - nrf52840 is making the http request via a BG95 modem that is using the generic gsm driver

rlubos, 2022-12-06:
Hi @mrclark25,

I don't think it's really a bug, but rather the nature of this http_client. The callback function registered with `http_request->response` is expected to receive "raw" http response, including chunking headers. If you're interested in receiving body payload data only, you can register `on_body` callback with `http_request->http_cb->on_body` pointer.

carlescufi, 2022-12-21:
Closing this due to inactivity and based on @rlubos analysis.

closed 2022-12-21 by carlescufi