zephyrproject-rtos/zephyr #52276: ST Kit B-L4S5I-IOT01A Octospi flash support

bdunlay, 2022-11-16:
**Is your enhancement proposal related to a problem? Please describe.**

The B-L4S5I-IOT01A board has an onboard QSPI flash, but the flash memory is not exposed in the device tree. 

**Describe the solution you'd like**

I would like the ability to use the external flash memory with the B-L4S5I-IOT01A board. 

**Describe alternatives you've considered**

I tried adding support for ospi flash to the device tree from v3.2.0, but ran into driver compatibility issues with the stm32l4 ospi hal (see below).

The [Stm32CubeL4 HAL implementation](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/stm32cube/stm32l4xx/drivers/include/stm32l4xx_hal_ospi.h) does not appear to be compatible with the [OSPI flash driver](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v3.2.0/drivers/flash/flash_stm32_ospi.c). Curiously there does appear to be a compatible implementation of stm32L4xx_hal_ospi out there (for example the [implementation used by the LoRaMac-Node](https://github.com/zephyrproject-rtos/loramac-node/blob/zephyr/src/boards/mcu/stm32/STM32L4xx_HAL_Driver/Inc/stm32l4xx_hal_ospi.h)).

### Errors from build: 

```bash
/home/brian/zephyrproject/zephyr/drivers/flash/flash_stm32_ospi.c: In function 'flash_stm32_ospi_init':
/home/brian/zephyrproject/zephyr/drivers/flash/flash_stm32_ospi.c:1766:29: error: 'OSPI_InitTypeDef' has no member named 'WrapSize'
 1766 |         dev_data->hospi.Init.WrapSize = HAL_OSPI_WRAP_NOT_SUPPORTED;
      |                             ^
/home/brian/zephyrproject/zephyr/drivers/flash/flash_stm32_ospi.c:1766:41: error: 'HAL_OSPI_WRAP_NOT_SUPPORTED' undeclared (first use in this function); did you mean 'HAL_DMA_ERROR_NOT_SUPPORTED'?
 1766 |         dev_data->hospi.Init.WrapSize = HAL_OSPI_WRAP_NOT_SUPPORTED;
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                         HAL_DMA_ERROR_NOT_SUPPORTED
/home/brian/zephyrproject/zephyr/drivers/flash/flash_stm32_ospi.c:1766:41: note: each undeclared identifier is reported only once for each function it appears in
/home/brian/zephyrproject/zephyr/drivers/flash/flash_stm32_ospi.c:1778:29: error: 'OSPI_InitTypeDef' has no member named 'Refresh'
 1778 |         dev_data->hospi.Init.Refresh = 0;
      |                             ^
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/brian/firmware/build
```

### Changes to the device tree:

```
diff --git a/dts/arm/st/l4/stm32l4r5.dtsi b/dts/arm/st/l4/stm32l4r5.dtsi
index 6bd658232a..d85ac9b546 100644
--- a/dts/arm/st/l4/stm32l4r5.dtsi
+++ b/dts/arm/st/l4/stm32l4r5.dtsi
@@ -6,6 +6,7 @@
 
 #include <mem.h>
 #include <st/l4/stm32l4.dtsi>
+#include <zephyr/dt-bindings/flash_controller/ospi.h>
 
 / {
        sram0: memory@20000000 {
@@ -246,6 +247,17 @@
                        };
                };
 
+    // 0xA000 1400 - 0xA000 13FF
+               octospi2: octospi@a0001000 {
+                       compatible = "st,stm32-ospi";
+      reg = <0xA0001000 0x400>; 
+                       interrupts = <76 0>;  
+      clocks = <&rcc STM32_CLOCK_BUS_AHB3 0x00000100>; 
+                       #address-cells = <1>;
+                       #size-cells = <0>;
+                       status = "disabled";
+               };
+
                can1: can@40006400 {
                        compatible = "st,stm32-can";
                        reg = <0x40006400 0x400>;
```

```
diff --git a/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts b/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts
index ffe1b59029..e2aa37a02e 100644
--- a/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts
+++ b/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts
@@ -205,6 +205,43 @@
        };
 };
 
+&octospi2 {
+    pinctrl-0 = <&octospim_p1_clk_pe10 &octospim_p1_ncs_pe11
+                 &octospim_p1_io0_pe12 &octospim_p1_io1_pe13
+                 &octospim_p1_io2_pe14 &octospim_p1_io3_pe15
+                 &octospim_p1_io4_pd4 &octospim_p1_io5_pd5
+                 &octospim_p1_io6_pd6 &octospim_p1_io7_pd7>;
+
+
+    pinctrl-names = "default";
+
+    status = "okay";
+
+    mx25r6435f: ospi-nor-flash@0 {
+            compatible = "st,stm32-ospi-nor";
+            reg = <0>;
+            ospi-max-frequency = <DT_FREQ_M(50)>;
+            size = <DT_SIZE_M(64)>;
+            spi-bus-width = <OSPI_QUAD_MODE>; /* access on 4 data lines */
+            data-rate = <OSPI_STR_TRANSFER>; /* access in STR */
+            status = "okay";
+    };
+
+    partitions {
+                          compatible = "fixed-partitions";
+                          #address-cells = <1>;
+                          #size-cells = <1>;
+
+                          partition@0 {
+                              label = "nor";
+                              reg = <0x00000000 DT_SIZE_M(4)>;
+                          };
+               };
+
+};
+
 &rtc {
        status = "okay";
 };
```

**Additional context**

I modeled my changes after [this pull request](https://github.com/zephyrproject-rtos/zephyr/pull/39770), though the st,stm32-ospi-nor binding has been updated to allow for quadspi mode since.

FRASTM, 2022-11-16:
The [B-L4S5I-IOT01A Discovery kit](file:///local/home/frq09495/Downloads/um2708-discovery-kit-for-iot-node-multichannel-communication-with-stm32l4-series-stmicroelectronics.pdf) has a  64-Mbit Quad-SPI Flash memory which must be declared as `compatible = "st,stm32-qspi-nor"`

FRASTM, 2022-11-16:
unfortunately, the west build -p auto -b b_l4s5i_iot01a samples/drivers/spi_flash still fails: 
```
ospi-nor-flash@0 SPI flash testing
==========================

Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Data read does not match data written!!
000ff000 wrote 55 read ff MISMATCH
000ff001 wrote aa read ff MISMATCH
000ff002 wrote 66 read ff MISMATCH
000ff003 wrote 99 read ff MISMATCH
```

FRASTM, 2022-11-18:
@GeorgeCGV, could you please look at

GeorgeCGV, 2022-11-22:
@FRASTM , `MX25R6435F` supports `PP` (_`Page program, PP (0x02) up to 256 bytes`_) and `4PP` (_that is `0x38` - `PP_1_4_4`_) [_[REV. 0.03, JAN. 30, 2015, p. 16, p. 52](https://pdf1.alldatasheet.com/datasheet-pdf/view/690092/MCNIX/MX25R6435F.html)_].

The following configuration should work when quad mode is enabled (_QE bit is `1`_):
```
spi-bus-width = <OSPI_QUAD_MODE>;
data-rate = <OSPI_STR_TRANSFER>;
writeoc="PP_1_4_4";
```

I would check that QE bit is set (_quad mode is enabled_) in the status register and write protection.

Ideally, driver checks QE bit and attempts to set it. But that happens only for `OSPI_QUAD_MODE` `spi-bus-width`. OSPI debug log shall provide more information (_quad mode enabled or failed, what program/write & read opcodes are used_).

Unfortunately, I don't have any board with L4 on-board or `MX25R6435F` to test that.

FRASTM, 2022-11-23:
With PR https://github.com/zephyrproject-rtos/zephyr/pull/52374, the SPI or DUAL or QUAD configurations are now all working, with DMA transfer or not

closed 2022-11-29 by fabiobaltieri