zephyrproject-rtos/zephyr #52298: CI fail multiple times due to download package from http://azure.archive.ubuntu.com/ubuntu failed.

RafaelLeeImg, 2022-11-16:
**Describe the bug**
CI fail due to unable to download packages from http://azure.archive.ubuntu.com/ubuntu

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 CI
 - What have you tried to diagnose or workaround this issue?
 Push again
 - Is this a regression?
 I don't know,

**To Reproduce**
Steps to reproduce the behavior:
1. creage a pull-request and then wait
2. Until you see "Unable to connect to azure.archive.ubuntu.com:http:"
**Expected behavior**
Pass CI

**Impact**
CI needs to run multiple times.

**Logs and console output**
[https://github.com/zephyrproject-rtos/zephyr/actions/runs/3473314153/jobs/5811634827](https://github.com/zephyrproject-rtos/zephyr/actions/runs/3473314153/jobs/5811634827)

```log
After this operation, 147 MB of additional disk space will be used.
Err:1 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libann0 amd64 1.1.2+doc-7build1
  Could not connect to azure.archive.ubuntu.com:80 (52.252.75.106), connection timed out
Err:2 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libcdt5 amd64 2.42.2-3build2
  Unable to connect to azure.archive.ubuntu.com:http:
Err:3 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libcgraph6 amd64 2.42.2-3build2
  Unable to connect to azure.archive.ubuntu.com:http:
Err:4 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libgts-0.7-5 amd64 0.7.6+darcs121130-4
  Unable to connect to azure.archive.ubuntu.com:http:
Err:5 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libpathplan4 amd64 2.42.2-3build2
  Unable to connect to azure.archive.ubuntu.com:http:
Err:6 http://azure.archive.ubuntu.com/ubuntu focal/universe amd64 libgvc6 amd64 2.42.2-3build2
  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/a/ann/libann0_1.1.2+doc-7build1_amd64.deb  Could not connect to azure.archive.ubuntu.com:80 (52.252.75.106), connection timed out
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/graphviz/libcdt5_2.42.2-3build2_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/graphviz/libcgraph6_2.42.2-3build2_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/gts/libgts-0.7-5_0.7.6+darcs121130-4_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/graphviz/libpathplan4_2.42.2-3build2_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/graphviz/libgvc6_2.42.2-3build2_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/graphviz/libgvpr2_2.42.2-3build2_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/universe/g/graphviz/liblab-gamut1_2.42.2-3build2_amd64.deb  Unable to connect to azure.archive.ubuntu.com:http:

```

**Environment (please complete the following information):**
 - OS: CI server
 - Toolchain: Not related
 - Commit: All commits are affected.

**Additional context**
Use some other package source?

closed 2022-11-16 by stephanosio

RafaelLeeImg, 2022-11-16:
The meaning of CI is reduce human interactions.
And I didn't find good way to trigger CI. I don't think create a new commit and force push is an good idea.

stephanosio, 2022-11-16:
> And I didn't find good way to trigger CI. I don't think create a new commit and force push is an good idea.

[image]

Not sure if you have access level required to do that though; if not, waiting a few hours and force pushing can be an option; alternatively, you can ping in the #ci channel on Discord (this would be the easiest way to understand what is going on and get the issue fixed).