zephyrproject-rtos/zephyr #52318: mgmt: mcumgr: Add settings commands

nordicjm, 2022-11-17:
**Is your feature request related to a problem? Please describe.**
A requested feature for zephyr-based devices is being able to write/read long data over BLE to/from a device in a simple to use way. MCUmgr allows for such a system as the protocol is simple and supports data which is larger than the BLE MTU (512 bytes), so would be a good fit for this. Therefore having an MCUmgr group for reading from/writing to settings through the settings API would be a useful feature for this.

**Describe the solution you'd like**
A simple MCUmgr group (using the existing group ID 3 - configuration) with commands for reading data from settings and writing data to settings, and optionally a way to iterate through all settings. This system would not have security for an initial release but may have security implemented in a future release e.g. to restrict access to some settings.

Laczen, 2022-11-22:
@nordicjm can't this feature be provided trough a shell command interface for mcumgr ? This would avoid doubling maintenance for a shell interface and a mcumgr interface.

nordicjm, 2022-11-22:
> @nordicjm can't this feature be provided trough a shell command interface for mcumgr ? This would avoid doubling maintenance for a shell interface and a mcumgr interface.

Then you would need to pull the whole shell system in which would add a huge bloat, and would be a big security vulnerability. Unless NVS has an API change made to it, nothing needs to be done with this.

Laczen, 2022-11-22:
@nordicjm, opening the nvs system to mcumgr would be a big security vulnerability also. If there would be a change to the nvs API the work would be doubled (for the shell and for mcumgr). As the author of nvs I also consider nvs to be eol.

nordicjm, 2022-11-22:
> @nordicjm, opening the nvs system to mcumgr would be a big security vulnerability also.

Not so, with recent revamped MCUmgr callback system, the plan would be to have an (optional) callback to check if access should be granted to particular keys - similar to how the fs_mgmt one works - see https://docs.zephyrproject.org/latest/services/device_mgmt/mcumgr_callbacks.html#structfs__mgmt__file__access

> As the author of nvs I also consider nvs to be eol.

What's the replacement?

PR #59345 (2023-08-25): mgmt: mcumgr: Add settings management
    Adds a settings management group to MCUmgr which allows for
    manipulation of the zephyr settings from a remote device.
    Includes callback hooks to secure access from an application.

- [x] Code
- [x] MCUmgr documentation update
- [x] Release notes update
- [x] Tests
- [x] Decide on group ID (either low - possible upstream MCUmgr group conflict, or high - zephyr custom group)
- [x] Doc update for load
- [x] Doc update for callback

Fixes #52318
closed 2023-08-25 by fabiobaltieri