zephyrproject-rtos/zephyr #52327: websocket: websocket_recv_msg breaks when there is no data after header

SavitskiDenis, 2022-11-17:
If there is no data after the header and the message length is 0, the next packet will be incorrectly analyzed, because payload data will be expected instead of the header. I tested it on zephyr 2.7.2, nothing has changed in version 3.2.99.
https://github.com/zephyrproject-rtos/zephyr/blob/56664826b27c1b716483fbad35961450794c3089/subsys/net/lib/websocket/websocket.c#L847
I catch this error when I receive a ping message without payload data (for ping messages, payload data is not required [[RFC6455](https://datatracker.ietf.org/doc/html/rfc6455#section-5.5.2)]).
I fixed it this way:
```c
if (ctx->tmp_buf_pos == 0) {
	/* No data after the header, let the caller call
		* this function again to get the payload.
		*/
	if (ctx->message_len == 0) {
		ctx->header_received = false;
		ctx->message_len = 0;
		ctx->message_type = 0;
		ctx->total_read = 0;
	}
	return -EAGAIN;
}
```

jukkar, 2022-11-18:
@SavitskiDenis could you propose a PR that has your fix

SavitskiDenis, 2022-11-18:
@jukkar My solution is duct tape, because in this case the function returns -EAGAIN. In a good way, you need to return 0 or another errno. if you select the option when the function returns 0 in this case, you need to change the result of the "socket closed" return.

GrixaYrev, 2022-11-21:
@jukkar, websocket frame can have no payload, for example ping/pong. So it is good return 0 when no payload, rather than socket closed. For socket closed state i suggest to use errno `-ENOTCONN`. If You and other community competent members agree that variant, i and @SavitskiDenis can propose a PR.

jukkar, 2022-11-21:
@GrixaYrev Your proposal sounds ok, so please prepare a PR for it if you can.

carlescufi, 2022-11-30:
@GrixaYrev will you post a Pull Request to fix this? Thanks!

GrixaYrev, 2022-11-30:
@carlescufi, i plan to post PR at end of current week.

rlubos, 2022-12-01:
Funnily, I've stumbled on this problem when investigating another issue.

Generally, I think what you propose make sense, however I have some caveat for the code snippet in the orignal post:
```
if (ctx->tmp_buf_pos == 0) {
	/* No data after the header, let the caller call
		* this function again to get the payload.
		*/
	if (ctx->message_len == 0) {
		ctx->header_received = false;
		ctx->message_len = 0;
		ctx->message_type = 0;
		ctx->total_read = 0;
	}
	return -EAGAIN;
}
```
I don't think the `message_len` check should be placed inside `if (ctx->tmp_buf_pos == 0) {` - the `temp_buf_pos` variable, indicating the length of data read from the socket may still be > 0, even if the `message_len` reported in the header was 0 - simply because during the socket `recv()` we could've already obtained part of the next header. Therefore I'd separate the two checks (`message_len` vs `tmp_buf_pos`) from each other.

GrixaYrev, 2022-12-01:
@rlubos, we (i and @SavitskiDenis) know this websocket have many bugs, not just empy-payload case. But other bugs are not critical. So we suggest hotfix.

It is good rewrite this library with clear FSM in future. May be we will make, but later.

Similar mistakes happens often, when TCP use like packet stream rather than byte stream. I think learning examples are cause.

GrixaYrev, 2022-12-05:
I can't fix websocket as soon as i wanted. And i think i should fix in full. I want to implement the FSM in `websocket_recv_msg()`. When jump to state `Opcode` is, the fuction exits.

<details>
<summary>FSM and algorithm</summary>

[image]

</details>

Please, give your comments.

rlubos, 2022-12-05:
> When jump to state Opcode is, the fuction exits.

Not sure I fully follow, what about the payload? Who's responsible for reading it from a socket?

Anyway, when I looked at the WebSocket code initially, I had a similar impression that some kind of parse-in-a-loop approach could be used, instead of constantly returning -EAGAIN if the data is incomplete. Please note, however, that the function accepts the `timeout` parameter which should be respected, i. e. the function should not block if the timeout is 0 for example.

GrixaYrev, 2022-12-05:
@rlubos, yes, i forgot to show function's exit when rx buffer is full. Other aspects of payload processing is same as in old version.

rlubos, 2022-12-13:
@GrixaYrev Any updates on the `websocket_recv_msg()` rework? It turns out that we have another WebSocket-related issue (https://github.com/zephyrproject-rtos/zephyr/issues/52632) which would benefit from an updated logic of `websocket_recv_msg()`.

GrixaYrev, 2022-12-13:
@rlubos I read this topic, and i see many people have problems with websocket. Sorry for my slowness. I start setup native-posix on wsl2, and i will strive to release the fix as soon as i can. 

I think my proposal can fix https://github.com/zephyrproject-rtos/zephyr/issues/52632#issuecomment-1347436171 at least.

GrixaYrev, 2022-12-13:
@rlubos, @jumo-sballing, @svenstorp
I have written new receive algorithm. At first i has updated tests (add receiving empty ping and fix receivng two messages), so that this two test become failure. With new algorithm all tests pass. Also i have tested with real websocket-server (simple python script) using `zephyr\samples\net\sockets\websocket_client` with little changes (add ping waiting before sending).

You can see my updates in [my repo](https://github.com/GrixaYrev/zephyr/tree/pr_websocket).
I does not formalize commit message yet. But you can test already. Note,  `websocket_recv_msg()` can return 0 now, and it is no error, it is empty payload. `-EAGAIN` will be returned when timeout occured, `-ENOTCONN` – when socket close.

<details>
<summary>New FSM and flowchart.</summary>

[image]

</details>

PR #53070 (2022-12-20): net: websocket: new receiving algorithm
`websocket_recv_msg()` is reworked with using fsm. Now the function return 0 when payload is empty, -ENOTCONN if socket close. Receiving empty ping and sending empty pong were added in tests.
Fixes #52327
commit 2a992c65 (2022-12-20): net: websocket: new receiving algorithm

websocket_recv_msg() is reworked with using fsm. Now the function
return 0 when payload is empty, -ENOTCONN if socket close. Receiving
empty ping and sending empty pong were added in tests.
Fixes #52327
closed 2022-12-20 by fabiobaltieri