zephyrproject-rtos/zephyr #52328: net: lwm2m: lwm2m_engine_send improvement (behaviour, callbacks, documentation)

broglep-work, 2022-11-17:
# Problem
The method `lwm2m_engine_send` has some behaviours, that at least to me, are somewhat unexpected:

1) If `confirmation_request` is enabled, a message timeout will trigger a lwm2m client re-registration 
2) Completion of `lwm2m_engine_send` only means that the message has been enqueued for transmission

Behaviour 1) could be considered an unwanted side effect of lwm2m_engine_send, depending on the usage. A timeout of single message could trigger a (potentially expensive) re-registration in case of just temporary transmissions problems 

Behaviour 2) and the lack of any possibility to pass in callbacks to the send method, poses imho the biggest shortcoming at the moment. The caller has no way to know if a send succeeded, not even if it was sent out onto the wire. At least that guarantee I'd expect from `lwm2m_engine_send`, in particular when `confirmation_request` is enabled. Less so if is it disabled, then I'd be aware that it has rather fire & forget semantics, but even in that case, I would imply that guarantee, as the method based on its signature looks like a blocking method

# Possible Solution
Keep current behaviour (no guarantees), document it accordingly and provide callbacks for: 
1) message has been sent on the wire
2) and if confirmable: timeout and reply callbacks

This would allow for most flexibility and allows caller define the behaviour and how he wants to deal with error conditions. A default callback for timeout callback that attempts an re-registration could be ok & good connivence for caller

# Alternatives
Introduce explicit `lwm2m_engine_send_sync`(or just `lwm2m_engine_send`) and `lwm2m_engine_send_async`. And internally for the send variant use `lwm2m_send_message` that guarantees that the message has been sent.

# Opinions / Suggestions? 
What do you think? Let's discuss

RomainPelletant, 2022-11-23:
I am currently facing the problem.
I would add a callback lwm2m_engine_send to get the simple result true or false (instead of only displaying "Send done!")
Is the lwm2m client re-registration mandantory is case of timeout? I mean, from user point of view, timeout is send failed only, isn't it?

SeppoTakalo, 2022-11-28:
I think the behaviour 1 is correct.
If we lose confirmable CoAP messages, it means that we have lost a connection to the server. This might be due to NAT table have expired, for example.
And because the connection is lost, only way to proceed is to try to re-establish the connection. Re-registration causes a DTLS handshake as well, and should be able to fix the issue.

Another thing is that LwM2M spec specifically denies us to use Non-Confirmable CoAP messages for LwM2M SEND.
https://www.openmobilealliance.org/release/LightweightM2M/V1_1_1-20190617-A/HTML-Version/OMA-TS-LightweightM2M_Transport-V1_1_1-20190617-A.html#6-4-0-64-URI-Identifier-and-Operation-Mapping

Therefore the send API should always use confirmable=true.

What comes to callbacks, I would like someone to create a new API
```c
int lwm2m_engine_send_cb(struct lwm2m_ctx *ctx, char const *path_list[], uint8_t path_list_size,
		     coap_reply_t *reply_cb);
```

And then the old API would just call
```c
__deprecated
static inline int lwm2m_engine_send(struct lwm2m_ctx *ctx, char const *path_list[], uint8_t path_list_size,
		      bool confirmation_request);
{
        if (!confirmation_request) {
                return -EINVAL;
        }
        return lwm2m_engine_send_cb(ctx, path_list, path_list_size, NULL);
}
```

Old API could be marked as deprecated.

Then application should be able to know when the request have been delivered. But if we instead receive event for registration update, we know that the packet have not been delivered.

broglep-work, 2022-12-21:
I think the behaviour 1 can be a sensible default if the whole lwm2m engine treats all confirmable message timeouts like this. Currently only lwm2m send and observe notify do have this behaviour (and with observe only if there is still an active observation, otherwise the timeout is ignored). 
The engine could be extended to make it more flexible what should happen on a engine timeout (e.g. only attempt re-registration if there are multiple timeouts). but that is probably another ticket. 

> Then application should be able to know when the request have been delivered. But if we instead receive event for registration update, we know that the packet have not been delivered.

Using this approach will be difficult for the caller. What the caller is interested in is if the message was delivered or not. He doesn't really wan't to need to know about the registration updates to figure out if the message was not delivered. Its might be even hard to correlate a registration update event to the message sending and figure out that one particular re-registration means that the previous send was not delivered

RomainPelletant, 2023-05-04:
@broglep-work @carlescufi 
Related PR has been merged if it successfully solved this issue

SeppoTakalo, 2023-05-15:
This issue can be closed.
PR https://github.com/zephyrproject-rtos/zephyr/pull/53235 is merged.

This means that for question 1: we have already documented the error handling in the state machine description https://docs.zephyrproject.org/latest/connectivity/networking/api/lwm2m.html#lwm2m-engine-and-application-events
For question 2: The fix was in the PR.

closed 2023-05-15 by SeppoTakalo