zephyrproject-rtos/zephyr #52339: usb: USB device re-enabling does not work on Nordic devices (regression)

matvey-mukha, 2022-11-17:
**Describe the bug**

If an application running on one of Nordic Semiconductor chips performs the following sequence:

* Enable the USB stack with a call to usb_enable()
* Disable the USB stack with a call to usb_disable()
* Re-enable the USB stack with a call to usb_enable()

the second call to usb_enable() would fail with an -EAGAIN error code.

Failure happens when the stack enables the control endpoint in usb_enable(), calling the usb_dc_ep_enable() function.
usb_dc_ep_enable() returns -EAGAIN as the control endpoint is not disabled in usb_disable().

This is a regression introduced by the commit from the following issue: https://github.com/zephyrproject-rtos/zephyr/pull/50475

**To Reproduce**

Build an application that does the call sequence mentioned above.

**Expected behavior**

The second call to usb_enable() returns 0.

desowin, 2022-11-18:
Thanks for the excellent bug report and pinpointing the actual faulty commit.

desowin, 2022-11-18:
It is worth mentioning that the commit that introduced the regression, essentially uncovered USB stack issue that has always been there. That is, the `usb_disable()` only calls `usb_dc_detach()` and does not call any `usb_dc_ep_disable()` on any endpoint. The reason why `usb_enable()`/`usb_disable()`/`usb_enable()` worked before seems incorrect.

Zephyr USB device controller drivers fall into two category:
- drivers that allow re-enabling already enabled endpoints (dw, native posix, rpi pico, sam usbc, sam usbhs, sam0, stm32)
- drivers that return error when enabling already enabled endpoints (kinetis, mcux, nrfx)

Before the regression nrfx was in a third category: drivers that lose information about already enabled endpoints on detach and therefore allow re-enabling already enabled endpoints after detach.

PR #52403 (2022-11-25): usb: device: Disable all endpoints on stack disable
Disable all enabled endpoints on stack disable to allow re-enabling USB device stack.

drivers: usb_dc_nrfx: Always allow endpoint disable. There is no point in allowing endpoint disable only when device is attached and ready. Remove the pointless check as it is actually harmful and prevents endpoints disable on USB stack disable.

Fixes: #52339

Cannot be merged before #51871
closed 2022-11-25 by carlescufi