zephyrproject-rtos/zephyr #52347: LE Audio: Use the BASE example from the BAP spec for the broadcast samples

Thalley, 2022-11-17:
**Is your enhancement proposal related to a problem? Please describe.**
The current broadcast samples does an OK job of showing significant amount of how the API can and should be used. However, they are not setting any metadata or creating multiple subgroups. 

My suggestion is to use the BASE example from the BAP spec as the Zephyr sample: 
[image]

This would showcase more of the API, and provide a more "real" scenario, as well as something that a developer can refer to from the spec. 

**Describe the solution you'd like**
Modify the Broadcast Audio Source and Broadcast Audio Sink sample to support the BASE example from BAP. 

**Describe alternatives you've considered**
We could/should consider offering support for mono, so that the number of BIS for each subgroup is reduced to 1 (2 total) instead of 2 (4 total), to support more controllers. 

**Additional context**
N/A

Thalley, 2022-11-17:
Ping @larsgk

Thalley, 2024-01-05:
Just FYI, the BASE from BAP 1.0.1 is not something that actual works: 4 streams with 48 kHz is simply not possible, and IIRC the BASE example will be updated to reflect that in a later BAP version.

kruithofa, 2024-01-30:
See https://bluetooth.atlassian.net/browse/ES-23582