zephyrproject-rtos/zephyr #52355: RFC: API change: Add KSCAN matrix API

ChiHuaL, 2022-11-18:
## Introduction
Add KSCAN APIs to support the keyboard matrix scanning mechanism.

### Problem description
Currently, all embedded controllers (EC) vendors (Nuvoton, ITE, and Microchip) put the keyboard matrix scanning algorithms within their own kscan drivers. It might not be a good choice since:

1. Scanning algorithm is not relevant to hardware. From a low-level driver's point of view, extracting it from the driver is reasonable. The other keyboard matrix vendors needn't re-invent the wheel again.
2. No solution if users have an alternative algorithm for better ghost key detection or specific circuit requirement. (Ex, the polarity of each matrix column might not be the same.)

### Proposed change
Hence, we'd like to add APIs to be an interface between the kscan driver and the scanning algorithms. The users can still use the internal matrix scanning algorithm internally by default. If not, these APIs can be used by the customized algorithm to support keyboard matrix scanning mechanism.

## Detailed RFC
Generally, a simple 4x4 keyboard matrix (that is, without diodes) will be wired up like this:
[image]
In this case, we assume all row and column pads are configured to open-drain and all row pads have PU internally.
And the attached picture indicates a general keyboard matrix scanning algorithm flow. We can split it into three stages.
[image]

- Stage 1: Key pressing detection
Driving all columns to low. Once a key is pressed, an edge signal on row pads can be detected. Then, go to stage 2.
- Stage 2: Read keyboard matrix
Disable interrupt of rows first. Then, driving columns of matrix one by one to read each rows' states sequentially.
- Stage 3: analyze keyboard matrix states
Analyze keyboard states. Once all keys are released, enable interrupt of rows and go back to stage 2.
During the whole keyboard matrix scanning process, the driver needs to provide the following interfaces:

1. Configure the ISR callback of for the input (KSI) detection ISR
2. Driving columns of matrix
3. Read each row's states
4. Enable/Disable interrupt of row's pad

### Proposed change (Detailed)
Please see the proposed PR in #42923

### Dependencies
This approach requires keyboard matrix hardware (GPIOs or specific module) support.

### Concerns and Unresolved Questions
None so far.

## Alternatives
None so far.

ChiHuaL, 2022-11-22:
@MulinChao

keith-zephyr, 2022-11-23:
Some additional context, PR https://github.com/zephyrproject-rtos/zephyr/pull/42923, proposes creating a `kscan_matrix` driver that connects to the existing KSCAN API.  The matrix style SoC drivers then connect to the KSCAN matrix API.  The KSCAN user API is unchanged by this proposal.

Logical organization
```
KSCAN API
  KSCAN matrix pseudo driver
    NPCX SoC matrix driver
    ITE SoC matrix driver
```

The KSCAN matrix pseudo driver doesn't require any devicetree bindings, only the SoC matrix drivers need a devicetree binding.  The SoC matrix drivers should do a `select KSCAN_MATRIX` to automatically enable the KSCAN matrix pseudo driver.

keith-zephyr, 2022-11-23:
@gmarull - I'd appreciate your review of this.

galak, 2022-12-15:
DevReview (Dec 15th, 2022): @mbolivar-nordic suggested to reach out to the ZMK maintainer (@petejohanson)

@galak suggestion to have library functions the lives in drivers/kscan similar to how i2c_bitbang works https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_bitbang.c.

closed 2023-12-06 by fabiobaltieri