zephyrproject-rtos/zephyr #52360: samples/net/socket/socketpair does not run as expected

rymaj, 2022-11-18:
**Describe the bug**
I tried to run samples/net/socket/socketpair, and got the folllowing issue first, 

```
*** Booting Zephyr OS build zephyr-v3.2.0-585-ge80a0c35e666  ***
Alpha: socketpair: 1 <=> 0
Bravo: socketpair: 3 <=> 2
socketpair failed: 23
Alpha closed fd 0
Bravo closed fd 2

```
Then I figure out I need to increase the default CONFIG_POSIX_MAX_FDS from 4 to 8 for 3 socket pairs (should this change be  covered in socketpair/prj.conf ?)

Even I add this setting, I saw this issue, seems the poll mechanism was broken?

```
*** Booting Zephyr OS build zephyr-v3.2.0-585-ge80a0c35e666  ***
Alpha: socketpair: 1 <=> 0
Bravo: socketpair: 3 <=> 2
Charlie: socketpair: 5 <=> 4
fd: 1: read 5 bytes
fd: 1: read 5 bytes
fd: 1: read 5 bytes
fd: 1: read 0 bytes
fd: 1: read 0 bytes
fd: 1: read 0 bytes
<keep looping>
fd: 1: read 0 bytes
fd: 1: read 0 bytes
fd: 1: read 0 bytes
fd: 1: read 0 bytes

```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?   
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

cfriedt, 2022-11-18:
@rymaj - can you please update the upstream commit in the bug report?

cfriedt, 2022-12-02:
It's kind of a PITA to have to jump through all of the hoops and enable a network device just to use socketpair, which really shouldn't require any kind of network interface at all.

Even just trying to run the sample application, I get an error:
```
west build -p auto -b mps2_an385 -t run samples/net/sockets/socketpair
...
qemu-system-arm: -serial unix:/tmp/slip.sock: Failed to connect to '/tmp/slip.sock': No such file or directory
qemu-system-arm: -serial unix:/tmp/slip.sock: could not connect serial device to character backend 'unix:/tmp/slip.sock'
FAILED: zephyr/CMakeFiles/run_qemu /Users/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/CMakeFiles/run_qemu 
cd /Users/cfriedt/workspace/zephyrproject/zephyr/build && /opt/homebrew/bin/qemu-system-arm -cpu cortex-m3 -machine mps2-an385 -nographic -vgf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake --build /Users/cfriedt/workspace/zephyrproject/zephyr/build --target run
```

cfriedt, 2023-02-02:
Actually, it seems at some point, `zephyr_app_main` was dropped and we introduced a weak `main`, but this file was never updated. I'll address that in the same PR as this fix.

cfriedt, 2023-02-04:
The output below:
```
fd: 1: read 0 bytes
fd: 1: read 0 bytes
fd: 1: read 0 bytes
fd: 1: read 0 bytes
```

Indicates to me that the socket is not operating in blocking mode. However, it was opened in blocking mode (by default), so this is potentially a bug in the sockets implementation.

I'm still working through it though, so will report more information as I find it.

PR #54458 (2023-02-08): socketpair improvements
4 commits:
* improvements for socketpair sample app (it was broken for a while)
* rename `socketpair_example.c` to `main.c`
* minor improvements for `socketpair.c`
* additional `<fcntl.h>` corrections in the networking subsystem

Fixes #54347
Fixes #52360
closed 2023-02-08 by stephanosio