zephyrproject-rtos/zephyr #52362: nrf_qspi_nor driver crash if power management is enabled

onkila, 2022-11-18:
when CONFIG_PM_DEVICE is enabled and system enters suspend pm_system_suspend() is called from idle-thread. This calls also drivers power management function and that tries to get the semaphore. Unfortunately that causes an assert because it seems that we can not wait to get the semaphore if current thead is idle thread. There is an assert for this in sched.c line 186:
__ASSERT_NO_MSG(!z_is_idle_thread_object(thread));

Ozone callstack is following:
Level,Function,Stack Frame,Source,PC,Return Address,Stack Used
0,z_priq_dumb_add, 0 @ 0x20029790,sched.c:186:2,0x00058562,: 0x000585A0,232
0,add_to_waitq_locked,16 @ 0x20029790,sched.c:719:3,0x0005852C,[0x2002979C]: 0x00058600,232
0,pend,32 @ 0x200297A0,sched.c:739:3,0x000585FC,[0x200297BC]: 0x0005866C,216
0,z_pend_curr,16 @ 0x200297C0,sched.c:814:2,0x00058668,[0x200297CC]: 0x00056CB0,184
0,z_impl_k_sem_take,32 @ 0x200297D0,sem.c:143:2,0x00056CAC,[0x200297EC]: 0x00042D22,168
0,qspi_lock, 0 @ 0x200297F0,kernel.h:1035:2,0x00042D1E,: 0x00042D22,136
0,qspi_device_init,16 @ 0x200297F0,nrf_qspi_nor.c:356:2,0x00042D10,[0x200297FC]: 0x00042E00,136
0,qspi_nor_pm_action,32 @ 0x20029800,nrf_qspi_nor.c:1315:3,0x00042DFC,[0x2002981C]: 0x00075B72,120
0,pm_device_action_run,24 @ 0x20029820,/zephyr/subsys/pm/device.c:80:2,0x00075B70,[0x20029834]: 0x0003019E, 88
0,pm_suspend_devices,32 @ 0x20029838,pm.c:75:3,0x0003019A,[0x20029854]: 0x00030408, 64
0,pm_system_suspend,16 @ 0x20029858,pm.c:238:3,0x00030404,[0x20029864]: 0x0005637E, 32
0,idle, 8 @ 0x20029868,idle.c,0x0005637A,[0x2002986C]: 0x00073C30, 16
0,z_thread_entry, 8 @ 0x20029870,thread_entry.c:36:2,0x00073C2E,[0x20029874]: 0xFFFFFFFF, 8
0,Top of stack - return address out of range: 0xFFFFFFFF,,,,,

onkila, 2022-11-18:
Solution would be to prevent nor flash driver to go to suspend state when it is busy. This happens easily by calling pm_device_busy_set()-funcion

carlescufi, 2022-12-21:
Downgrading to Low since this is restricted to a very specific set of circumstances, and is in fact a general problem.

PR #57193 (2023-04-25): drivers: flash: nrf_qspi_nor: Mark device as busy when locked
Mark the device as busy when it is locked to perform a flash operation so that the PM subsystem does not try to suspend it. Also refuse to run a PM action when the device is busy (in case `qspi_nor_pm_action()` is called directly, not by the PM subsystem).

Fixes #52362.
closed 2023-04-25 by MaureenHelm