zephyrproject-rtos/zephyr #52393: Central misses notifications from connected peripherals after increasing MTU

ahmedmoheb-nordic, 2022-11-20:
**Describe the bug**
When central is connected to multiple peripherals and the MTU is increased to 247 bytes, central gets notifications only from the last connected peripheral

**To Reproduce**
Steps to reproduce the behavior:
1. Central connects to more than one peripheral
2. For each connection, increase the MTU
3. Send notifications in both ways between the central and connected peripherals

**Expected behavior**
1. Central receives notifications from the last connected peripheral
2. Peripherals that aren't communicating disconnect with time out disconnection reason

**Environment**
 - OS: Ubuntu 22.04
 - Toolchain : Zephyr SDK 0.15.1
 - Commit SHA 3607a05fbaf1d85a1e11d19c6ff33b869c5e88bb
 - PR https://github.com/zephyrproject-rtos/zephyr/pull/52392

jori-nordic, 2022-11-25:
I think it's a controller issue, the test works fine (there were some logic errors) when peripherals are compiled with `CONFIG_BT_AUTO_DATA_LEN_UPDATE=n`.

It seems the central controller has issues ACK-ing one of the peripheral after switching to the max LL packet length. When the LL data length stays at 27, there are no issues whatsoever. 

@kruithofa and @cvinayak To reproduce:
- check out this branch: https://github.com/jori-nordic/zephyr/tree/long-packets-no-ack
- run the build&test script `tests/bluetooth/bsim_bt/bsim_test_conn_stress/tests_scripts/conn_stress.sh`

ahmedmoheb-nordic, 2022-11-25:
I've tested the same scenario using the desktop application "nrf connect for desktop Bluetooth Low Energy" in place of the central device while keeping the same peripherals and following the same behavior of the central application and there were no issues found.

kruithofa, 2022-12-16:
might be fixed by PR #52012 and #53095, I will reproduce the issue first, and see if these PRs fix it

kruithofa, 2022-12-19:
@ahmedmoheb-nordic @jori-nordic PR #52012 fixes this issue, but requires that the event-length update feature is enabled by adding following 2 lines to the prj.conf files:
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_CTLR_CONN_EVENT_LEN_UPDATE=y
@ahmedmoheb-nordic can you confirm?

PR #52012 (2023-01-19): Bluetooth: controller: fix post DLE/PHY update event length
Fix the controller implementation to perform connection event length reservation based on the completed Data Length Update and/or PHY Update Procedure.

Note: some optimisation is possible with regards to the length allocated, currently the maximum rx and tx time are used, which leaves a lot of unused airtime in case that the 2M phy is used. If necessary this will be fixed in a separate PR
fixes #21995
fixes #52393
closed 2023-01-19 by carlescufi