zephyrproject-rtos/zephyr #52395: Cannot build applications with dts having (unused) external flash partition and disabling those drivers

thedjnK, 2022-11-20:
**Describe the bug**
Seeing an issue with building mcuboot in zephyr whereby there is an external flash on QSPI which has a partition, `lfs_partition`, this partition is not used by mcuboot but the application fails to link unless the QSPI driver is enabled with an `undefined reference to '__device_dts_ord_133'` error, which is in `default_flash_map`. This leads to a complete waste of flash space by requiring completely unused flash drivers be enabled in builds.
If the `lfs_parition` part of the dts is commented out, the application builds without failing to link.

**To Reproduce**
Build mcuboot (or any application with a flash driver) which has an unused flash partition in an external device and build it with the flash driver for that partition disabled.

**Expected behavior**
Application to build without needing unused flash partition drivers

**Impact**
Annoyance, waste of resources

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.2
 - Commit SHA or Version used: 0baf45f00af7296caa089e02e75193ae4c2ae08c

Laczen, 2022-11-21:
@thedjnK, it is possible to remove the driver while building for mcuboot by supplying a dts overlay file that deletes the lfs-partition node.

thedjnK, 2022-11-27:
> @thedjnK, it is possible to remove the driver while building for mcuboot by supplying a dts overlay file that deletes the lfs-partition node.

I did try this for a while and was not able to get it working, just tried it again and it is working so must have made a mistake previously.

nashif, 2023-05-30:
@de-nordic  can you please take another look?

de-nordic, 2023-05-31:
> @de-nordic can you please take another look?

@nashif  The issue is there. It appeared when we have switched to device macros.
The Flash Map creates entry for every partition that is visible in DTS and attaches device to it, so it requires device driver to be compiled in. At this point Zephyr does not provide a method to match driver enabling Kconfig to DT node, which means that we have no way to just not include references to devices that will not be there, because their drivers will not be linked.
I am currently working, again, at dropping the Flash Map in current form, but the work requires significant modification on the side of MCUboot and Zephyr and takes time.

closed 2025-01-09 by github-actions[bot]