zephyrproject-rtos/zephyr #52409: STM32H7: ethernet: Device stops receiving IP over ethernet completely after it has been receiving UDP/IP multicast at a constant rate

cperera-aud, 2022-11-21:
**Describe the bug**
I've been able to reproduce this issue on the Nucleo-H753ZI (STM32H7) evaluation board. I have created a very basic example which exhibits the issue. The example application uses ethernet networking. It starts up DHCP and does nothing afterwards. However, the device is on a ethernet network that has a constant rate of 10Mbps IPv4 UDP multicast traffic e.g. audio over IP. The example does not open any multicast sockets to receive data (I can modify it if requested). The ethernet MAC receives the multicast data and forwards it to the network stack. There is no IGMP snooping on the network. Also the eth_stm32_hal driver doesn't enable multicast hash filtering. Eventually after the device has been running for about 20 minutes or so the device completely stops receiving IPv4 over ethernet packets. Please note that IPv4 was specifically mentioned since this is what was tested. I have not tested IPv6 since we do not have a usecase for it. This was confirmed by checking the output from net stats shell command. It was also further confirmed by checking the behaviour of the various system threads using Segger SystemView. 

I have used Segger SystemView to get some insight into the issue and eventually I can see that ethernet DMA receive interrupts stop firing. The stm_eth thread takes a semaphore (`dev_data->rx_int_sem`) but it times out after 500ms because the DMA receive interrupt is not signaling the semaphore. 

On the actual software that I'm working on, it opens up a multicast socket and receives the multicast data and eventually runs into the same issue of not receiving ethernet packets. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
	**Nucleo-H753ZI board**
 - What have you tried to diagnose or workaround this issue?
	**I have been using Segger SystemView to help diagnose what is going on. It is from SystemView I released that the ethernet receive DMA interrupts are not firing after sometime after receiving UDP/IP multicast data. The workaround is to use unicast but our product needs to support both unicast and multicast**
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
	**Not that I'm aware of, I'm somewhat new to Zephyr and I've been using v3.1 and the issue is present in v3.2 too.**

**To Reproduce**
Steps to reproduce the behavior:
1. Unzip the attached archive (hello_wold.zip) and put it in the zephyr/samples directory
2. cd zephyr/samples/hello_world
3. west build -p auto -b nucleo_h753zi -- -DDTC_OVERLAY_FILE="dts/test.overlay" .
4. west flash
5. Run the device on a ethernet network that has multicast traffic with a constant rate of 10Mbps (I used our own equipment to generate the IPv4 UDP/IP multicast traffic, sorry I'm not aware if there are any off the shelf software/HW that can do the same)
6. Run net stats shell command to see if the Bytes received or IPv4 recv is going up
7. Eventually after about 20mins or even less the Bytes received or IPv4 recv counters from net stats will not be going up anymore. Please note that at times it might take longer to reproduce the issue
8. ping to the device will fail

**Expected behavior**
The device should continue to receive IP data over ethernet, i.e. the Bytes received or IPv4 recv counters in the net stats command should continue to increase. 

**Impact**
We are not able to ship a product with this type of failure. The device is not able to operate normally and receive multicast IP data over ethernet. 

**Logs and console output**
A log from Segger SystemView is attached. Apologies for the overflows shown the SystemView logs. The SystemView logs indicate that at 11:46.693 567 127 (75392682) the device received its last ethernet receive DMA interrupt. Since then the device has stopped servicing ethernet receive DMA interrupts. 

**Environment (please complete the following information):**
 - OS: **Windows**
 - Toolchain: **GNU ARM Embedded 10/2020 Q4 Major, Zephyr SDK 0.15.1**
 - Commit SHA or Version used: **Zephyr version 3.2.99**

**Additional context**
While investigating this issue, I found that the PTP related config options in the attached prj.conf file appears to be contributing this issue. If I do not enable CONFIG_PTP_CLOCK and CONFIG_PTP_CLOCK_STM32_HAL the device can recieve multicast without any issues.
Specifically, the CONFIG_PTP_CLOCK option seems to be what is causing the issue, however, looking at the code it enables it is not clear how this could contribute to an error. 
Please note that the options in the prj.conf might seem superfluous however, they are necessary for the software for our product to work. Our software heavily relies on networking (both on loopback and external interfaces).

cperera-aud, 2022-11-23:
Hi @erwango, if there is anything you would like for me to try out on my end I'm more than happy to give it a go.

erwango, 2022-11-23:
> Hi @erwango, if there is anything you would like for me to try out on my end I'm more than happy to give it a go.

The issue description is quite complete. We'll not be able to reproduce your issue on our ends though, but we'll reach eth experts internally to check if this could be a known issue.

Though, you need to know that current eth HAL in use in Zephyr drive is being deprecated in favor of a new HAL API that should provide significant improvments. As a consequence, I expect experts' feedback will be: use new HAL.
A PR is on going to bring it up to Zephyr, see https://github.com/zephyrproject-rtos/zephyr/pull/45770. Issue is that, for now, this porting does not provide PTP support though. 

So potentially, what you'd need to do is to try this new API and port PTP support on the new API on your own, and see if fixing the issue.

Anyway, I'll keep you posted on the discussion with STM32 Eth experts.

cperera-aud, 2022-11-23:
Thanks and much appreciated for getting back @erwango. Yes we certainly would be interested in trying out the new HAL. The ethernet interface is pretty much the bread and butter of all of our devices.

erwango, 2022-11-24:
> Thanks and much appreciated for getting back @erwango. Yes we certainly would be interested in trying out the new HAL. The ethernet interface is pretty much the bread and butter of all of our devices.

New HAL support is going to be merged soon. It will be disabled by default. Once merged, you'll be able to enable it using a Kconfig symbol and test it on your end.
If eth is the core of you application, I encourage you to test it proactively and report to us bugs or good status so it can be enabled by default as soon as possible.

cperera-aud, 2022-11-24:
Thanks for the update, looking forward to trying it out on our end.

Desvauxm-st, 2022-11-25:
About the UDP/IP muticast when Clock PTP (EEE 1588) is used:
I had found one thing about issue PTP on the STM32Hxx
as above:
EEE 1588 PTP Clock Source in STM32H743
https://community.st.com/s/question/0D53W000004lxNhSAI/ieee-1588-ptp-clock-source-in-stm32h743
may be could you have a look at this issue about PTP clock.
Maybe it is link to this issue.

Desvauxm-st, 2022-12-05:
about New HAL support the ticket in ST in relation with this issue could be  :

Bug:
Ticket 79615 - Ethernet: Enable and handle the DMA
Ticket 67970 Read Buffer Underflow interrupt

How to reproduce:
Not handled in HAL_ETH_IRQHandler

Origin:
[Ticket 79615](https://intbugzilla.st.com/show_bug.cgi?id=79615) - Ethernet: Enable and handle the DMA Read Buffer Underflow interrupt

Status : implemented

How is resolved:  Enable RBU Interrupt

cperera-aud, 2022-12-06:
Thanks @Desvauxm-st for the links and cases, I'll have a look.

Also thanks @erwango for putting in the effort to get the new STM32 Eth API merged I'll give it a go on my end and see if it improves my situation. Just checking we still need to implement the PTP side of things to make use of the new API?

erwango, 2022-12-07:
> Just checking we still need to implement the PTP side of things to make use of the new API?

This is my understanding, but maybe @gleichdick can confirm

cperera-aud, 2022-12-15:
Hi @erwango, I managed to get an opportunity to try out the new V2 eth HAL API. Unfortunately I'm noticing some new issues now. 
- PTP is not working as it used to before, which is expected. [@gleichdick](https://github.com/gleichdick) would you be able to confirm or if there is anything I need to enable?
- When I recreate the same scenario (about 10Mbps of multicast) I continuously see the following error message 
`<err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:4480`
- If there's more multicast traffic then eventually there's a hard fault

cperera-aud, 2022-12-15:
Also, I must admit that to get a build going with my current setup I had to resort to cherry picking the changesets in https://github.com/zephyrproject-rtos/zephyr/pull/45770. We're currently at v3.1 for Zephyr. If I go to the tip of the main branch I end up with linker errors that I'm having a hard time sorting out. I'll just mention it in case if anyone has seen it before:

`'_GLOBAL_OFFSET_TABLE_' referenced in section '.text.<propritery implementation>' of ../lib/<proprietary library>.a(proprietary file.cpp.obj): defined in discarded section '.got.plt' of /opt/toolchains/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v7e-m/nofp/crtbegin.o`

Not just one but many. All of them were related to C++ files. With the C++ sample in zephyr I don't hit this error.

erwango, 2022-12-15:
> Also, I must admit that to get a build going with my current setup I had to resort to cherry picking the changesets in #45770. I was able to go to the tip of main for hal_stm32. We're currently at v3.1 for Zephyr. If I go to the tip of the main branch I end up with linker errors that I'm having a hard time sorting out. I'll just mention it in case if anyone has seen it before:
> 
> `'_GLOBAL_OFFSET_TABLE_' referenced in section '.text.<propritery implementation>' of ../lib/<proprietary library>.a(proprietary file.cpp.obj): defined in discarded section '.got.plt' of /opt/toolchains/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v7e-m/nofp/crtbegin.o`
> 
> Not just one but many. All of them were related to C++ files. With the C++ sample in zephyr I don't hit this error.

Could  you have a try with latest Zephyr SDK ? (at least 0.15.2)

erwango, 2022-12-15:
About :

> - When I recreate the same scenario (about 10Mbps of multicast) I continuously see the following error message
<err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:4480
> - If there's more multicast traffic then eventually there's a hard fault

Would you be able to reproduce these use cases and errors using `samples/net/zperf` ?

cperera-aud, 2022-12-16:
> About :
> 
> > * When I recreate the same scenario (about 10Mbps of multicast) I continuously see the following error message
> >    eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:4480
> > * If there's more multicast traffic then eventually there's a hard fault
> 
> Would you be able to reproduce these use cases and errors using `samples/net/zperf` ?

Hi @erwango, yes I was able to repro the `eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:4480` error with the `samples/net/zperf` and nucleo-H753ZI board. However, for the issue to come up I had to add the following to the prj.conf in `samples/net/zperf`:

```
CONFIG_ETH_STM32_HAL_API_V2=y
CONFIG_NET_L2_PTP=y
CONFIG_PTP_CLOCK=y
CONFIG_PTP_CLOCK_STM32_HAL=y
CONFIG_NET_PKT_TIMESTAMP=y
CONFIG_NET_PKT_TIMESTAMP_THREAD=y
```

I wasn't able to reproduce the hard fault though. 

I suppose the lack of proper PTP support is what's causing this issue? Does your team have a timeline on when PTP support would be added to work with the new eth HAL API?

erwango, 2022-12-16:
> I suppose the lack of proper PTP support is what's causing this issue? 

Regarding PTP:
There are actually some parts of PTP available with the new API, but until we're clear that support is complete I would not encourage enabling it.
We're actually currently lacking testing means on PTP feature. Would you be able to perform a basic round of PTP functional tests on your side, to make a first status ?  

Regarding the issue you're facing:
Can you confirm that issue is showing up only when PTP is enabled ?
Can you provide the zperf commands you're using so we can reproduc the exact use case ? (zperf enables a wide range of tetsing conditions)

On our side, w/o PTP enabled, we're also seeing some perf related issues on the new API, and the DMA error you report, but rather on F4 target, here is the use case:
- `iperf -c  192.168.0.111 -u  -t 1  -b 10M -P 1`
- `zperf udp download 5001`
This is still under characterization

cperera-aud, 2022-12-16:
> We're actually currently lacking testing means on PTP feature. Would you be able to perform a basic round of PTP functional tests on your side, to make a first status ?

Absolutely, we're more than happy to try out if you have a branch or something equivalent. We started hitting performance issues with PTP enabled. 

> Can you confirm that issue is showing up only when PTP is enabled ?

Yes correct. Without the additional PTP config options that I mentioned in my comment: https://github.com/zephyrproject-rtos/zephyr/issues/52409#issuecomment-1354088911 I didn't notice any issue when running the zperf app.

> Can you provide the zperf commands you're using so we can reproduc the exact use case ? (zperf enables a wide range of tetsing conditions)

Actually I didn't run any commands at all... I just ran the app without any commands but had about 10Mpbs of multicast traffic on my network using our devices. Sorry I must have misunderstood your earlier comment about running zperf. Generally with unicast traffic the performance of the ethernet driver has been alright. We haven't fully been able to test our application with unicast yet (i.e. full extent of the data throughput). In the cast of unicast there is a lot more things that we need to get correct for the application to work properly. In the case of multicast things are more simple in terms of setup so hence why we wanted to try out multicast first. 

That being said, I was able to reproduce the issue without our devices but by running iperf on a PC using the following command:
`iperf -c 239.254.1.10 -u -T 32 -t 10 -i 1 -b 10M`

In this situation I'm not running any zperf commands on the device, the device has started up and its running the zperf application without any commands.

If there's any logs or anything that you would like me to collect I'm more than happy to provide them.

cperera-aud, 2022-12-16:
I ought to mention to get iperf to work properly on my PC I had to disable the network interfaces except the interface I wanted the multicast to come out on.

cperera-aud, 2022-12-19:
> From what I can see, there might not be much to do though, only following section would require an update:
https://github.com/zephyrproject-rtos/zephyr/blame/b8f8c3c10eda6d1529d0e616b24ba9719efab2c8/drivers/ethernet/eth_stm32_hal.c#L490

Thanks @erwango for the pointer, I'll look into it.

cperera-aud, 2022-12-19:
In the case of the error previously mentioned:

> '_GLOBAL_OFFSET_TABLE_' referenced in section '.text.<propritery implementation>' of ../lib/<proprietary library>.a(proprietary file.cpp.obj): defined in discarded section '.got.plt' of [...]

I was able to get around it by using `CONFIG_COMPILER_OPT`. Previously with Zephyr 3.1 we were setting some additional C/CXX flags directly in our application CMakeLists.txt. With Zephyr 3.2 we pass in these additional compiler flags in `CONFIG_COMPILER_OPT`. I mentioned it in case anyone else runs into this issue.

cperera-aud, 2022-12-20:
Hi @erwango, Just a quick update, we've started looking into getting PTP working with the V2 API. There was a one line change we had to make to the eth_stm32_hal driver to get the timestamping to work. Despite that PTP is not working for us as it used to with the V1 API. One of the issues that we're noticing now is the receive timestamps (timestamp recovered from an incoming packet) is off by milliseconds (>10ms) This was observed when doing a trace in wireshark. We'll continue to investigate why this is the case. If you have any suggestions or pointers feel free to share them with us.

erwango, 2022-12-20:
@cperera-aud Thanks for the update.

Btw, we've detected what seems to be an actual regression on V2 when coming to performances. We'll keep you posted on progress in the analysis.

Also, here are some information on changes that were made in V2 API:
https://community.st.com/s/question/0D53W00001Gi9BoSAJ/ethernet-hal-driver-reworked-by-st-and-available-in-22q1-preview-now-available-on-github

cperera-aud, 2022-12-21:
Thank you @erwango for the update from your end. I'll check back regularly on the link that you've provided.

Desvauxm-st, 2022-12-21:
about the issue you said :
                    **When I recreate the same scenario (about 10Mbps of multicast) I continuously see the following error message
                    <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:4480**

I had reproduce the same error message with the board nucleo_F429zi on my side with iperf.
I made a modification about DMACIER register  ( disable the interrupt  bit :ETH_DMACIER_AIE) to remove this error message.

can you test that on the board Nucleo-H753ZI on your side :
do like that:

Into the file :  zephyrproject\modules\hal\stm32\stm32cube\stm32h7xx\drivers\src\stm32h7xx_hal_eth.c
into the function:  HAL_StatusTypeDef HAL_ETH_Start_IT(ETH_HandleTypeDef *heth)
Approximatively in line 800
Remove the bit ETH_DMACIER_AIE ( Abnormal Interrupt Summary Enable) like that:

   // modification just to try about : Abnormal Interrupt Summary Enable ==> removed this IRQ
   // __HAL_ETH_DMA_ENABLE_IT(heth, (ETH_DMACIER_NIE | ETH_DMACIER_RIE | ETH_DMACIER_TIE  |
   //                                ETH_DMACIER_FBEE | ETH_DMACIER_AIE | ETH_DMACIER_RBUE));
  **__HAL_ETH_DMA_ENABLE_IT(heth, (ETH_DMACIER_NIE | ETH_DMACIER_RIE | ETH_DMACIER_TIE  |
                                   ETH_DMACIER_FBEE  | ETH_DMACIER_RBUE));**
   // END of modification just to try : Abnormal Interrupt Summary Enable ==> removed this IRQ

cperera-aud, 2022-12-22:
Hi @Desvauxm-st,

Thank you much appreciated for the suggestion. I gave it a try on my end and interestingly enough it seems like the multicast receive performance with the V2 API has significantly improved. In one instance the device had hard faulted after running for more than an hour (unfortunately I didn't have the debugger attached to see what really happened). Second time around the device was up for more than 3 hours and it didn't crash. Also the device was running fine and receiving multicast without any issues. I noticed that in the V1 API the ETH_DMACIER_AIE bit is set? 

I'm still looking into why PTP is not working. PTP is working quite flawlessly for us with the V1 API. Sorry if I'm adding unrelated things to this case but it seems like when PTP clocking is enabled in the prj.conf IGMP reports are not being sent out too. This happens regardless of using the V1 or V2 eth HAL API. Not sure if you or anyone else has noticed this issue. In this situation it becomes a bit difficult to deploy the multicast hash filter for the STM32H7 as discussed in https://github.com/zephyrproject-rtos/zephyr/pull/39089/commits/56d2b377125918982d43b1f573c705cfa2e01dda because the net_if_mcast_cb mentioned in https://github.com/zephyrproject-rtos/zephyr/pull/39089/commits/56d2b377125918982d43b1f573c705cfa2e01dda#diff-0d2e5611dcfdafb92af335bba5c381139ed9ce88182b5fe03d9655f7984725bcR792. Again apologies for dragging in IGMP and multicast hash filtering. 

Regarding the link that you had posted in https://github.com/zephyrproject-rtos/zephyr/issues/52409#issuecomment-1327419957. The recommendations are already part of the zephyr eth_stm32_hal driver.

cperera-aud, 2022-12-22:
Hi @erwango, @Desvauxm-st,

I have some good news, I was able to add a fix to the V2 HAL API to get PTP to work. I tried my best to see if I can make the change to the zephyr eth_stm32_hal driver as it was done with the V1 API but wasn't able to. But instead it seemed it was more convenient to make the change in the V2 HAL API. In summary what I did was exactly outlined in the reference manual. The code in the zephyr eth_stm32_hal driver did something similar with the V1 API. In summary what I did was:
- In the `HAL_ETH_ReadData` function where it checks for the last descriptor, I added a checked if the TSA bit was set in DESC1
- If the TSA bit is set then have a peak at the context descriptor which should be the one after the last descriptor
- If the CTXT bit is set in the context descriptor then extract the timestamps

Apologies if the fix seems like a hack. With the change I see the exact same PTP performance that I saw with the V1 API.

erwango, 2022-12-22:
@cperera-aud Great news, thanks for the heads up. Can you share your modifications (HAL Fix and Zephyr driver update for PTP)? We'll fw this to HAL dev team and try to get their feedback.

Then, now that you're able to get PTP functional on V2, does it fix the initial multicast and PTP issue you reported ?

cperera-aud, 2022-12-22:
Hi @erwango,

Sure, I've attached the fixes as patch files (sorry the code is not 100% cleaned up). I hope the patch files for now is alright? 

[zephyr_eth_stm32_hal_ptp.patch](https://github.com/zephyrproject-rtos/zephyr/files/10287088/zephyr_eth_stm32_hal_ptp.patch)
[stm32_eth_hal_v2_ptp.patch](https://github.com/zephyrproject-rtos/zephyr/files/10287089/stm32_eth_hal_v2_ptp.patch)

> Then, now that you're able to get PTP functional on V2, does it fix the initial multicast and PTP issue you reported ?

Yes I would say that the initial issue that was reported appears to be fixed. We certainly will be doing more tests around multicast. After Friday (23rd), we'll be back in the new year. Happy holidays!

erwango, 2022-12-22:
Great news. We'll fw this internally. On our side, we're closing this evening and are back on 3rd. We'll keep you posted on the progress.
Happy holidays to you!

erwango, 2023-01-11:
@cperera-aud Best wishes for this new year.
About PTP enablement and HAL fies. Do you plan to push them upstream ?

cperera-aud, 2023-01-11:
Hi @erwango,

Happy new year and best wishes for the new year too! 

Yes I'd be more than happy to get the changes upstream. I was wondering if the HAL team were ok with the changes? The zephyr changes were fairly minor so I can start with it if the HAL team needs more time to review the HAL related changes?

erwango, 2023-01-11:
>  The zephyr changes were fairly minor so I can start with it

Great, txs!

>  I was wondering if the HAL team were ok with the changes? [...] if the HAL team needs more time to review the HAL related changes?

We've forwarded the point. Waiting for feedback for now. Meanwhile, I see no issue in pushing a temp fix in zephyr HAL "fork".

erwango, 2023-01-11:
Oops, already started by @Desvauxm-st:
https://github.com/zephyrproject-rtos/zephyr/pull/53694

cperera-aud, 2023-01-11:
Sounds good thanks @erwango. I looked at the changes by @Desvauxm-st and they look good to me 👍

cperera-aud, 2023-01-11:
@erwango, I forgot to ask, how did you want to proceed with the HAL changes?

cperera-aud, 2023-01-27:
Hi @erwango, @Desvauxm-st, I just wanted to revisit the comment: https://github.com/zephyrproject-rtos/zephyr/issues/52409#issuecomment-1361149149

On our end our application so far is performing alright. Our application is an audio over IP application with PTP and control information. We still haven't fully completed our application yet. The control and PTP parts are fine and basic audio over IP is working. We're moving towards verifying data integrity. So far we're currently ignoring the errors from the HAL API. 

I suppose I was wondering, are these errors real and do they need to be fixed at some point in time? I think there are some pending fixes which were mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/52409#issuecomment-1359081592? Any update or any information that you can share is much appreciated!

erwango, 2023-01-30:
> I suppose I was wondering, are these errors real and do they need to be fixed at some point in time? I think there are some pending fixes which were mentioned in [...]

Issue mentioned was described in [later comment](https://github.com/zephyrproject-rtos/zephyr/issues/52409#issuecomment-1361149149). The actual fix is simply to not log errors (`ETH_DMACIER_AIE` neablement should be kept for statistics if we want to be able to count them in stats).

There isn't any other bug that remain to be fixed on our side (known to us at the moment).

Btw, considering the progress on the initial issue reported, I consider it would be alright to close current issue. If required, other specific issues could be open. Is that fine for you ?

cperera-aud, 2023-01-30:
Thanks @erwango for getting back to me. Yes its fine to close the case. What I was a bit interested in your previous comment regarding 

> Btw, we've detected what seems to be an actual regression on V2 when coming to performances. We'll keep you posted on progress in the analysis.

Was there any update from the last time you mentioned it?

erwango, 2023-01-31:
> Was there any update from the last time you mentioned it?

No update:
- there are no more console logs on DMA error, which fixes the main issue
- DMA Allocation error IRQ is kept (to enable error stats logging)

erwango, 2023-01-31:
Forgot to add: Thanks for your patience and help in testing and investigating.

closed 2023-01-31 by erwango

cperera-aud, 2023-01-31:
You're welcome, glad to be able to help!