zephyrproject-rtos/zephyr #52415: tests: kernel: timer: timer_behavior: kernel.timer.timer fails

nordic-piks, 2022-11-21:
**Describe the bug**
The tests/kernel/timer/timer_behavior/kernel.timer.timer fails

Observed for
- nrf9160dk_nrf9160_ns
- nrf5340dk_nrf5340_cpuapp_ns

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf9160dk_nrf9160 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/timer/timer_behavior -p nrf9160dk_nrf9160_ns --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.2.0-1729-g1374415a2ce8 ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift
periodic timer behavior test
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 32768, expected period 1000 us, samples 10000, period 32.768000 cycles, expected period drift 7.080078 us
statistics samples 10000, rollovers 0, mean 1007.086182 us, variance 0.372492 us, stddev 0.610321 us, min 1007.080078 us, max 1068.115234 us
statistics samples 10000, rollovers 0, mean 33.000200 cycles, variance 0.000400 cycles, stddev 0.019999 cycles, min 33 cycles, max 35 cycles
timer start cycle 14129, end cycle 344131, total time 10070861.816406, expected time 10000000.000000, expected time drift 70800.781250, difference 61.035156
PASS - test_jitter_drift in 10.342 seconds
===================================================================
TESTSUITE timer_jitter_drift succeeded
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 32768Hz, Expecting 81920 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 0
Faking busy work, remaining timers is 4, timer callbacks 6852
Faking busy work, remaining timers is 4, timer callbacks 13528
Faking busy work, remaining timers is 4, timer callbacks 20209
Faking busy work, remaining timers is 4, timer callbacks 26888
Faking busy work, remaining timers is 4, timer callbacks 33564
Faking busy work, remaining timers is 4, timer callbacks 40242
Faking busy work, remaining timers is 4, timer callbacks 46918
Faking busy work, remaining timers is 4, timer callbacks 53590
Faking busy work, remaining timers is 4, timer callbacks 60265
Faking busy work, remaining timers is 4, timer callbacks 66943
Faking busy work, remaining timers is 4, timer callbacks 73623
Faking busy work, remaining timers is 4, timer callbacks 80300
One Tick Timer Train Done, took 10192 ms, busy loop ran 13 times
Timer 0 max delta 5 cycles or 153 us
Timer 1 max delta 5 cycles or 153 us
Timer 2 max delta 5 cycles or 153 us
Timer 3 max delta 5 cycles or 153 us
Assertion failed at ../../../../../../../../../../zephyr/zephyr/tests/kernel/timer/timer_behavior/src/tick_timer_train.c:143: timer_tick_train_test_one_tick_timer_train: (busy_loops > acceptable_busy_loops is false)
Expected thread to run while 1 tick timers are firing
FAIL - test_one_tick_timer_train in 10.308 seconds
===================================================================
TESTSUITE timer_tick_train failed.
===================================================================
RunID: 482c2e87f678597647531b8b7928e428
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: zephyr-v3.2.0-1729-g1374415a2ce8

nordicjm, 2022-11-21:
@nordic-piks did you try with this disabled? https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nrf5340dk_nrf5340/Kconfig.defconfig#L100

nordic-piks, 2022-11-21:
No, I did not try. If I understand correctly, this setting is enabled both for nrf5340dk_nrf5340_cpuapp and nrf5340dk_nrf5340_cpuapp_ns, but test are failing only at nrf5340dk_nrf5340_cpuapp_ns.

PR #54021 (2023-01-25): drivers: nrf_rtc_timer: Adjust set_absolute_alarm to prevent CC misses
This is a follow-up to commit https://github.com/zephyrproject-rtos/zephyr/commit/205e684958ebea4d7f2a15d75b94aac880fb2390.

Add an nrf_rtc_timer test case that schedules a set of consecutive timeouts, each one on the next cycle of the system timer but each time after a slightly larger delay. This test case reveals a problem in the current nrf_rtc_timer implementation that sometimes a compare value can be missed what leads to timeouts delayed by 512 seconds.
Adjust the implementation of `set_absolute_alarm()` to avoid missi [... cut]
closed 2023-01-25 by fabiobaltieri