zephyrproject-rtos/zephyr #52417: DMA callback API

fouge, 2022-11-21:
**Is your enhancement proposal related to a problem? Please describe.**

The DMA driver has the ability to configure a callback called whenever a DMA transfer is completed. Below is the current description for this callback which is set when configuring the DMA device (see `struct dma_config`):

```c
/**
 * @typedef dma_callback_t
 * @brief Callback function for DMA transfer completion
 *
 *  If enabled, callback function will be invoked at transfer completion
 *  or when error happens.
 *
 * @param dev Pointer to the DMA device calling the callback.
 * @param user_data A pointer to some user data or NULL
 * @param channel The channel number
 * @param status 0 on success, a negative errno otherwise
 */
typedef void (*dma_callback_t)(const struct device *dev, void *user_data,
			       uint32_t channel, int status);
```

Usage of DMA on different microcontrollers might differ. For example, most STM32 microcontrollers have two interrupts: one for half complete, and one for fully complete transfer. It might be needed to call the DMA callback for both of these interrupts and the user of the DMA device might need to be informed about which interrupt occurred. It is not possible at this point to pass this kind of information using the current API description: `0 on success` meaning when DMA transfer completes.

The `dma_get_status()` function could be used to guess which interrupt occurred but the current status of DMA might have changed by the time the user can call the `dma_get_status()` function (it happened to me when working on #51795) and we might want to process the buffers differently depending on the source of the interrupt, not the current status.

**Describe the solution you'd like**

My idea is to change the `dma_callback_t` description so that any positive `status` gives information about the source of the interrupt, which might differ depending on the platform: STM32 might use 2 status codes for fully and half complete while other platforms might use only 1 status code.

**Describe alternatives you've considered**

Store several callbacks, one for each interrupt sources in the platform-dependent implementation, like done in `dma_dw_common.h`:

```c
struct dw_dma_chan_data {
        // ...
	dma_callback_t dma_blkcallback; // block complete callback
	void *blkuser_data;
	dma_callback_t dma_tfrcallback; // transfer complete callback 
	void *tfruser_data;
}
```

I don't think this is a bad idea and actually stumbled upon this implementation while writing this issue. It has the advantage of having one user data pointer per callback instead of one for all the interrupt sources.

Let me know if you have preferences.

fouge, 2022-11-21:
Also worth mentioning #37372 which proposes to use an enum to give the status, which would be shared across all the DMA implementations

fouge, 2022-12-02:
I dived a bit more into it and the `dma.h` `struct dma_config` only has one callback used when configuring a DMA device so I don't think it's that easy to store several callbacks. It would involve calling the `.config` function of `struct dma_driver_api` with several parameters to configure the DMA clients differently depending on `struct dma_config`...

I think the idea of having a negative status for errors and any positive value for status about a DMA interrupt is the best option.

cc @teburd let me know what you think, I can submit a PR to change the DMA API description and status handling (make sure `status < 0` for error instead of `status != 0`.

teburd, 2022-12-02:
I'd say send a PR, I'm ok with updating the status to indicate more information than error or success.

For the circular modes this makes good sense. This change will require some careful review though of the code base to ensure status != 0 isn't being used to check for the error case. While not exactly breaking the API, it is changing expectations of how the API works. So in that regard we may need to ensure broadcast this change in a few places. I need to review the projects requirements around stable API changes and if this would constitute one.

teburd, 2022-12-07:
Please change the issue title to one similar to the stable API change RFC title noted here, https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes and make sure all the fields are here (I believe they are).

fouge, 2022-12-21:
@teburd I just modified the issue so that it can be treated as an RFC

teburd, 2023-01-04:
Thanks @erwango was out for the holidays

fouge, 2023-01-17:
@teburd any change to see this discussed in the weekly meeting?

mbolivar-nordic, 2023-01-31:
Architecture WG: Some agreement, some request for more design on possible use cases beyond this STM32 part's half-full/full behavior.

galak, 2023-01-31:
For the https://github.com/zephyrproject-rtos/zephyr/pull/51795 is a change in the status really needed?  If I'm understanding the code change there is more about if the callback is a cyclic callback or not.  Can't that be determined in the callback in the uart based on how `cyclic` got configured?

fouge, 2023-02-01:
@galak 
Passing the status from the DMA callback is desired because `dma_get_status()` provides status of a the DMA peripheral that might have changed since the interrupt fired.

Let's say we are receiving data into a RX buffer that gets full but the status shows a write index at the beginning of the RX buffer like at index `1` for example (data wrapped around). As we keep track of the last write index into the DMA buffer between interrupts, we can correctly use the data from the buffer but because the UART API expects a continuous chunk of memory (see API definition below), we then need to call the UART callback **twice** ([see `uart_callback_t`](https://github.com/zephyrproject-rtos/zephyr/blob/5d9c65fe7f817e67cce7649874f429db9b6abf3c/include/zephyr/drivers/uart.h#L325)), instead of once, because the write index made progress, even though we are not done receiving UART data (idle interrupt didn't fire).

```c
/**
 * @brief UART RX event data.
 *
 * The data represented by the event is stored in rx.buf[rx.offset] to
 * rx.buf[rx.offset+rx.len].  That is, the length is relative to the offset.
 */
struct uart_event_rx {
	/** @brief Pointer to current buffer. */
	uint8_t *buf;
	/** @brief Currently received data offset in bytes. */
	size_t offset;
	/** @brief Number of new bytes received. */
	size_t len;
};
``` 

Here is a piece of logs from my running example. My buffer is 2048-byte long, and the message is 688-byte long.
The first line will lead to 2 calls to the UART callback: one for data from index `1729` to `2048` and one from `0` to `1`, instead of 1 call if we use information from the DMA module that the buffer is full.
```
2023-02-01 11:18:15.373 [00:03:35.665,000] <dbg> uart_stm32: async_evt_rx_rdy: rx_rdy: (1729 1)
2023-02-01 11:18:15.381 [00:03:35.668,000] <dbg> uart_stm32: async_evt_rx_rdy: rx_rdy: (1 369)
```

Because we know the interrupt reason from the DMA module, I thought it would be more efficient to directly use this piece of information and pass it to the DMA user, in this case the UART module.

I made a running example this morning that doesn't involve DMA API modifications and can probably open a new PR with theses changes.

teburd, 2023-02-23:
@galak please take a look, thanks!

galak, 2023-03-09:
We should make the return code well defined.  So 0 means DMA complete, and 1 means circular water mark complete.

teburd, 2023-03-10:
@erwango @fouge please see the response here from Kumar, does this make sense? Effectively the request is the status values need to be well defined for the success cases. 0 being "transfer complete" and 1 being "watermark complete". In the case of st's circular dma setup 1 would imply the half full (watermark) completion event.

This could be useful as well for other DMAs that implement circular buffers with potentially configurable/fixed watermarks.

Thanks!

fouge, 2023-03-10:
Hey
if you look at my UART implementation, I actually use 3 different statuses in case DMA times out. In this case, the `dma_get_status` function is used, [see here](https://github.com/worldcoin/zephyr/blob/worldcoin/stm32_dma_circular_mode/drivers/serial/uart_stm32.c#L964).

teburd, 2023-03-10:
> Hey if you look at my UART implementation, I actually use 3 different statuses in case DMA times out. In this case, the `dma_get_status` function is used, [see here](https://github.com/worldcoin/zephyr/blob/worldcoin/stm32_dma_circular_mode/drivers/serial/uart_stm32.c#L964).

I only see the one added status code here https://github.com/zephyrproject-rtos/zephyr/pull/51795/files#diff-14bcedc892b7643865d068e532c3dab525465a9814a505eba6724db5026028b7R122

fouge, 2023-03-10:
@teburd Indeed you're right, the other status code is only used internally in `uart_stm32.c` level so I guess the API can be changed with these 2 status code. 
I will send an updated PR next week

PR #52780 (2023-05-08): dma: callback to allow for positive statuses on success
Make use of positive status values in the DMA callback to pass info to the DMA client after a successful DMA operation. The status integer is dependent to the DMA implementation and might differ on each platform.

Fixes #52417

cc @teburd
closed 2023-05-08 by carlescufi