zephyrproject-rtos/zephyr #52434: drivers: pinctrl: Use common pinctrl DTS bindings for Cypress psoc6

henrikbrixandersen, 2022-11-21:
**Is your enhancement proposal related to a problem? Please describe.**
As noted in #39740 the Cypress PSoC6 implementation uses a custom devicetree pinctrl representation (as seen in `dts/arm/cypress/psoc6-pinctrl.dtsi`).

**Describe the solution you'd like**
The Cypress PSoC6 implementation should use one of the standard Zephyr pinctrl devicetree representations as described in https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#devicetree-representation along with a pinctrl driver.

**Describe alternatives you've considered**
None.

**Additional context**
None.

gmarull, 2023-06-13:
@npal-cy This has been open for a while, is the platform maintained? Should it be removed or is it going to be fixed?

ifyall, 2023-06-22:
@gmarull, these platforms are being maintained. We will generate a plan internally to bring them up to standards.

henrikbrixandersen, 2023-08-22:
@ifyall Any updates on this?

keithmwheeler, 2024-04-23:
Our team will be addressing this issue over the next three weeks; hopefully will complete sooner.

npal-cy, 2024-05-03:
Working on this now.

npal-cy, 2024-05-23:
PR is ready for review: https://github.com/zephyrproject-rtos/zephyr/pull/73233

closed 2024-06-04 by aescolar