zephyrproject-rtos/zephyr #52435: drivers: pinctrl: add pinctrl driver for armv7-m/armv8-m

henrikbrixandersen, 2022-11-21:
**Is your enhancement proposal related to a problem? Please describe.**
As noted in #39740 the armv7m/armv8-m implementations uses a custom pinctrl implementation (as seen in https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/v2m_musca_b1/pinmux.c and https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/mps2_an385/pinmux.c).

**Describe the solution you'd like**
The armv7m/armv8-m implementations should use one of the standard Zephyr pinctrl devicetree representations as described in https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#devicetree-representation along with a pinctrl driver.

**Describe alternatives you've considered**
None.

**Additional context**
None.

microbuilder, 2023-12-29:
@ithinuel Do you have the bandwidth to look at this after the holidays? If you don't, let me know and I'll schedule time in for it myself. It completely fell off my radar. 😬

ithinuel, 2024-01-04:
I think I should be àble to do that at some point in this quarter. 👍

henrikbrixandersen, 2024-08-17:
@ithinuel Any plans to address this?

ithinuel, 2024-08-22:
Yes, although low in priority. Should it be escalated to a higher prio?

henrikbrixandersen, 2024-09-02:
> Yes, although low in priority. Should it be escalated to a higher prio?

Well, pinmux was deprecated in Zephyr v3.2 and the plan was to remove all non-pinctrl muxing too. This was nearly two years ago. I think this deserves some attention.

ithinuel, 2025-02-17:
Little, we are working on this 👍.