zephyrproject-rtos/zephyr #52436: drivers: pinctrl: add pinctrl driver for ARC emsk

henrikbrixandersen, 2022-11-21:
**Is your enhancement proposal related to a problem? Please describe.**
As noted in #39740 the ARC emsk platform uses a custom pinctrl driver (as seen in https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arc/em_starterkit/pmodmux.c).

**Describe the solution you'd like**
The ARC emsk implementation should use one of the standard Zephyr pinctrl devicetree representations as described in https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#devicetree-representation along with a pinctrl driver.

**Describe alternatives you've considered**
None.

**Additional context**
See also #45771.

gmarull, 2023-05-30:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/56926

gmarull, 2023-06-13:
Reopening since it looks like some ARC boards still use legacy pinctrl methods:

https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arc/em_starterkit/pmodmux.c
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arc/hsdk/platform.c

henrikbrixandersen, 2024-08-17:
@ruuddw, @evgeniy-paltsev Ang plans to address this?

ruuddw, 2024-08-19:
For EMSK I expect we will not add this. These boards are phased out in favor of the EMSDP boards. New EMSK hardware boards are not available anymore, new projects use EMSDP.

henrikbrixandersen, 2024-08-19:
> For EMSK I expect we will not add this. These boards are phased out in favor of the EMSDP boards. New EMSK hardware boards are not available anymore, new projects use EMSDP.

Should we close this, then?

gmarull, 2024-08-19:
> For EMSK I expect we will not add this. These boards are phased out in favor of the EMSDP boards. New EMSK hardware boards are not available anymore, new projects use EMSDP.

Should they be deleted if they are not keeping up with Zephyr developments?

ruuddw, 2024-08-19:
No, or not yet - they are still used, though mainly focusing on the ARC processor on it, not peripherals/pinctrl.

gmarull, 2024-08-19:
> No, or not yet - they are still used, though mainly focusing on the ARC processor on it, not peripherals/pinctrl.

so can legacy pinctrl stuff be removed?

ruuddw, 2024-08-19:
There is no legacy or other pinctrl dependency in this solution, just a small local "pmodmux" is used. As far as I know, the legacy pinctrl was removed long time ago already.

evgeniy-paltsev, 2024-08-19:
> As far as I know, the legacy pinctrl was removed long time ago already.

Can confirm. Now we only platform code where we just write to MMIO.

ruuddw, 2024-08-19:
Based on comments and discussion, closing this issue as not planned.

closed 2024-08-19 by ruuddw