zephyrproject-rtos/zephyr #52448: esp32: subsys: settings: not working properly on esp_wrover_kit

honestech74, 2022-11-22:
**Describe the bug**
The `settings` subsys doesn't work properly on `esp_wrover_kit`.

But it works on esp32-wroom and esp32-c3.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b esp_wrover_kit samples/subsys/settings/ -p
2. See error
```
save <alpha/beta/voltage> key directly: fail (err -2)
...
delete <alpha/length>: fail (err -2)
...
save <gamma> key directly: fail (err -2)
```

**Expected behavior**
It seems to work for reading but doesn't work for saving and deleting.
When tried to build as `esp32` board and ran it on ESP32 WROVER board, it failed on `settings_subsys_init()` with error code is `(EDEADLK)`.

**Logs and console output**

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6596
load:0x40078000,len:15216
ho 0 tail 12 room 4
load:0x40080400,len:3992
entry 0x40080634
I (29) boot: ESP-IDF 2cb20ac6c 2nd stage bootloader
I (29) boot: compile time 18:31:54
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (111) esp_image: segment 1: paddr=00010044 vaddr=3ffb0000 size=001e8h (   488) load
I (120) esp_image: segment 2: paddr=00010234 vaddr=3ffb01e8 size=0013ch (   316) load
I (128) esp_image: segment 3: paddr=00010378 vaddr=40080000 size=041a4h ( 16804) load
I (144) esp_image: segment 4: paddr=00014524 vaddr=00000000 size=0bb14h ( 47892) 
I (162) esp_image: segment 5: paddr=00020040 vaddr=3f400040 size=015c4h (  5572) map
I (165) esp_image: segment 6: paddr=0002160c vaddr=00000000 size=0ea0ch ( 59916) 
I (190) esp_image: segment 7: paddr=00030020 vaddr=400d0020 size=03abch ( 15036) map
I (197) boot: Loaded app from partition at offset 0x10000
I (197) boot: Disabling RNG early entropy source...
�*** Booting Zephyr OS build zephyr-v3.2.0  ***

*** Settings usage example ***

settings subsys initialization: OK.
subtree <alpha> handler registered: OK
subtree <alpha/beta> has static handler

##############
# iteration 0
[... 236 lines cut ...]
save <gamma> key directly: fail (err -2)

=================================================
Inject the value to the setting destination in runtime

<alpha/beta/source> = RT
injected <alpha/beta/source>: OK.
  The settings destination off the key <alpha/beta/source> has got value: "RT"

=================================================
Read a value from the setting destination in runtime

fetched <alpha/beta/source>: OK.
  String value "rtos" was retrieved from the settings destination off the key <alpha/beta/source>

*** THE END  ***
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Toolchain: Zephyr SDK v0.15.1
 - Commit SHA or Version used: zephyr-v3.2.0 release

**Additional context**
No clue but I think it might be related to PSRAM and flash on the same SPI bus.

Laczen, 2022-11-22:
@honestech74, the eror message is indicating that there already is data in the flash area used for settings. Could it be that this area was not erased? As a protection against wrongly erasing data that could be vital for the user the settings_subsys_init might not remove what is already there.

honestech74, 2022-11-22:
@Laczen Tried to erase the flash using `esptool.py erase_flash` but got the same errors.

sylvioalves, 2022-11-22:
@honestech74, there reason is that `esp_wrover_kit` does not have an overlay file in this sample code.  
To get this properly working, you can create a custom `boards/esp_wrover_kit.conf` and copy `boards/esp32.conf` content or using default `esp32.conf` file:  
`west build -b esp_wrover_kit samples/subsys/settings/ .. -DOVERLAY_CONFIG=boards/esp32.conf
`

honestech74, 2022-11-23:
@sylvioalves I forgot those kconfigs. It works now. Thank you.

closed 2022-11-23 by honestech74