zephyrproject-rtos/zephyr #52449: net: ip: igmp: IGMP v2 membership reports are sent to 224.0.0.2 instead of the group being reported

cperera-aud, 2022-11-22:
**Describe the bug**
The Zephyr network stack is sending IGMP v2 membership reports to the 224.0.0.2 multicast address instead of the multicast group it is being reported. When reading through RFC 2236 (http://www.faqs.org/rfcs/rfc2236.html) Section 9 this seems to be a violation. I've copy pasted the relevant text from the RFC:

```
9.  Message destinations

   This information is provided elsewhere in the document, but is
   summarized here for convenience.

   Message Type                  Destination Group
   ------------                  -----------------
   General Query                 ALL-SYSTEMS (224.0.0.1)
   Group-Specific Query          The group being queried
   Membership Report             The group being reported
   Leave Message                 ALL-ROUTERS (224.0.0.2)
```

The reason why I am reporting this issue is when I was reporting https://github.com/zephyrproject-rtos/zephyr/issues/52409 , with IGMP snooping enabled on my Cisco SG300 switch the STM32H7 device was not able to receive UDP IPv4 multicast data when it opened up a socket. What I observed was despite sending membership reports for 224.0.0.2 the device was able to receive mDNS multicast however, it was not able to receive multicast from a 239.255.x.x address. Hence why this issue was raised. 

The leave reports are sent to the correct destination address of 224.0.0.2. 

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? **Nucleo-H753ZI Evaluation board**
 - What have you tried to diagnose or workaround this issue? **This issue cannot be worked around since IGMP snooping is required to avoid flooding a device with multicast**
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? **This issue is present in v3.1 and v3.2**

**To Reproduce**
Steps to reproduce the behavior:
1. Run wireshark
2. Open a socket to receive UDP/IP multicast data
3. Join the multicast group address (`net_ipv4_igmp_join`)

**Expected behavior**
The device should send the IGMP membership report to the multicast group that is being reported and be able to receive the UDP/IP multicast data on a network with IGMP snooping enabled.

**Impact**
We cannot ship our product that is not able to receive UDP/IP multicast in a network with IGMP snooping enabled.

**Additional context**
I have a fix for this issue: in zephyr/subsys/net/ip/igmp.c: `igmp_send_generic`, the following should be changed:

```
ret = igmp_v2_create_packet(pkt, &all_routers, addr,
			join ? NET_IPV4_IGMP_REPORT_V2 : NET_IPV4_IGMP_LEAVE);
```

to

```
ret = igmp_v2_create_packet(pkt, join ? addr : &all_routers, addr,
			join ? NET_IPV4_IGMP_REPORT_V2 : NET_IPV4_IGMP_LEAVE);
```

Whoever is reviewing this ticket agrees with my change, feel free to assign the ticket to me and I will send a PR for the change and quote the ticket number in the change commit.

jukkar, 2022-11-22:
Thanks for the report, please send a PR for it.

cperera-aud, 2022-11-24:
Sorry @jukkar for the delay in submitting the PR: https://github.com/zephyrproject-rtos/zephyr/pull/52521. This was my first time sending a PR so it took some time to sort out the various issues around pushing changes github.

PR #52521 (2022-11-24): net: ip: Fixed IGMP V2 membership report destination address.
As per RFC 2236 Section 9, the IGMP V2 membership reports should be sent to the group multicast address being reported. Without this fix, when IGMP snooping is enabled on a network a device is not able to receive multicast from certain multicast addresses e.g. 239.255.x.x.

Fixes #52449
commit 67db7d07 (2022-11-24): net: ip: Fixed IGMP V2 membership report destination address.

As per RFC 2236 Section 9, the IGMP V2 membership reports should be sent
to the group multicast address being reported. Without this fix, when
IGMP snooping is enabled on a network a device is not able to receive
multicast from certain multicast addresses e.g. 239.255.x.x.

Fixes #52449