zephyrproject-rtos/zephyr #52457: compilation error with "west build -b lpcxpresso54114_m4 samples/subsys/ipc/openamp/ "

arnopo, 2022-11-22:
**Describe the bug**

runing the command `west build -b lpcxpresso54114_m4 samples/subsys/ipc/openamp/` result in a compilation error

```
/local/home/frq07267/views/zephyrproject/zephyr/soc/arm/nxp_lpc/lpc54xxx/soc.c:31:10: fatal error: zephyr_image_info.h: No such file or directory
   31 | #include <zephyr_image_info.h>
      |          ^~~~~~~~~~~~~~~~~~~~~

```
The bug seems to have been introduced in  PR https://github.com/zephyrproject-rtos/zephyr/pull/48709
by the commit 407fe95af3 ("samples: openamp: add lpc54114 configuration settings")

**Impact**
Fails the OpenAMp CI (https://github.com/OpenAMP/open-amp/issues/450)

arnopo, 2022-11-22:
FYI @danieldegrasse

danieldegrasse, 2022-11-22:
Hi @arnopo- This sample has been updated to use systembuild, which allows multiple Zephyr images to be built using one command. If you build the sample with the following command: `west build --sysbuild -b lpcxpresso54114_m4 samples/subsys/ipc/openamp/`, the build should pass. Will this work for your CI? The missing header is one generated by the M0 core image build as collateral, which contains load address information for the M0 image. This information is used by the M4 image to determine where to copy the M0 image on boot before kicking off the M0 core.

arnopo, 2022-11-22:
Thanks for your reactivity, I will test it in my CI and close the issue if fixed

arnopo, 2022-11-22:
That solves the CI issue on OpenAMP , thanks for the advise @danieldegrasse

closed 2022-11-22 by arnopo