zephyrproject-rtos/zephyr #52458: Add setting default radio TX power for IEEE 802.15.4 in OpenThread

interfer, 2022-11-22:
**Is your enhancement proposal related to a problem? Please describe.**

While trying to set the transmission power for **OpenThread** running `samples/net/sockets/echo_server` with `CONFIG_NET_L2_IEEE802154_RADIO_DFLT_TX_POWER` and `CONFIG_NET_CONFIG_IEEE802154_RADIO_TX_POWER` but none of them works.
AFAIK that dbm value is coming into IEE driver API function (`ieee802154_radio_api.set_txpower`) from `zephyr/modules/openthread/platform/radio.c`.
The [`static int8_t tx_power`](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/openthread/platform/radio.c#L86) there is used in getter/setter functions initialized by zero. So the problem is it doesn't change in runtime, **there are no** `otPlatRadioSetTransmitPower` or `otPlatRadioSetChannelMaxTransmitPower` **calls doing this**. Only if explicitly called in driver.

**Describe the solution you'd like**

Apparently in `modules/lib/openthread/src/core/config/misc.h` there's a `OPENTHREAD_CONFIG_DEFAULT_TRANSMIT_POWER` define which might be used for these purposes as it's done [here](https://github.com/openthread/openthread/pull/394) in OT repo.
Or consider creating Kconfig option for initializing the TX power value `static int8_t tx_power` in `zephyr/modules/openthread/platform/radio.c`

fgrandel, 2024-07-24:
Reviewed and removed myself as assignee as this seems to be a nRF and/or OT problem only. The IEEE 802.15.4 driver API should not be the problem here. If I'm wrong feel free to re-add me. @rlubos what do you think?

rlubos, 2024-07-25:
> Or consider creating Kconfig option for initializing the TX power value static int8_t tx_power in zephyr/modules/openthread/platform/radio.c

This seems to have been implemented in https://github.com/zephyrproject-rtos/zephyr/pull/53125, the default TX power for OpenThread can be set with `CONFIG_OPENTHREAD_DEFAULT_TX_POWER`. I hope this suits your needs.

rlubos, 2024-07-25:
As this seems to be resolved, closing the issues. Please reopen if that's not the case.

closed 2024-07-25 by rlubos