zephyrproject-rtos/zephyr #52462: uart: stm32: UART clock source not initialized

r2r0, 2022-11-22:
**Describe the bug**
When there is no secondary clock defined in dts (`clocks` property) then UART clock (`usart_ker_ck`) it is not set at all (in `RCC_CCIPR`).
In case when bootloder has different settings for `RCC_CCIPR` then UART in application stops working (wrong baudarate is set).

Please also mention any information which could help others to understand
the problem you're facing:
 - target platform: STM32WB running on HSE with PLL at 64MHz
 - Without bootloader LPUART works as expected.

**To Reproduce**
Steps to reproduce the behavior:
Use mcuboot with different (than in application) dts clocks settings for one of UARTs:
 i.e. mcuboot: `clocks = <&rcc STM32_CLOCK_BUS_APB1_2 0x00000001>, <&rcc STM32_SRC_HSI LPUART1_SEL(2)>;`
application uses default MCU `clocks` property: `clocks = <&rcc STM32_CLOCK_BUS_APB1_2 0x00000001>;`

**Expected behavior**
UART driver should setup all neccessary clocks to be able to work as specified in dts.

**Impact**
annoyance

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain:  Zephyr SDK 0.15.1
 - Commit SHA: 4c70bf663058b2e2e1592e51692cb1470ed59c6a

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

erwango, 2022-11-22:
@r2r0 Do I understand correctly that if bootloader sets a domain clock, it remains somehow set (in RCC configuration) for application as well (application uart clock setting doesn't set any domain clock,), which leads to bad uart clock configuration in the end ?

r2r0, 2022-11-22:
@erwango Yes, exactly as you wrote.
We used HSI as a second entry in `clocks` property for LPUART in mcuboot and no `clocks` was defined in application dts - the effect was that in application after UART initilization LPUART1SEL in RCC had value equal to 2.
When bootloader was not programmed everthing went well because default value (0) was set and it matched with default dts `clocks` settings.

r2r0, 2022-12-06:
Yes, this is working.
Does it mean that "default" clock cannot be applied to UART (as `usart_ker_ck`) but if there is additional one then it can be?
For now `clock_control_configure` is called only in the case when `config->pclk_len > 1` - why it cannot be called also in case when `config->pclk_len == 1` with default value (`0`)?

erwango, 2022-12-07:
> For now clock_control_configure is called only in the case when config->pclk_len > 1 - why it cannot be called also in case when config->pclk_len == 1 with default value (0)?

Yes, but this is basically often useless additional code for each driver (enforcing reset value), for a specific case that can easily be handled otherwise.

erwango, 2022-12-16:
@r2r0, I'm closing the issue as there is not much more that we plan to do.

closed 2022-12-16 by erwango