zephyrproject-rtos/zephyr #52479: incorrect canopennode SDO CRC

iperry, 2022-11-22:
I am testing block SDO transfers from a master using the Lely Industries CANOpen stack to a slave running the Zephyr canopennode stack. 

When attempting to use block SDO uploads (reading a test chunk from the slave), CRC errors are encountered. I narrowed the problem down to the CRC check used inside the canopen stack, and its interaction with the zephyr CRC.

1. The Zephyr CO stack is configured to use Zephyr internal CRC16 implementation (CO_driver_target.h)
```
/* Use Zephyr provided crc16 implementation */
#define CO_USE_OWN_CRC16
```
2. The CANopen CRC implementation and the Zephyr implementation differ in call signatures. The CO stack uses the following signature (modules/lib/canopennode/stack/crc16-ccitt.h):
```
#ifdef CO_USE_OWN_CRC16
extern
#endif
unsigned short crc16_ccitt(
        const unsigned char     block[],
        unsigned int            blockLength,
        unsigned short          crc);
```
But, the zephyr signature is (zephyr/include/sys/crc.h)
```uint16_t crc16_ccitt(uint16_t seed, const uint8_t *src, size_t len);```
3. The zephyr version of crc16_ccitt() is not the same version as specified by [CIA 301](https://forum.opencyphal.org/uploads/short-url/mNWuvY23DYckSFoSbfam0ji3YWn.pdf). The Zephyr version of the CRC as specified in CIA 301 is named `crc16_itu_t` in crc.h.
I tested this by looking at the test message with data "123456789,", which should yield the checksum 0x31C3 (see 7.2.4.3.16 CRC calculation algorithm to verify SDO block transfer).

I applied the following patch and it "works for me:"
```diff --git a/stack/crc16-ccitt.h b/stack/crc16-ccitt.h
index 9d3344f..aebf442 100644
--- a/stack/crc16-ccitt.h
+++ b/stack/crc16-ccitt.h
@@ -56,12 +56,15 @@ extern "C" {
  * @return Calculated CRC.
  */
 #ifdef CO_USE_OWN_CRC16
-extern
-#endif
+extern uint16_t crc16_itu_t(uint16_t seed, const uint8_t *src, size_t len);
+#define crc16_ccitt(block, blockLength, crc) \
+  crc16_itu_t(crc, block, blockLength)
+#else
 unsigned short crc16_ccitt(
         const unsigned char     block[],
         unsigned int            blockLength,
         unsigned short          crc);
+#endif
 
 #ifdef __cplusplus
 }
```
 
Please also mention any information which could help others to understand
the problem you're facing:
 - Zephyr v3.2.0
 - See patch above for workaround

**To Reproduce**
Steps to reproduce the behavior:
1. Run canopennode example
2. Send SDO block read
3. Check CRC 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr SDK 0.15.1
 - v3.2.0
[sdo_test.tar.gz](https://github.com/zephyrproject-rtos/zephyr/files/10071385/sdo_test.tar.gz)

henrikbrixandersen, 2023-01-09:
@iperry Thank you. Please test https://github.com/zephyrproject-rtos/zephyr/pull/53620 and report back.

closed 2023-01-11 by carlescufi

iparobin, 2023-04-12:
I am testing the sending of a large number of data and I wanted to know how to select in the config the choice between segmented date or block transfer ?

PR #53620 (2023-01-11): manifest: update CANopenNode to fix SDO block transfer CRC function
Update CANopenNode revision to fix the CRC calculation for SDO block transfers. Add SDO block transfer support for the CANopen program download west runner. SDO block transfers are experimental in the underlying CANopen for Python.

Fixes: #52479
commit 16552e86 (2023-01-11): manifest: update CANopenNode to fix SDO block transfer CRC function

Update CANopenNode revision to fix the CRC calculation for SDO block
transfers.

Fixes: #52479