zephyrproject-rtos/zephyr #52490: STM32: create synchronious PWMs

krebsbstn, 2022-11-23:
**Describe the bug**
Hi, 
I am Working on a STM32-H7A3ZI-Q, trying to drive a TCD1304-Sensor.
Therefore it is needed to create at least two PWMs with 2MHz and 500KHz respectively.
The Datasheet of STM32-H7A3ZI-Q points out, that there is a possibility of Timer Link feature.

To get those PWMs synchronized to each other, I tried using the same Timer with different channels.
This is not working because the output signals will have the same frequency but different duty cycles.

Creating the two PWMs with different Timers they have a drift referring to each other.
As consequence the Timing requirements of TCD1304 cannot be held. 

Without Zephyr you might have to drive the two Timers as Slaves of another Timer to get them synchronized.
Is there a Way to Synchronize the PWMs with the Device tree?

**To Reproduce**
create a new Project with the following contents for a STM-H7A3ZI-Q.
*prj.conf*
```
CONFIG_PWM=y
CONFIG_SYS_CLOCK_TICKS_PER_SEC=100
```

*main.c*
```
#define FM_PERIOD 2000000U
#define SH_PERIOD 500000U

void main(){
    /* Initialize and start PWM Signals */
    const struct device *fM_device = device_get_binding("FM");
    const struct device *SH_device = device_get_binding("SH");
    if (!fM_device | !SH_device){
        printk("TCM1304 - PWM initialize failed.");
        return;
    }
    pwm_set(fM_device, 1, FM_PERIOD, 0.5*FM_PERIOD, 0);
    pwm_set(SH_device, 4, SH_PERIOD, 0.5*SH_PERIOD, 0);
    while(1){
        ;
    }
};
```
*nucleo_h7a3zi_q.overlay*
```
...
&timers1 {
    st,prescaler = <10>;
    status = "okay";
    
    pwm1: pwm {
        status = "okay";
        label = "FM";
        pinctrl-names = "default";
        pinctrl-0 = <&tim1_ch1_pe9>;
    };
};

&timers4 {
    st,prescaler = <100>;
    status = "okay";

    pwm3: pwm {
        status = "okay";
        label = "SH";
        pinctrl-names = "default";
        pinctrl-0 = <&tim4_ch4_pd15>;
    };
};
...
```

**Expected behavior**
Seeing two synchronized PWMs on the both Pins.

**Impact**
It is not possible to drive complex Sensors with PWMs.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - STM32-CubeIDE

**Additional context**
The PWM Signals can be measured with an oscilloscope at the pins PE9 and PD15

jpowen898, 2024-02-14:
@zephyrbot and @erwango Is this feature still in the works? I am working on an existing project using stm32's and we are wanting to switch zephyr in the near future but we the cascaded timer feature. Can you give me a timeline on when this would be implemented? Or is there a work around where I could access the stm32 hal directly and implement a temporary work around?

erwango, 2024-02-15:
@jpowen898 This feature is not in our priority list. To access STM32 HAL directly from your application see https://github.com/zephyrproject-rtos/hal_stm32?tab=readme-ov-file#use-stm32cube-in-your-application