zephyrproject-rtos/zephyr #52491: Value of EVENT_OVERHEAD_START_US is set to low

KKopyscinski, 2022-11-23:
**Describe the bug**
Commit https://github.com/zephyrproject-rtos/zephyr/commit/236d0351e70abd0aa4b1a0cdfddf99e18162f63d set to low value of EVENT_OVERHEAD_START_US.

While testing Bluetooth Mesh 1.1 with tester app I encountered a problem with assert in `lll`:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c#L476-L477

I was using application build from https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/bluetooth/tester, with `prj.conf`. Logs were collected using test `MESH/CFGCL/KR/BV-04-C`, so `CONFIG_BT_PRIVACY` was also enabled, but we can observe this also for other tests with default config. This bug typically happens at provisioning step or when big amounts of data are being advertised/received. I'm running app on nrf52840, pca10056. Calculated value varies from run to run, typically is above 300us.

**To Reproduce**
Build tester application and run test - this can also be encountered when mesh_shell is used, in the same way on same platform.

**Expected behavior**
I think value that triggers assert should be higher, previously was 750us - I haven't seen value that high during testing, but I don't know how low can it be set safely.

**Impact**
This is making testing very difficult, and some tests are not able to pass at all. Showstopper. 

**Logs and console output**
```
[00:00:07.877,441] <dbg> bttester_mesh: provision_node: 
[00:00:07.877,471] <inf> bt_mesh_main: bt_mesh_provision: Primary Element: 0x0003
[00:00:07.877,502] <dbg> bt_mesh_main: bt_mesh_provision: net_idx 0x0000 flags 0x00 iv_index 0x0000
[00:00:07.887,451] <dbg> bttester_mesh: prov_complete: net_idx 0x0000 addr 0x0003
[00:00:08.040,893] <dbg> bttester_mesh: provision_adv: 
[00:00:08.567,230] <dbg> bttester_mesh: link_open: bearer 0x01
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:476
	lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 305
[00:00:09.968,444] <err> os: esf_dump: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:00:09.968,475] <err> os: esf_dump: r3/a4:  0x20003098 r12/ip:  0x2000d72c r14/lr:  0x00002133
[00:00:09.968,475] <err> os: esf_dump:  xpsr:  0x41000011
[00:00:09.968,505] <err> os: esf_dump: Faulting instruction address (r15/pc): 0x0000213e
[00:00:09.968,536] <err> os: z_fatal_error: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:09.968,566] <err> os: z_fatal_error: Fault during interrupt handling

[00:00:09.968,597] <err> os: z_fatal_error: Current thread: 0x20005878 (idle)
[00:00:17.464,599] <err> os: k_sys_fatal_error_handler: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu Linux 22.04 LTS
 - Toolchain: Zephyr SDK
 - Build from private fork for testing Mesh 1.1 (controller not modified, changes only to host)

KKopyscinski, 2022-11-23:
FYI @omkar3141

cvinayak, 2022-11-23:
@KKopyscinski Please check if your branch has this commit: https://github.com/zephyrproject-rtos/zephyr/commit/13c29940bf69a30e6d3285859ebd9d00708cb2d7

KKopyscinski, 2022-11-23:
It has not, but that's different issue. This issue would result in very big values (obviously)  but the problem is that values are small and reasonable, just slightly above threshold.

jakkra, 2022-12-15:
@cvinayak adding myself to the list have this issue. I'll try with the patch in https://github.com/zephyrproject-rtos/zephyr/commit/13c29940bf69a30e6d3285859ebd9d00708cb2d7 and come back. 
I don't have a comples scenario to reproduce, it pretty much happens as soon I start periodic advertising. 

[prj_conf.zip](https://github.com/zephyrproject-rtos/zephyr/files/10235294/prj_conf.zip)

jakkra, 2022-12-15:
@cvinayak This is a custom PCB with NRF52833 I'm running on.

This is the assert I get:
```
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:476
lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 14465
```
The actual number changes.

I replaced the assert with a LOG_ERR and I get lots of those.
Actually I think now that this may be a different problem. My application uses LVGL to draw on a screen. If I don't draw, I have no asserts. 

I debugged a bit and it's not LVGL that is the problem, it's SPI.
See this log:
```
SPI send 11396 bytes
00> [00:00:15.675,598] <err> bt_ctlr_lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 35217
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 2072 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 6080 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 144 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 2016 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 112 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 1056 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 11396 bytes
00> [00:00:16.675,933] <err> bt_ctlr_lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 15563
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 2072 bytes
00> SPI send 4 bytes
00> SPI send 4 bytes
00> SPI send 6080 bytes
00> [00:00:16.872,192] <err> bt_ctlr_lll: lll_preempt_calc: Actual EVENT_OVERHEAD_START_US = 1831
```

Every time a larger chunk is sent I get that assert/log in lll.c

I did not have any asserts using nRF Connect 2.1, I got this after upgrading to 2.2 due to BLE disconnects after about ~30s on 2.1 which now works fine in 2.2. 

Do you have any pointers in what the problem here could be? This is the "problematic" SPI code in my display driver:
```
static inline int gc9a01_write_cmd(const struct device *dev, uint8_t cmd,
				   uint8_t *data, size_t len)
{
	const struct gc9a01_config *config = dev->config;
	struct spi_buf buf = {.buf = &cmd, .len = sizeof(cmd)};
	struct spi_buf_set buf_set = {.buffers = &buf, .count = 1};
	gpio_pin_set_dt(&config->dc_gpio, 0);
	if (spi_write_dt(&config->bus, &buf_set) != 0) {
		printk("FAIL\n");
		return -EIO;
	}

	if (data != NULL && len != 0) {
		buf.buf = data;
	        buf.len = len;
		printk("SPI send %d bytes\n", len);
		gpio_pin_set_dt(&config->dc_gpio, 1);
		if (spi_write_dt(&config->bus, &buf_set) != 0) {
			printk("FAIL\n");
			return -EIO;
		}
	}

	return 0;
}

Sorry for hijacking this issue if this is unrelated, then let me know.

cvinayak, 2022-12-16:
> Please try to use lower priorities for other SoC peripherals used in your application by having the below
>
> #define NRF_DEFAULT_IRQ_PRIORITY 5
>
> inside your application folder in dts/arm/nordic/override.dtsi file.
> You can verify the assigned interrupt priority for peripherals in the build generated `zephyr.dts` file where the peripherals have the line `interrupts = < 0xN 0x5 >;`

This is the first solution I ask when the `EVENT_OVERHEAD_START_US` assertion is reported. (in fact, I added the assert deliberately to catch users having ISRs that introduce higher than acceptable ISR latencies). The Controller design remains the same before and after the introduction of the assert check, if the check is removed and user application have higher latency the BLE Radio events get skipped (not a big issue for low throughput applications). Repeated skips lead to supervision/sync timeouts.

mniestroj, 2022-12-21:
I've just run into this assertion. My app uses DFU over ESP-AT WiFi chip, which means lots of UART interrupts and flash write/erase operations. I see muliple issues on Github already related to this assertion. Should this assertion recommend steps to reconfigure application, such as overriding `NRF_DEFAULT_IRQ_PRIORITY`? This option seems to help my case.

jakkra, 2022-12-22:
Thanks, `#define NRF_DEFAULT_IRQ_PRIORITY 5` fixed it for me too.

xudongzheng, 2023-03-11:
I am running Zephyr 3.3 and am getting the same assert. This is for a board running 24/7 and the assert happens approximately 2-3 times a week.

I've isolated the issue to https://github.com/zephyrproject-rtos/zephyr/blob/v3.3.0/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c#L473-L476 based on core dump analyzed with GDB.

The board has SPI (WS2812B) and QSPI (P25Q16H) though neither should have been in use at the time of the crash if my logs are accurate.

This happens both with and without `#define NRF_DEFAULT_IRQ_PRIORITY 5` as suggested in https://github.com/zephyrproject-rtos/zephyr/issues/52491#issuecomment-1354207194.

I've confirmed that build/zephyr/zephyr.dts includes `interrupts = < 0xN 0x5 >`:

```
clock: clock@40000000 {
	compatible = "nordic,nrf-clock";
	reg = < 0x40000000 0x1000 >;
	interrupts = < 0x0 0x5 >;
	status = "okay";
};
power: power@40000000 {
	compatible = "nordic,nrf-power";
	reg = < 0x40000000 0x1000 >;
	interrupts = < 0x0 0x5 >;
	status = "okay";
};
radio: radio@40001000 {
	compatible = "nordic,nrf-radio";
	reg = < 0x40001000 0x1000 >;
	interrupts = < 0x1 0x5 >;
	status = "okay";
	ieee802154-supported;
	ble-2mbps-supported;
	ble-coded-phy-supported;
	tx-high-power-supported;
	ieee802154: ieee802154 {
		compatible = "nordic,nrf-ieee802154";
		status = "disabled";
	};
};
```

I noticed that the priority changes for the radio as well I'm not familiar enough with Zephyr to know whether that's good or bad.

Please let me know if anyone has other suggestions for debugging this crash.

cvinayak, 2023-03-12:
> I am running Zephyr 3.3 and am getting the same assert. This is for an nRF52840 board (Seeeduino XIAO, not the sense version) running 24/7 and the assert happens approximately 2-3 times a week.

Are you too using mesh stack? Please provide the actual assert message, details of building, flashing and steps to reproduce the assertion.

xudongzheng, 2023-03-14:
Oops, I should have specified. I'm not using the mesh stack.

It's a custom firmware with 2 Bluetooth connections (as central) to peripherals and about 4-6 Bluetooth connections (as peripheral) to centrals. Unfortunately I don't have a reliable way of reproducing the assert. It just appears randomly while the firmware is mostly idle with devices connected. It appears on average 2-3 times a week so it's mostly just a minor inconvenience.

I've modified my firmware to print the EVENT_OVERHEAD_START_US value and will report back once I have that information.

cvinayak, 2023-03-14:
@xudongzheng Thank you for following up and help debug. Will wait from you for the assert message value it prints.

If I understand correct, you have a device that is central to 2 other peripherals and also is multiple peripheral to other central devices, concurrently.

I am suspecting connection events drift onto each other over a period of time and possibly a bug in controller's scheduling implementation under certain edge conditions uncovers the said assertion.

Once you reproduce the assertion, may be you can help me verify if this PR helps fix your issue: https://github.com/zephyrproject-rtos/zephyr/pull/54848

You can cherry-pick the commit into your repo.

xudongzheng, 2023-03-23:
When I replaced `LL_ASSERT_MSG` with `LOG_ERR`, I got `Actual EVENT_OVERHEAD_START_US = 305`. Per my logs, everything continued operating normally despite the `EVENT_OVERHEAD_START_US` error message; there were no disconnects to any of the connected centrals/peripherals.

I subsequently cherry picked #54848 while keeping `LOG_ERR`. The resulting firmware has been running for approximately 6 days and the `EVENT_OVERHEAD_START_US` error has not reappeared. Given the rarity of the assert/error (only happening about twice a week), it's too soon to say definitely that #54848 fixed the issue but it does look like it thus far.

cvinayak, 2023-03-24:
Thank you @xudongzheng for reporting your observations and help verify.

> everything continued operating normally despite the EVENT_OVERHEAD_START_US error message;

It is safe to remove this assertion check (, continue the rest of the prepare function and `return 0`), the controller is design to operate with some radio events be skipped if for reasons the preparation is delayed. Say, an application design needs uninterrupted use of CPU and can tolerate skipped radio events.
The assertion is added to ensure there is no repeated skipping of events that can lead to supervision timeouts or livelocks. An improvement could be to assert on n repeated delays, maybe.

> I subsequently cherry picked https://github.com/zephyrproject-rtos/zephyr/pull/54848 while keeping LOG_ERR. The resulting firmware has been running for approximately 6 days and the [...]

#54848 is a valid fix the same assertion discovered during manual testing continuous scanning alongwith more than one other radio role being active. Yes, the assertion check is entirely timing dependent, having thread and multiple ISR executing in some particular order that could delay the radio prepare routines.

Other related bug fixes that reduce the EVENT_OVERHEAD_START_US value are:
https://github.com/zephyrproject-rtos/zephyr/pull/55945
https://github.com/zephyrproject-rtos/zephyr/pull/55443

Hope, you will continue to run your tests and please report back if the assertion/log shows up. Thanking you again.

PR #55443 (2023-03-24): Bluetooth: Controller: Fix ticker to reduce RTC min compare offset
Fix ticker to explicitly trigger worker when ticks_to_expire is less than or equal to elapsed time.

Fixes #52491.
closed 2023-03-24 by carlescufi