zephyrproject-rtos/zephyr #52493: net: lwm2m: add 32 bits floating point support

RomainPelletant, 2022-11-23:
**Is your enhancement proposal related to a problem? Please describe.**
lwm2m_engine_set_float function only supports double and not directly 32 bits float.
Maybe there is a reason why 32 bits float has been deprecated?

**Describe the solution you'd like**
Replace lwm2m_engine_set_float for 32 bits floating points and add lwm2m_engine_set_double for 64 bits ones.

**Describe alternatives you've considered**
For low impacts on existing code, you could implement lwm2m_engine_set_float for 64 bits and lwm2m_engine_set_float32 for 32 bits ones.

rlubos, 2022-11-23:
Can you elaborate on the rationale for why do you need a 32-bit float support in the first place? Having to support both, 32-bit and 64-bit floating point types for **data storage** complicates the LwM2M float processing logic **a lot**. And it's not a limitation from the LwM2M perspective, the engine can parse both, 32-bit and 64-bit floats if present in the message.

> Maybe there is a reason why 32 bits float has been deprecated?

Well, I wouldn't say it has been deprecated provided that native 32-bit float type has never been supported by the engine. The engine used to have special structs to represent 32-bit and 64-bit float types, but it has been generally agreed that they only overcomplicate the LwM2M logic, w/o clear benefits. Hence it was replaced by native `double` type for data storage. You can follow the discussions/rationale here:
https://github.com/zephyrproject-rtos/zephyr/pull/37146
https://github.com/zephyrproject-rtos/zephyr/pull/39367

> For low impacts on existing code, you could implement lwm2m_engine_set_float for 64 bits and lwm2m_engine_set_float32 for 32 bits ones.

There'll never be a low impact for such a change, as the engine would need to contain extra logic to encode/decode 32-bit floats. Therefore I'm rather against such changes unless reasonably argumented.

RomainPelletant, 2022-11-29:
Not a real need, nor any rationale. Just asking cause we are using double and wanted to know if it was possible to use 32 bits float directly.

closed 2022-11-29 by RomainPelletant