zephyrproject-rtos/zephyr #52506: GPIO gap incorrect location compare

28itay, 2022-11-23:
**Describe the bug**
The gpio configuration failed because of a gpio pin that "fall" in a gap.

I'm using a custom board with MIMXRT1021CAF4A MCU (it was only 100 pins compare to it's bigger versions), there are 2 gaps in gpio2 one from 0 to 3 and the second from 10 to 15, this was generated using imx_cfg_utils.py (from https://github.com/zephyrproject-rtos/hal_nxp).

[image]

While trying to configure gpio_emc_18 the configurate returned -134 (ENOTSUP) error, while debugging the code I found that `mcux_igpio_configure` in drivers/gpio/gpio_mcux_igpio.c in line 58 checks if the pin number fall inside a gap, thinks that pin 18 is inside the second gap, because the gaps values are absolute while the value that it's being compared to is it's relative value.

**To Reproduce**
This isn't was straight forwarded.

I used MIMXRT1020_EVK as a reference board, changed it's MCU to MIMXRT1021CAF4A, generated the relevant gpio pin out using imx_cfg_utils.py.

And tried to configured gpio_emc_18.

**Expected behavior**
For the gpio to be configured as it does exist.

**Impact**
I can't configured the 4 existing gpios that are above the second gap which are gpio_emc_16 through gpio_emc_19.

**Logs and console output**
[image]
[image]

**Environment (please complete the following information):**
 - OS: WSL
 - Toolchain Zephyr SDK 0.15.1
 - Zephyr commit SHA: 4c53a4067a6750987e4ebca354d7e00514b5bd45
 - HAL/NXP commit SHA: 3e3648526824cc5b52ab59d5dd308473e291b760

28itay, 2022-11-27:
@DerekSnell 
I'm not sure hot to submit a code 😅, but the change is really simple.
Change line 59 from:
`if (cfg_idx < (config->pin_gaps[i].start +`
to:
`if (pin< (config->pin_gaps[i].start +`

Basically checking if `pin` (the absolute pin value) is in side the current gap (which is also being specified by an absolute values)

danieldegrasse, 2022-11-28:
Hi @28itay, I've tested your proposed fix on an RT1015 EVK (since the SOC present there has the same pin mux gaps as the RT1021 SOC you have). PR #52610 should fix this issue, it appears that both checks should use `pin` over `cfg_idx`

PR #52610 (2022-12-22): drivers: gpio: fix gpio-reserved-ranges handling in MCUX iGPIO driver
Fix handling of gpio-reserved-ranges within MCUX iGPIO driver, to ensure that the configuration IDX will be correctly calculated for pins where multiple reserved ranges are present on the GPIO controller

Fixes #52506
commit 21c4957c (2022-12-22): drivers: gpio: fix gpio-reserved-ranges handling in MCUX iGPIO driver

Fix handling of gpio-reserved-ranges within MCUX iGPIO driver, to ensure
that the configuration IDX will be correctly calculated for pins where
multiple reserved ranges are present on the GPIO controller

Fixes #52506
closed 2022-12-22 by carlescufi