zephyrproject-rtos/zephyr #52510: twister: truncated handler.log reported as "failed"

FRASTM, 2022-11-23:
**Describe the bug**
When running a test campaign on stm32 platforms, some testcases have uncomplete handler.log  and twister report it as failed but test is individually passed

Missing the :
```
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: xxxxxxxxxxxxxxxxxxxxxxxxx
PROJECT EXECUTION SUCCESSFUL
```

**Expected behavior**
full handler.log   file stored and test reported "passed"

**Impact**
all test declared as _failed_ are not.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
COMMAND : twister --ninja --runtime-artifact-cleanup --disable-suite-name-check --hardware-map ./zephyr/../map.yaml --device-testing --outdir ../logs/221121/log_dt_main_59f3316d33_2104

For example: Failed tests: zephyr/tests/subsys/shell/shell/shell.core, nucleo_wb55rg
json test results:
```
       {
            "name":"zephyr/tests/subsys/shell/shell/shell.core",
            "arch":"arm",
            "platform":"nucleo_wb55rg",
            "run_id":"0d46eb7c4d7a84ef4b7d49b7f0e37533",
            "runnable":true,
            "retries":0,
            "status":"error",
            "reason":"No Console Output(Timeout)",
            "log":"*** Booting Zephyr OS build zephyr-v3.2.0-1833-g59f3316d33a1 ***\r\nRunning TESTSUITE shell\r\n====================================================>
            "execution_time":"66.84",
            "testcases":[
                {
                    "identifier":"shell.core.cmd_help",
                    "execution_time":"0.02",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_clear",
                    "execution_time":"0.02",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_dict",
                    "execution_time":"0.01",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_history",
                    "execution_time":"0.02",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_mand_1_opt_raw",
                    "execution_time":"0.02",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_resize",
                    "execution_time":"0.04",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_select",
                    "execution_time":"0.03",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.cmd_shell",
                    "execution_time":"0.28",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.mand_2_opt_raw",
                    "execution_time":"0.03",
                    "status":"passed"
                },
               {
                    "identifier":"shell.core.max_argc",
                    "execution_time":"0.01",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.section_cmd",
                    "execution_time":"0.02",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.set_root_cmd",
                    "execution_time":"0.02",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.shell_fprintf",
                    "execution_time":"0.00",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.shell_module",
                    "execution_time":"0.03",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.shell_wildcards_dynamic",
                    "execution_time":"0.01",
                    "status":"passed"
                },
                {
                    "identifier":"shell.core.shell_wildcards_static",
                    "execution_time":"0.03",
                    "status":"passed"
                }
            ]
        },
```

nucleo_wb55rg/zephyr/tests/subsys/shell/shell/shell.core/handler.log : 
```
...
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [shell]: pass = 15, fail = 0, skip = 0, total = 15 duration = 0.575 seconds
- PASS - [shell.test_cmd_clear] duration = 0.017 seconds
- PASS - [shell.test_cmd_dict] duration = 0.012 seconds
- PASS - [shell.test_cmd_history] duration = 0.019 seconds
- PASS - [shell.test_cmd_mand_1_opt_raw] duration = 0.017 seconds
- PASS - [shell.test_cmd_resize] duration = 0.039 seconds
- PASS - [shell.test_cmd_select] duration = 0.027 seconds
- PASS - [shell.test_cmd_shell] duration = 0.284 seconds
- PASS - [shell.test_mand_2_opt_raw] duration = 0.029 seconds
- PASS - [shell.test_max_argc] duration = 0.015 seconds
- PASS - [shell.test_section_cmd] duration = 0.021 seconds
- PASS - [shell.test_set_root_cmd] duration = 0.019 seconds

```

FRASTM, 2022-11-23:
attached : ../logs/221121/log_dt_main_59f3316d33_2104/nucleo_wb55rg/zephyr/tests/subsys/shell/shell/shell.core/handler.log for the testcase
[handler.log](https://github.com/zephyrproject-rtos/zephyr/files/10077652/handler.log)

erwango, 2022-12-19:
@hakehuang Do you see similar issues on your side ?

PerMac, 2023-02-23:
Twister reports that the proper log was not found within the timeout. However, execution times indicates that it was a quick one, it's just those last lines you mentioned missing. Is thiserror a random one or easily reproduceable? Have you try looking directly at the output in real-time e.g. with minicom? When error is reported, are always those last lines missing, or bigger chunks of the output as well? 
I am trying to figure out at which level the error is located. Right now it doesn't look like twister error to me. Twister expects a certain output which is not provided. run_id is also missing. I would first try to figure out why those lines are missing. You are already disabling suite name check, which I guess was needed due to some early output which can be missing in your logs. I think it is fine since run_id printed at the end is already a strong check making sure the output comes from an exact build. If this check is also disable (and it is missing in your output) than IMO you would loose a reliable verification of what was executed.
BTW we haven't seen issues with missing parts of outputs in our (Nordic) internal on-target twister setups.

golowanow, 2023-03-13:
@FRASTM, could you please check #55659 recently (13/Mar/2023) merged to the master branch ?
Does it resolve this issue observed on your setup ?

hakehuang, 2023-03-14:
> @hakehuang Do you see similar issues on your side ?

sorry for later reply, I did not see such issue. the only simliar issue at my side is the test takes too long on my board such as cmsis-dap cases.

nashif, 2023-04-06:
not able to reproduce, please reopen if still a bug and you are able o reproduce with latest codebase.

closed 2023-12-17 by github-actions[bot]