zephyrproject-rtos/zephyr #52513: sample.modules.chre fails on qemu_leon3

fabiobaltieri, 2022-11-23:
**Describe the bug**

`zephyr/samples/modules/chre/sample.modules.chre` fails to run on `qemu_leon3`. The test started to try and run on this platform since e478a4e85083dddcbc2cd7282f413f0d89d3a1eb, so it probably never ran successfully.

This is the run and the output:

```
$ west build -t run -p -b qemu_leon3 samples/modules/chre
...
*** Booting Zephyr OS build zephyr-v3.2.0-1913-g8f72c8d53b50 ***
Failed to set thread name to "CHRE": -88
Hello CHRE!
I: CHRE init, version: undefined
I: EventLoop start
D: Instance ID 1 assigned to app ID 0x0000000000000001
EchoApp::nanoappStart()
E: 
E: tt = 0x07, mem_address_not_aligned
E: 
E:       INS        LOCALS     OUTS       GLOBALS
E:   0:  40008bb0   f34000c6   00000000   00000000
E:   1:  00000003   40007130   00000001   40008ad0
E:   2:  4000a7dc   40001658   ffffffff   00000000
E:   3:  400050fc   00000001   ffffffff   00000033
E:   4:  ffffffff   00000002   0000000f   00000001
E:   5:  4000a73c   00000000   00000000   00000080
E:   6:  4000d128   40000060   4000d0c8   00000000
E:   7:  400050e8   00000000   40004b30   00000000
E: 
E: psr: f30000c5   wim: 00000080   tbr: 40000070   y: 00000000
E:  pc: 40004b38   npc: 40004b3c
E: 
E:       pc         sp
E:  #0   40004b38   4000d0c8
E:  #1   400050e8   4000d128
E:  #2   4000165c   4000d190
E:  #3   40005b7c   4000d200
E:  #4   40001820   4000d260
E:  #5   40003468   4000d2c0
E: 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x40009e00 (unknown)
E: Halting system
```

```
$ addr2line -e build/zephyr/zephyr.elf 40004b38
/usr/local/google/home/fabiobaltieri/zephyrproject/modules/lib/chre/core/event_loop.cc:497
```

**To Reproduce**
`west build -t run -p -b qemu_leon3 samples/modules/chre`

**Expected behavior**
Does not crash.

**Impact**
Breaking CI, excluding the platform.

**Environment (please complete the following information):**
Linux, Zephyr SDK 0.15.0, a95c93af34 (current main)

tbr-tt, 2023-10-02:
It looks like the memory provided by `memoryAlloc()` in the chre module is not properly aligned for an object of type `chreNanoappInfo`.

The following change should make the sample run to completion:

```diff
--- a/platform/zephyr/memory.cc
+++ b/platform/zephyr/memory.cc
@@ -24,7 +24,7 @@ K_HEAP_DEFINE(chre_mem, CONFIG_CHRE_DYNAMIC_MEMORY_SIZE);
 namespace chre {
 
 void *memoryAlloc(size_t size) {
-  return k_heap_alloc(&chre_mem, size, K_NO_WAIT);
+  return k_heap_aligned_alloc(&chre_mem, __alignof__(z_max_align_t), size, K_NO_WAIT);
 }
```

fabiobaltieri, 2023-10-02:
@tbr-tt thanks, could you open a PR for this? Alternatively, @yperess could you help this gets fixed upstream?

yperess, 2023-10-02:
> @tbr-tt thanks, could you open a PR for this? Alternatively, @yperess could you help this gets fixed upstream?

Yes, I'll try to do it this week

yperess, 2023-10-17:
Still haven't looked into this, but haven't forgotten

closed 2023-12-31 by github-actions[bot]