zephyrproject-rtos/zephyr #52526: Simplify lvgl.h file by creating more header files inside module's sub-filoders (src/widgets and src/layouts)

bbkarim, 2022-11-24:
**Make the lvgl.h more simple and short by adding new header files and include them.**

**Describe the solution you'd like**
Create below header files:
- `src/libs/lv_libs.h` with the content:
```c
#include "src/libs/bmp/lv_bmp.h"
#include "src/libs/fsdrv/lv_fsdrv.h"
#include "src/libs/png/lv_png.h"
#include "src/libs/gif/lv_gif.h"
#include "src/libs/qrcode/lv_qrcode.h"
#include "src/libs/sjpg/lv_sjpg.h"
#include "src/libs/freetype/lv_freetype.h"
#include "src/libs/rlottie/lv_rlottie.h"
#include "src/libs/ffmpeg/lv_ffmpeg.h"
#include "src/libs/tiny_ttf/lv_tiny_ttf.h"
#include "src/libs/bmp/lv_bmp.h"
#include "src/libs/fsdrv/lv_fsdrv.h"
#include "src/libs/png/lv_png.h"
#include "src/libs/gif/lv_gif.h"
#include "src/libs/qrcode/lv_qrcode.h"
#include "src/libs/sjpg/lv_sjpg.h"
#include "src/libs/freetype/lv_freetype.h"
#include "src/libs/rlottie/lv_rlottie.h"
#include "src/libs/ffmpeg/lv_ffmpeg.h"
#include "src/libs/tiny_ttf/lv_tiny_ttf.h"
```
- `src/widgets/lv_widgets.h` with the content:
```c
#include "src/widgets/animimg/lv_animimg.h"
#include "src/widgets/arc/lv_arc.h"
#include "src/widgets/bar/lv_bar.h"
#include "src/widgets/btn/lv_btn.h"
#include "src/widgets/btnmatrix/lv_btnmatrix.h"
#include "src/widgets/calendar/lv_calendar.h"
#include "src/widgets/canvas/lv_canvas.h"
#include "src/widgets/chart/lv_chart.h"
#include "src/widgets/checkbox/lv_checkbox.h"
#include "src/widgets/colorwheel/lv_colorwheel.h"
#include "src/widgets/dropdown/lv_dropdown.h"
#include "src/widgets/img/lv_img.h"
#include "src/widgets/imgbtn/lv_imgbtn.h"
#include "src/widgets/keyboard/lv_keyboard.h"
#include "src/widgets/label/lv_label.h"
#include "src/widgets/led/lv_led.h"
#include "src/widgets/line/lv_line.h"
#include "src/widgets/list/lv_list.h"
#include "src/widgets/menu/lv_menu.h"
#include "src/widgets/meter/lv_meter.h"
#include "src/widgets/msgbox/lv_msgbox.h"
#include "src/widgets/roller/lv_roller.h"
#include "src/widgets/slider/lv_slider.h"
#include "src/widgets/span/lv_span.h"
#include "src/widgets/spinbox/lv_spinbox.h"
#include "src/widgets/spinner/lv_spinner.h"
#include "src/widgets/switch/lv_switch.h"
#include "src/widgets/table/lv_table.h"
#include "src/widgets/tabview/lv_tabview.h"
#include "src/widgets/textarea/lv_textarea.h"
#include "src/widgets/tileview/lv_tileview.h"
#include "src/widgets/win/lv_win.h"
```
- `src/layouts/lv_layouts.h` with the content:
```c
#include "src/layouts/flex/lv_flex.h"
#include "src/layouts/grid/lv_grid.h"
```
- `src/others/lv_others.h` with the content:
```c
#include "src/others/snapshot/lv_snapshot.h"
#include "src/others/monkey/lv_monkey.h"
#include "src/others/gridnav/lv_gridnav.h"
#include "src/others/fragment/lv_fragment.h"
#include "src/others/imgfont/lv_imgfont.h"
#include "src/others/msg/lv_msg.h"
#include "src/others/ime/lv_ime_pinyin.h"
#include "src/others/file_explorer/lv_file_explorer.h"
```

And include them from `lvgl.h`
```c
#include "src/libs/lv_libs.h"
#include "src/widgets/lv_widgets.h"
#include "src/layouts/lv_layout.h"
#include "src/others/lv_others.h"
```

carlescufi, 2022-11-24:
@bbkarim thanks for the suggestion. Would you be able to send a Pull Request implementing this?

bbkarim, 2022-11-24:
Sorry this is not targetign this project !

bbkarim, 2022-11-24:
INVALID

closed 2022-11-24 by bbkarim