zephyrproject-rtos/zephyr #52556: adc driver sample was failing with STM32 nucleo_f429zi board

rjmaris, 2022-11-25:
An overlay file for the nucleo_f429zi board for the adc driver sample is missing. Hence ADC is not operating despite suggestion in the board description.

The overlay file has been created now, based on another overlay file. However I don't succeed in integrating both inputs of adc1 as well as adc3. Therefore, I outcommented the input description for the one adc3-input, which is tied to A3 of the Arduino shield connector. It was my intention to define four channels, related to A0...A3. Instead, I defined A0...A2 + A6. A6 is located on CN10 which makes quite sense because analog GND is present at CN10.

Please find a tested (and gitlint checked) patch.

[0001-boards-nucleo_f429zi-add-adc-overlay-file-specifiy-f.patch.txt](https://github.com/zephyrproject-rtos/zephyr/files/10093309/0001-boards-nucleo_f429zi-add-adc-overlay-file-specifiy-f.patch.txt)

gautierg-st, 2023-01-03:
I just tested your patch and I think you are just missing adc3 activation and the definition of the adc3 pin:
```
&adc3 {
	pinctrl-0 = <&adc3_in9_pf3>;
	pinctrl-names = "default";
	status = "okay";
```

erwango, 2023-02-07:
@rjmaris Did it work on your side with proposal from @gautierg-st ?

erwango, 2023-02-07:
Converting to discussion