zephyrproject-rtos/zephyr #52559: Compiler cannot include C++ headers when both PICOLIBC and LIB_CPLUSPLUS options are set

r2r0, 2022-11-25:
**Describe the bug**
When `CONFIG_PICOLIBC` and `CONFIG_LIB_CPLUSPLUS` options are set at the same time then compiler is not able to include C++ headers i.e. `cstdint`.
When there is used `picolib` without enableing usage of C++ standard library then everything works well.
After adding `CONFIG_LIB_CPLUSPLUS` standard C++ headers cannot be found (see log below).

**To Reproduce**
Steps to reproduce the behavior:
Enable picolibc
Enable C++
Enable C++ standard library

**Expected behavior**
If C++ standard library cannot be used with picolib then this should be blocked at the kconfig level, otherwise compilation should succeed.

**Impact**
Showstopper for switching from newlib to picolib.

**Logs and console output**
```
ccache /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++ ......
In file included from /home/hid/aurora/app/src/low_power/wakeup_button.cpp:18:
/home/hid/aurora/app/../shared/src/std_def.h:10:10: fatal error: cstdint: No such file or directory
   10 | #include <cstdint>
      |          ^~~~~~~~~

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA 27ec69eca183887c6b3948b3e92b8cc4f935532e

**Additional context**
C++ related config:
```
CONFIG_PICOLIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_STD_CPP17=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_CPP_MAIN=y
```

stephanosio, 2022-11-25:
Using libstdc++ with picolibc requires the toolchains (i.e. Zephyr SDK) to include the picolibc, which is not currently the case.

The upcoming Zephyr SDK 0.16.0 release will include picolibc and the libstdc++ compiled for it: see https://github.com/zephyrproject-rtos/zephyr/issues/49922#issue-1361765276.

r2r0, 2022-11-25:
Thanks a lot for a prompt answer.
Do you think that it would be a good idea to not allow users to select this combination of options for now?

closed 2023-03-16 by r2r0